osx-cross/homebrew-avr #4: Error: dyld: Library not loaded: /usr/local/lib/libisl.10.dylib

ladislas, 2015-05-21:
As of today, when I tried compiling my code, I got the following error:

```
dyld: Library not loaded: /usr/local/lib/libisl.10.dylib
  Referenced from: /usr/local/Cellar/avr-gcc/4.9.2/libexec/gcc/avr/4.9.2/cc1plus
  Reason: image not found
avr-g++: internal compiler error: Trace/BPT trap: 5 (program cc1plus)
libbacktrace could not find executable to open
Please submit a full bug report,
with preprocessed source if appropriate.
See <http://gcc.gnu.org/bugs.html> for instructions.
make: *** [/Users/ladislas/dev/leka/moti/bin/ColorChanger/uno/main.o] Error 4
```

I `brew remove`d everything and try installing the tool chain again but got an error saying that `ISL` can't be found after trying different versions (I'm sorry I can provide the real error, I closed my terminal window...)

Anyway, after some research it seems that `isl` has been updated in hombrew to `v0.14` which is not supported by `avr-gcc 4.9.x`.

The work around is to `brew tap homebrew/versions` and to modify the formula to install `isl012`. Now the installation works and I can compile my code.

Do any of you have the same issue? If so, maybe we should find a fix as it is not really straight forward to first time users.

Or we can ask ourselves "why is `--with-isl` used?" Do we really need it to compile code on AVR?

larsimmisch, 2015-05-28:
I can reproduce this after a `brew update; brew upgrade; brew cleanup`. I'll take a look at it and maybe update gcc while I am at it

larsimmisch, 2015-05-28:
Sigh. gcc 5.1 is problematic; I can't get it to compile, it crashes in gen-avr-mmcu-specs.

See https://www.mail-archive.com/gcc-bugs@gcc.gnu.org/msg454439.html and also, related, http://savannah.nongnu.org/bugs/?44869

larsimmisch, 2015-05-28:
I have given gcc 5.1 another serious try and I could get it to cross-compile for AVR - but it was not of much use - I couldn't use that `avr-gcc` to compile my test programs.

There is a pretty prominent warning of sorts about gcc 5.1 and AVR-LibC in https://gcc.gnu.org/gcc-5/changes.html:

> If the compiler is used together with AVR-LibC, this requires at least GCC 5.2 and a version of AVR-
> LibC which implements feature #44574.

And, to answer @ladislas question: I don't think we *need* `--with-isl`. It enables some clever loop optimizations, so we may want it eventually, but for the time being, we should just get the recipes back in working order.

closed 2015-05-28 by larsimmisch

ladislas, 2015-05-29:
thanks @larsimmisch for trying this out! I guess we'll have to wait for `gcc 5.2` then ;)

I agree that ISL is not really needed right now. I'm not an expert, but as far as I understand, it is only useful for very complexe applications that need a lot of optimization. So it might not be really useful for avr...

DouglasUrner, 2015-12-24:
For whatever it might be worth, this problem still exists (wrong version of libisl). It would be nice to have slightly clearer instructions on how to use the older version.

DouglasUrner, 2015-12-24:
This is how I modified the avr-gcc formula to use the older ISL:

1. `brew tap homebrew/versions`
1. `brew edit avr-gcc`
1. Change the two instances of 'isl' to 'isl012'
1. `brew reinstall avr-gcc`

ShinDarth, 2016-08-05:
I have the same problem

adrianawise, 2017-07-06:
After I upgraded to gcc 7.1.0, I was still getting the following error:

dyld: Library not loaded: /usr/local/lib/libisl.10.dylib

I just replaced every instance of libisl.10.dylib with libisl.15.dylib, which is the version installed for MacOS Sierra 10.12.5, in the _binary_ files:

/usr/local/Cellar/gfortran/4.8.2/gfortran/libexec/gcc/x86_64-apple-darwin13.0.0/4.8.2/f951
/usr/local/lib/libcloog-isl.4.dylib

This fixed the build for lapack-3.7.1 for me on a MacOS Sierra 10.12.5. I had already built blas-3.7.1. After this, lapackpp-2.5.4 gave me no trouble.

ladislas, 2017-07-07:
@adrianawise we do not support avr-gcc 7 yet and isl has been removed from the formula a long time ago.

are you sure your comment is related to avr?

commit ff9a60ea (2015-05-28): Remove fast moving dependecies libisl and libcloog.

gcc 4.9.2 does compile with libisl v0.14 from upstream homebrew, and gcc 5.1 is
problematic with the avr targets. Remove the dependencies and wait for gcc 5.2

Closes #4