osx-cross/homebrew-avr #6: Could not resolve host: gnu.cardse.net

nixpulvis, 2015-06-08:
Not sure what `gnu.cardse.net` is, but it's not resolving, the mirror does seem to work however.

nixpulvis, 2015-06-08:
The build also fail FWIW.

```
==> ../configure --target=avr --prefix=/usr/local/Cellar/avr-gcc/4.9.2 --enable-languages=c,c++ --with-gnu-as --with-gnu-ld --with-ld=/usr/local/opt/avr-binutils/bin/avr-ld --with-as=/usr/local/opt/avr-bi
==> make
make[4]: *** [_satfractUDQUSQ.o] Error 4
make[3]: *** [multi-do] Error 1
make[2]: *** [all-multi] Error 2
make[1]: *** [all-target-libgcc] Error 2
make: *** [all] Error 2
```

Aaron-Hartwig, 2015-07-20:
I have had the same issue, but now after two tries it is still not working. It just freezes up after the `make`.

```
==> ../configure --target=avr --prefix=/usr/local/Cellar/avr-gcc/4.9.2 --enable-
==> make

```

leojrfs, 2015-07-20:
We should update the formula to 4.9.3 and also make a new one for the 5 series, there has been some nice changes for the avr
https://gcc.gnu.org/gcc-5/changes.html

Aaron-Hartwig, 2015-07-21:
@leojrfs I actually got a successful install. It took ~20 minutes to `make` on the successful run.  To be honest I forgot the process was running and when I looked back it had made progress. So perhaps on my "failed" runs it was really just my impatience (I had only waited ~10 minutes).

leojrfs, 2015-07-21:
@Aaron-Hartwig gcc usually takes a quite long time to compile... :smile: when in doubt use the '-v' option

closed 2015-07-29 by leojrfs

ladislas, 2015-09-04:
this has been fixed with pull request #9