osx-cross/homebrew-avr #7: Stack level too deep on gcc4.8 formula

leojrfs, 2015-07-29:
can anyone confirm this?

```
leo@colossus ~ > brew install avr-gcc48
Error: stack level too deep
Please report this bug:
    https://git.io/brew-troubleshooting
/usr/local/Library/Homebrew/formulary.rb:21
```

smammy, 2015-08-10:
Yes, and I also get this when trying to install `avr-libc48`:
```
sam@shiny ~> brew install avr-libc48
Error: stack level too deep
Please report this bug:
    https://git.io/brew-troubleshooting
/usr/local/Library/Homebrew/formulary.rb:22
```

ladislas, 2015-08-10:
@leojrfs : I think the way we install old versions is not good. It was a hack I found for my personal usage but it's too complicated.

I might be wrong, but homebrew doesn't seem to support multiple versions of the same formulae, say 4.8, 4.9 and 5.2, with the ability to only link the one you want. You can only have the latest versions, which is quite dumb I think...

smammy, 2015-08-10:
This is the full error output: https://gist.github.com/smammy/d81fa2ddcaa31245962b

leojrfs, 2015-12-31:
can anyone confirm this is working?
either way, 4.8 will be left without much attention until the arrival of the first 5 series stable to our repo

ladislas, 2016-09-25:
I don't have issues as of now to install 4.8 and 4.9 -- but #21 should fix the problem :)

closed 2016-10-13 by ladislas