osx-cross/homebrew-avr #10: Download failed: avr-libc-1.8.1.tar.bz2

hryamzik, 2015-10-12:
```bash
brew install avr-libc
```

Results in:

```
...
==> Installing osx-cross/avr/avr-libc
==> Downloading http://download.savannah.gnu.org/releases/avr-libc/avr-libc-1.8.1.tar.bz2
==> Downloading from http://public.p-knowledge.co.jp/Savannah-nongnu-mirror//avr-libc/avr-libc-1.8.1.tar.bz2

curl: (7) Failed to connect to public.p-knowledge.co.jp port 80: Connection refused
Error: Failed to download resource "avr-libc"
Download failed: http://public.p-knowledge.co.jp/Savannah-nongnu-mirror//avr-libc/avr-libc-1.8.1.tar.bz2
```

Are there any other mirrors for this package?

hryamzik, 2015-10-12:
In fact, here's the second [link](http://download.savannah.gnu.org/releases/avr-libc/) from google.

hryamzik, 2015-10-12:
So the workaround is to change the url:

```bash
brew edit avr-libc
```

```ruby
url 'http://mirror.lihnidos.org/GNU/savannah/avr-libc/avr-libc-1.8.1.tar.bz2'
```

leojrfs, 2015-12-31:
this is a external issue.
The ```http://download.savannah.gnu.org/releases/avr-libc/avr-libc-1.8.1.tar.bz2``` URL is resolved by some kind of [savannah](http://savannah.gnu.org)'s mirrorlist.

closed 2015-12-31 by leojrfs