osx-cross/homebrew-avr #13: About the different versions of gcc

ladislas, 2015-11-28:
Hi there!

The formula still has two versions of gcc. Changing from one to the other used to work well just for me but I don't think it's a suitable way to do it.

I thought about this: why not avec different repository for different versions of gcc:

- homebrew-avr - latest gcc, the most up-to-date version (now 4.9 but tomorrow 5.0 and so on)
- homebrew-avr-4.8 - for the 4.8 version
- and so on...

this way people can use which ever version they want. 

what do you guys think?

leojrfs, 2015-11-28:
Thats a solution...
Another one to avoid duplication or extra effort/complication is to chose a "stable" from one of the versions that being the "main" one. The others would be compiled with the version on the binary name, like they do it on the gcc formulas on the homebrew/versions 
Example:
gcc (gcc 4.9.3)
gcc-5 (gcc5.2.0)
gcc-4.8 (gcc4.8.5)

Another solution on the mean time is also keep them on one repo, and  add `conflicts_with` and a message to each formula. [Example](https://github.com/Homebrew/homebrew-versions/commit/c70582a2055ea6649cc1974076f57001f8c471a3)

ladislas, 2015-11-28:
it's not just about conflicting, it's about different things: each time we have to compile `avr-libc` with the right version of `avr-gcc`, you can not just switch as you like.

having them in separate repositories with the conflict message and keg only, except for the main formula.

leojrfs, 2015-11-28:
We can compile and link avr-libc for each version the same way it is done with the gcc naming binaries I think.

but I don't know... we should decide, but first we must definitely chose a stable version (that we support). And then maybe we can move the other versions to their own repos until we can figure out if it is really necessary to have the effort of maintaining multiple versions or not.

What do you think?

ladislas, 2015-11-28:
> We can compile and link avr-libc for each version the same way it is done with the gcc naming binaries I think.

I wanted to do that but I never succeeded... :)

> first we must definitely chose a stable version (that we support). 

:+1: 

> And then maybe we can move the other versions to their own repos until we can figure out if it is really necessary to have the effort of maintaining multiple versions or not.

:+1:

leojrfs, 2015-11-28:
lets maybe focus on gcc5 then? and when we have a working version we move the others.

Sounds good?

What about the rest of you guys? @osx-cross/owners

ladislas, 2015-11-28:
I'm all with using the cutting edge version. But I remember when I started working on my fork at WeAreLeka that people where really surprised I wanted to used a version of gcc newer than the one provided with Arduino ;)

larsimmisch, 2015-11-28:
I'd prefer to have cutting-edge versions of gcc in homebrew-avr. It has a certain risk, but for me - so far - the benefits have outweighed the inconvenience.

ladislas, 2016-10-13:
Hi guys --

"Reopening" this issue as to what we should do.

@cosmo0920 seemed to have found to a solution expressed here: https://github.com/osx-cross/homebrew-avr/pull/21

What should we do?

cosmo0920, 2016-10-13:
Shortly speaking, it need to use "resource syntax" to bundle libc into each of gcc Cellar like #21.

ladislas, 2016-10-13:
thanks a lot @cosmo0920! I'll try that :)

ladislas, 2017-08-04:
fixed with #54

closed 2017-08-04 by ladislas