osx-cross/homebrew-avr #17: Common include path

nixpulvis, 2015-12-20:
Currently there is no good way to reliably link to the libraries in `avr-libc`. I'd like to see something akin to `usr/local/include` for avr, maybe `usr/local/avr/(include | lib)`.

nixpulvis, 2015-12-20:
This would also give people building libraries a place to put headers and libraries. Currently I'm kinda hacking in the libraries with `avr-libc` with homebrew.

```rb
    def install
        system "make PREFIX=#{Formula["avr-libc"].prefix}/avr"
        system "make install PREFIX=#{Formula["avr-libc"].prefix}/avr"
    end
```

leojrfs, 2015-12-31:
Did you take a look at https://github.com/osx-cross/homebrew-avr/issues/13?

luqasz, 2017-01-14:
I had to go to `/usr/local/Cellar/avr-gcc/6.2.0` and do `ln -s /usr/local/Cellar//avr-libc/2.0.0/avr avr`. After that I `avr/io.h: No such file or directory` error disappeared.

luqasz, 2017-05-08:
After upgrading to `avr-gcc` I had to remove all versions of `avr-gcc` and install newset since I had:
```
Compiling: main.c
avr-gcc -c -mmcu=atmega32 -I. -DF_CPU=8000000UL -Os -Wall -Wextra -std=gnu11 /Users/lkostka/workspace/avr/main.c -o .builds/main.o
/Users/lkostka/workspace/avr/main.c:1:20: fatal error: avr/io.h: No such file or directory
 #include <avr/io.h>
```

ladislas, 2017-05-08:
@luqasz yes, upgrading to a newer version of gcc requires to rebuild avr-libc at the same time. Note that you could also unlink the libraries and keep them if you need them for the first version of avf-gcc you installed.

we are trying to change this behavior by including the avr-libc as ressources in the branch `update-formula` - https://github.com/osx-cross/homebrew-avr/tree/update-formula

if you want to experiment and install the latest version of gcc, you can do the following:

```
# cd to the tap
$ cd /usr/local/Homebrew/Library/Taps/osx-cross/homebrew-avr

# checkout the update-formula branch
$ git checkout update-formula

# try to install avr-gcc
$ brew install avr-gcc
```
after that, you'll find all you need in `/usr/local/Cellar/avr-gcc/6.3.0/avr/include`

Let me know if this behavior can be acceptable for you.

luqasz, 2017-05-08:
Thx. I will stick to just removing and reinstalling. I do not switch between versions.

ladislas, 2017-05-08:
you can still try the new branch if you want gcc 6.3 -- it works great for me but I'd love feedback

luqasz, 2017-05-08:
I already have 6.3 installed.

ladislas, 2017-08-04:
as there is no activity, I'm closing the issue. Feel free to reopen if needed

closed 2017-08-04 by ladislas