osx-cross/homebrew-avr #18: avr-gdb does not find ISL

luismarques, 2015-12-27:
```
$ brew install avr-gdb
(...)
checking for version 0.10 of ISL... no
checking for version 0.11 of ISL... no
checking for version 0.12 of ISL... no
configure: error: Unable to find a usable ISL.  See config.log for details.
```

I have `isl` (0.15) and `isl012` (0.12) both installed. I have also tried with only either of them installed, and with none. `brew doctor` is clean. I tried to do an `xcode-select --install` (saw that in a similar bug) but that didn't help either.

(I have `avr-binutils`, `avr-gcc`, `avr-libc` and `avrdude` all installed without issue)

leojrfs, 2015-12-31:
@osx-cross/team I can confirm this

leojrfs, 2015-12-31:
related to https://github.com/osx-cross/homebrew-avr/pull/3 that was added by @jpommerening

cosmo0920, 2016-01-24:
This is because `"--with-isl=#{Formula["isl"].opt_prefix}"` tends to refer to Homebrew mainline ISL.
For example, we should use homebrew-versions' ISL 0.12 with `"--with-isl=#{Formula["isl012"].opt_prefix}"`.

closed 2016-10-13 by ladislas