osx-cross/homebrew-avr #19: avr-libc could not install with `--sandbox` option

cosmo0920, 2016-01-23:
Here is log:

```log
% brew install --sandbox avr-libc
==> Installing avr-libc from osx-cross/avr
==> Installing dependencies for osx-cross/avr/avr-libc: avr-binutils, avr
==> Installing osx-cross/avr/avr-libc dependency: avr-binutils
==> Using the sandbox
==> Downloading http://ftpmirror.gnu.org/binutils/binutils-2.25.tar.gz
==> Downloading from http://mirror.jre655.com/GNU/binutils/binutils-2.25.tar.gz
######################################################################## 100.0%
==> Downloading https://gist.github.com/larsimmisch/4190960/raw/b36f3d6d08698000
==> Downloading from https://gist.githubusercontent.com/larsimmisch/4190960/raw/
######################################################################## 100.0%
==> Patching
patching file binutils/size.c
Hunk #1 succeeded at 34 (offset -2 lines).
Hunk #2 succeeded at 67 (offset -2 lines).
Hunk #3 succeeded at 82 (offset -2 lines).
Hunk #4 succeeded at 335 (offset -2 lines).
Hunk #5 succeeded at 335 (offset -2 lines).
Hunk #6 succeeded at 357 (offset -2 lines).
Hunk #7 succeeded at 366 (offset -2 lines).
Hunk #8 succeeded at 397 (offset -2 lines).
Hunk #9 succeeded at 407 (offset -2 lines).
Hunk #10 succeeded at 422 (offset -2 lines).
Hunk #11 succeeded at 455 (offset -2 lines).
Hunk #12 succeeded at 507 (offset -2 lines).
Hunk #13 succeeded at 867 (offset -2 lines).
==> ../configure --disable-debug --disable-dependency-tracking --target=avr --pr
==> make
==> make install
🍺  /usr/local/Cellar/avr-binutils/2.25: 138 files, 19.0M, built in 2 minutes 36 seconds
==> Installing osx-cross/avr/avr-libc dependency: avr-gcc
==> Downloading ftp://gcc.gnu.org/pub/gcc/releases/gcc-4.9.3/gcc-4.9.3.tar.bz2
######################################################################## 100.0%
==> ../configure --target=avr --prefix=/usr/local/Cellar/avr-gcc/4.9.3 --enable-
==> make
==> make install
🍺  /usr/local/Cellar/avr-gcc/4.9.3: 285 files, 509.0M, built in 46 minutes 25 seconds
==> Installing osx-cross/avr/avr-libc
==> Downloading http://download.savannah.gnu.org/releases/avr-libc/avr-libc-1.8.
==> Downloading from http://public.p-knowledge.co.jp/Savannah-nongnu-mirror//avr
######################################################################## 100.0%
==> ./configure --build=x86_64-apple-darwin15.3.0 --prefix=/usr/local/Cellar/avr
==> make install
==> copying /usr/local/Cellar/avr-libc/1.8.1/avr -> /usr/local/Cellar/avr-gcc/4.
Error: Operation not permitted - /usr/local/Cellar/avr-gcc/4.9.3/avr
```

To solve this issue, it should bundle avr-libc into avr-gcc formula with resource syntax.

larsimmisch, 2016-01-23:
Hmmm, care to send a pull request or elaborate?

cosmo0920, 2016-01-24:
Sure, but ordinary homebrew users do not affect by this issue.
`--sandbox` option prohibits to write outside of the formula's prefix and some temporary directories.
This option is mainly used by homebrew mainline contributors and its CI.

cosmo0920, 2016-01-24:
I've registered PR which solves this issue, thanks!

nooges, 2017-08-03:
This issue needs to be looked at again, because with this commit: https://github.com/Homebrew/brew/commit/fb310c57b18c3835ed69e25e653cddc3570f64ac, sandboxing is turned on for non-homebrew taps.

Installation of avr-libc now results in this:
```
==> make install
==> copying /usr/local/Cellar/avr-libc/2.0.0/avr -> /usr/local/opt/avr-gcc
Error: Operation not permitted - /usr/local/opt/avr-gcc/avr
```

Related issues:
https://github.com/Homebrew/brew/issues/2994
https://github.com/Homebrew/brew/issues/2986

cosmo0920, 2017-08-03:
Does anyone enough motivation and time to make up-to-date this PR? https://github.com/osx-cross/homebrew-avr/pull/21
It should fix this issue.

nooges, 2017-08-03:
Actually, it looks like this PR should take care of the issue: https://github.com/osx-cross/homebrew-avr/pull/54

cosmo0920, 2017-08-04:
Thanks for notifying it.
#54 should solve this issue.

ladislas, 2017-08-04:
@cosmo0920 it should be fixed now, I just merged #54 :)

closed 2017-08-04 by ladislas

cosmo0920, 2017-08-04:
Awesome. Thank you!