osx-cross/homebrew-avr #27: Can't Install

chris1111111, 2016-03-09:
When I run the install into terminal I have the following problem

I can run brew tap osx-cross/avr with no issues 
the second step of running the following is were an error occurs 

brew install avr-libc

Last 15 lines from /Users/chris/Library/Logs/Homebrew/avr-binutils/02.make:
looking for an incomplete stdint.h in stdint.h, checking for uint32_t... no
looking for an incomplete stdint.h in inttypes.h, checking for uint32_t... no
looking for an incomplete stdint.h in sys/inttypes.h, checking for uint32_t... no
looking for u_intXX_t types in sys/types.h, checking for u_int32_t... no
looking for u_intXX_t types in inttypes.h, checking for u_int32_t... no
looking for u_intXX_t types in sys/inttypes.h, checking for u_int32_t... no
checking what to include in bfd_stdint.h... stddef.h (using manual detection)
checking size of void *... (cached) 0
checking size of long... (cached) 0
checking size of int... 0
checking size of short... 0
checking size of char... 0
checking for type equivalent to int8_t... configure: error: no 8-bit type, please report a bug
make[1]: *** [configure-bfd] Error 1
make: *** [all] Error 2

josherick, 2016-03-24:
Exact same issue here

josherick, 2016-03-24:
Running `xcode-select --install` fixed for me. Maybe something was out of date.

cesareolivera, 2016-04-02:
I have this problem after i updated to os x 10.11.4. Please help me

CesarOlivera$ avr-gcc -Wall -Wextra -Wpedantic -std=gnu99 -mmcu=atmega328p -Os -o go_avr.elf push_debouncer_AVR.c

dyld: Library not loaded: /usr/local/lib/libmpfr.4.dylib
  Referenced from: /usr/local/opt/libmpc/lib/libmpc.3.dylib
  Reason: Incompatible library version: libmpc.3.dylib requires version 6.0.0 or later, but libmpfr.4.dylib provides version 5.0.0
avr-gcc: internal compiler error: Trace/BPT trap: 5 (program cc1)
libbacktrace could not find executable to open
Please submit a full bug report,
with preprocessed source if appropriate.
See <http://gcc.gnu.org/bugs.html> for instructions.

evangs, 2016-06-04:
josherick's fix worked for me as well

leojrfs, 2016-07-29:
run `xcode-select --install`

closed 2016-07-29 by leojrfs

AlphaOrionis42, 2017-08-03:
I am having issues with my install as well. I have the Xcode command line tools installed already. When trying to install the running brew install avr-libc I end up with the following.
`Gregs-MacBook-Pro:~ gregt$ brew install avr-libc
==> Installing avr-libc from osx-cross/avr
==> Installing dependencies for osx-cross/avr/avr-libc: avr-gcc
==> Installing osx-cross/avr/avr-libc dependency: avr-gcc
==> Using the sandbox
==> Downloading https://gcc.gnu.org/pub/gcc/releases/gcc-7.1.0/gcc-7.1.0.tar.bz2
Already downloaded: /Users/gregt/Library/Caches/Homebrew/avr-gcc-7.1.0.tar.bz2
==> ../configure --target=avr --prefix=/usr/local/Cellar/avr-gcc/7.1.0 --enable-
==> make
==> make install
🍺  /usr/local/Cellar/avr-gcc/7.1.0: 761 files, 621.5MB, built in 16 minutes 40 seconds
==> Installing osx-cross/avr/avr-libc
==> Downloading https://download.savannah.gnu.org/releases/avr-libc/avr-libc-2.0
==> Downloading from http://savannah.spinellicreations.com/avr-libc/avr-libc-2.0
######################################################################## 100.0%
==> ./configure --build=i686-apple-darwin16.6.0 --prefix=/usr/local/Cellar/avr-l
==> make install
==> copying /usr/local/Cellar/avr-libc/2.0.0/avr -> /usr/local/opt/avr-gcc
Error: Operation not permitted - /usr/local/opt/avr-gcc/avr`

I am not sure why the copy operation is not being permitted.