osx-cross/homebrew-avr #29: Issues with installation on 10.11.5

edasque, 2016-07-08:
On a brand new installation of 10.11.15, I can't seem to install avr-gcc:
```
brew install avr-libc
==> Installing avr-libc from osx-cross/avr
==> Installing dependencies for osx-cross/avr/avr-libc: avr-gcc
==> Installing osx-cross/avr/avr-libc dependency: avr-gcc
==> Downloading ftp://gcc.gnu.org/pub/gcc/releases/gcc-4.9.3/gcc-4.9.3.tar.bz2
Already downloaded: /Users/edasque/Library/Caches/Homebrew/avr-gcc-4.9.3.tar.bz2
==> ../configure --target=avr --prefix=/usr/local/Cellar/avr-gcc/4.9.3 --enable-languages=c,c++ --with-gnu-as --with-g
==> make
Last 15 lines from /Users/edasque/Library/Logs/Homebrew/avr-gcc/02.make:
make[2]: *** [s-opinit] Error 1
make[2]: *** Waiting for unfinished jobs....
/bin/sh ../../gcc/../move-if-change tmp-attr-common.h insn-attr-common.h
echo timestamp > s-codes
echo timestamp > s-flags
/bin/sh ../../gcc/../move-if-change tmp-attr.h insn-attr.h
echo timestamp > s-attr-common
/bin/sh ../../gcc/../move-if-change tmp-config.h insn-config.h
echo timestamp > s-attr
echo timestamp > s-config
/bin/sh ../../gcc/../move-if-change tmp-mlib.h multilib.h
echo timestamp > s-mlib
rm gcc.pod
make[1]: *** [all-gcc] Error 2
make: *** [all] Error 2

READ THIS: https://git.io/brew-troubleshooting
If reporting this issue please do so at (not Homebrew/brew):
  https://github.com/osx-cross/homebrew-avr/issues

These open issues may also help:
Bundle avr-libc to avr-gcc to resolve sandbox issue https://github.com/osx-cross/homebrew-avr/pull/21
```

Any idea how I could fix it? I tried to install 48 with similar issues. It doesn't work on this work mac but works on my home Mac for some reason.

```
brew --version
Homebrew 0.9.9 (git revision 8282; last commit 2016-07-08)
Homebrew/homebrew-core (git revision 2302; last commit 2016-07-08)
```

```
gcc --version
Configured with: --prefix=/Applications/Xcode.app/Contents/Developer/usr --with-gxx-include-dir=/usr/include/c++/4.2.1
Apple LLVM version 7.3.0 (clang-703.0.31)
Target: x86_64-apple-darwin15.5.0
Thread model: posix
InstalledDir: /Applications/Xcode.app/Contents/Developer/Toolchains/XcodeDefault.xctoolchain/usr/bin
```

leojrfs, 2016-07-29:
You still have this problem?
its working for me,  do you have xcode command line tools installed? 

please install it with `--verbose` option to see the compiler output

closed 2016-07-31 by leojrfs

edasque, 2016-07-31:
I still have that problem on my work computer, I can give you the dump Monday.

leojrfs, 2016-07-31:
Also check if you have the xcode command line tools

edasque, 2016-07-31:
I sure did

edasque, 2016-08-01:
Now on MacOS 10.11.6

```bash
➜  ~ xcode-select --install
xcode-select: error: command line tools are already installed,
use "Software Update" to install updates
```

```bash
➜  ~ gcc --version
Configured with: --prefix=/Applications/Xcode.app/Contents/Developer/usr --with-gxx-include-dir=/usr/include/c++/4.2.1
Apple LLVM version 7.3.0 (clang-703.0.31)
Target: x86_64-apple-darwin15.6.0
Thread model: posix
InstalledDir: /Applications/Xcode.app/Contents/Developer/Toolchains/XcodeDefault.xctoolchain/usr/bin
```

I run with verbose and capture what I thought was the interesting part:
```
checking for fcntl.h... echo "#define LOCALEDIR \"/usr/local/Cellar/avr-gcc/4.9.3/share/locale\"" > localedir.new
../../libcpp/../move-if-change localedir.new localedir.h
make[2]: ../../libcpp/../move-if-change: Permission denied
make[2]: *** [localedir.hs] Error 1
make[2]: *** Waiting for unfinished jobs....
yes
checking for unistd.h... (cached) yes
checking for sys/file.h... yes
checking for sys/time.h... yes
checking for sys/mman.h... yes
checking for sys/resource.h... yes
checking for sys/param.h... yes
checking for sys/times.h... yes
checking for sys/stat.h... (cached) yes
checking for direct.h... make[1]: *** [all-libcpp] Error 2
make[1]: *** Waiting for unfinished jobs....
no
checking for malloc.h... no
checking for langinfo.h... yes
checking for ldfcn.h... no
checking for locale.h... yes
checking for wchar.h... yes
checking for thread.h... no
checking for pthread.h... yes
checking for CHAR_BIT... yes
checking whether byte ordering is bigendian... no
checking how to run the C++ preprocessor... clang++ -E
checking for unordered_map... yes
checking for tr1/unordered_map... no
checking for ext/hash_map... yes
checking dependency style of clang++... gcc3
checking for collect2 libraries... none required
checking for library containing exc_resume... no
checking for library containing kstat_open... no
checking for library containing ldexp... none required
checking for inttypes.h... yes
checking for times... yes
checking for clock... yes
checking for kill... yes
checking for getrlimit... yes
checking for setrlimit... yes
checking for atoll... yes
checking for atoq... no
checking for sysconf... yes
checking for strsignal... yes
checking for getrusage... yes
checking for nl_langinfo... yes
checking for gettimeofday... yes
checking for mbstowcs... yes
checking for wcswidth... yes
checking for mmap... yes
[... 290 lines cut ...]
Xcode: 7.3.1
CLT: 7.3.1.0.1.1461711523
X11: N/A
==> ENV
HOMEBREW_CC: clang
HOMEBREW_CXX: clang++
SDKROOT: /Applications/Xcode.app/Contents/Developer/Platforms/MacOSX.platform/Developer/SDKs/MacOSX10.11.sdk
MAKEFLAGS: -j8
CMAKE_PREFIX_PATH: /usr/local
CMAKE_INCLUDE_PATH: /usr/include/libxml2:/System/Library/Frameworks/OpenGL.framework/Versions/Current/Headers
CMAKE_LIBRARY_PATH: /System/Library/Frameworks/OpenGL.framework/Versions/Current/Libraries
MACOSX_DEPLOYMENT_TARGET: 10.11
PKG_CONFIG_LIBDIR: /usr/lib/pkgconfig:/usr/local/Library/Homebrew/os/mac/pkgconfig/10.11
PATH: /usr/local/Library/Homebrew/shims/super:/usr/local/opt/avr-binutils/bin:/usr/bin:/bin:/usr/sbin:/sbin

Error: osx-cross/avr/avr-gcc 4.9.3 did not build
Logs:
     /Users/edasque/Library/Logs/Homebrew/avr-gcc/01.configure
     /Users/edasque/Library/Logs/Homebrew/avr-gcc/01.configure.cc
     /Users/edasque/Library/Logs/Homebrew/avr-gcc/02.make
     /Users/edasque/Library/Logs/Homebrew/avr-gcc/02.make.cc

These open issues may also help:
Bundle avr-libc to avr-gcc to resolve sandbox issue https://github.com/osx-cross/homebrew-avr/pull/21
```

edasque, 2016-08-01:
Seems the first error is here:
```
build/genattrtab ../../gcc/config/avr/avr.md insn-conditions.md \
		-Atmp-attrtab.c -Dtmp-dfatab.c -Ltmp-latencytab.c
make[2]: build/genattrtab: Permission denied
/bin/sh ../../gcc/../move-if-change tmp-config.h insn-config.h
build/genemit ../../gcc/config/avr/avr.md \
	  insn-conditions.md > tmp-emit.c
make[2]: *** [s-attrtab] Error 1
make[2]: *** Waiting for unfinished jobs....
```