osx-cross/homebrew-avr #32: Support for ATtiny441/841

sgoll, 2016-08-02:
It seems the latest version (`avr-gcc -v` says 4.9.3) has no support for ATtiny441 and the related ATtiny841, installed with default options (`brew install avr-libc`).

Am I missing something?

ladislas, 2016-10-13:
I don't use those, so i've never tried. 

would you have a github repository with code and Makefile I could use to try it myself?

sgoll, 2016-10-13:
@ladislas Not necessary, simply check if `avr-gcc -mmcu=attiny841` fails with `cannot access device-specs for (…)` (in which case it should also output the list of supported devices).

On my Debian system, this is the expected output:

```
$ avr-gcc -mmcu=attiny841
avr-gcc: fatal error: no input files
compilation terminated.
```

My compiler there is also older than the one provided by Homebrew, so it should definitely be able to support those devices:

```
$ avr-gcc -v
Using built-in specs.
Reading specs from /usr/lib/gcc/avr/4.9.2/device-specs/specs-avr2
COLLECT_GCC=avr-gcc
COLLECT_LTO_WRAPPER=/usr/lib/gcc/avr/4.9.2/lto-wrapper
Target: avr
Configured with: ../src/configure -v --enable-languages=c,c++ --prefix=/usr/lib --infodir=/usr/share/info --mandir=/usr/share/man --bindir=/usr/bin --libexecdir=/usr/lib --libdir=/usr/lib --enable-shared --with-system-zlib --enable-long-long --enable-nls --without-included-gettext --disable-libssp --build=x86_64-linux-gnu --host=x86_64-linux-gnu --target=avr CFLAGS='-g -O2 -fstack-protector-strong -Wformat ' CPPFLAGS=-D_FORTIFY_SOURCE=2 CXXFLAGS='-g -O2 -fstack-protector-strong -Wformat ' FCFLAGS='-g -O2 -fstack-protector-strong' FFLAGS='-g -O2 -fstack-protector-strong' GCJFLAGS='-g -O2 -fstack-protector-strong' LDFLAGS=-Wl,-z,relro OBJCFLAGS='-g -O2 -fstack-protector-strong -Wformat ' OBJCXXFLAGS='-g -O2 -fstack-protector-strong -Wformat '
Thread model: single
gcc version 4.9.2 (GCC) 
```

My guess is that there is some kind of configure option when building `avr-gcc`? The official documentation simply includes the two MCU types `attiny441` and `attiny841` in their list:

https://gcc.gnu.org/onlinedocs/gcc/AVR-Options.html

ladislas, 2016-10-14:
well, with the latest version it works well with the latest version. Have you tried removing everything and starting again?

[image]

sgoll, 2016-10-14:
Yes, now it works. And like you I now have avr-gcc version 6.2.0.

closed 2016-10-14 by sgoll

ladislas, 2016-10-14:
glad I could help :)