osx-cross/homebrew-avr #39: configure: error: cannot compute suffix of object files: cannot compile

randre03, 2016-12-24:
Here is the error when I try to install via Homebrew:

`Last 15 lines from /Users/Library/Logs/Homebrew/avr-gcc/02.make:
checking for --enable-version-specific-runtime-libs... no
checking for a BSD-compatible install... /usr/bin/install -c
checking for gawk... awk
checking for avr-ar... avr-ar
checking for avr-lipo... avr-lipo
checking for avr-nm... /private/tmp/avr-gcc-20161224-889-xdcwwo/gcc-6.2.0/build/./gcc/nm
checking for avr-ranlib... avr-ranlib
checking for avr-strip... avr-strip
checking whether ln -s works... yes
checking for avr-gcc... /private/tmp/avr-gcc-20161224-889-xdcwwo/gcc-6.2.0/build/./gcc/xgcc -B/private/tmp/avr-gcc-20161224-889-xdcwwo/gcc-6.2.0/build/./gcc/ -B/usr/local/Cellar/avr-gcc/6.2.0/avr/bin/ -B/usr/local/Cellar/avr-gcc/6.2.0/avr/lib/ -isystem /usr/local/Cellar/avr-gcc/6.2.0/avr/include -isystem /usr/local/Cellar/avr-gcc/6.2.0/avr/sys-include  -mmcu=avrtiny
checking for suffix of object files... configure: error: in `/private/tmp/avr-gcc-20161224-889-xdcwwo/gcc-6.2.0/build/avr/avrtiny/libgcc':
configure: error: cannot compute suffix of object files: cannot compile
See `config.log' for more details.
make[1]: *** [configure-target-libgcc] Error 1
make: *** [all] Error 2`

ladislas, 2016-12-26:
try removing everything and reinstalling from scratch following the instructions here: https://github.com/osx-cross/homebrew-avr#installing-homebrew-avr-formulae

ladislas, 2017-08-04:
as there is no activity, I'm closing the issue. Feel free to reopen if needed

closed 2017-08-04 by ladislas