osx-cross/homebrew-avr #44: fatal error: 'libelf.h' file not found #include <libelf.h>

wenhuizhang, 2017-04-18:
```
wenhui-2:sim WenhuiZhang$ brew link gcc@4.8
Linking /usr/local/Cellar/gcc@4.8/4.8.4... 25 symlinks created
wenhui-2:sim WenhuiZhang$ brew install --HEAD osx-cross/avr/simavr
==> Installing simavr from osx-cross/avr
==> Cloning https://github.com/buserror/simavr.git
Updating /Users/WenhuiZhang/Library/Caches/Homebrew/simavr--git
==> Checking out branch master
==> make install DESTDIR=/usr/local/Cellar/simavr/HEAD-1ae1912 HOMEBREW_PREFIX=/usr/local RELEASE=1
Last 15 lines from /Users/WenhuiZhang/Library/Logs/Homebrew/simavr/01.make:
/Applications/Xcode.app/Contents/Developer/usr/bin/make libsimavr run_avr
CC sim/sim_cmds.c
CC sim/sim_cycle_timers.c
CC sim/sim_avr.c
sim/sim_elf.c:32:10: fatal error: 'libelf.h' file not found
#include <libelf.h>
         ^
1 error generated.
make[2]: *** [obj-x86_64-apple-darwin16.4.0/sim_elf.o] Error 1
make[2]: *** Waiting for unfinished jobs....
CC sim/sim_hex.c
CC sim/sim_core.c
CC sim/sim_gdb.c
make[1]: *** [all] Error 2
make: *** [install] Error 2
```

# Please always follow these steps:
- [ ] Confirmed this is a problem with Homebrew/brew and not one or more formulae? If it's a formulae-specific problem please file this issue at https://github.com/Homebrew/homebrew-core/issues/new
- [ ] Ran `brew update` and retried your prior step?
- [ ] Ran `brew doctor`, fixed as many issues as possible and retried your prior step?
- [ ] Ran `brew config` and `brew doctor` and included their output with your issue?

To help us debug your issue please explain:
- What you were trying to do (and why)
- What happened (include command output)
- What you expected to happen
- Step-by-step reproduction instructions (by running `brew` commands)

**Please note we may immediately close your issue without comment if you do not fill out the issue template and provide ALL the requested information.**

# Or propose a feature:
Please replace this section with a detailed description of your proposed feature, the motivation for it and alternatives considered.
Please note we may close this issue or ask you to create a pull-request if it's something we're not actively planning to work on.

ladislas, 2017-04-18:
my guess is that your need to use avr-gcc instead of gcc.

wenhuizhang, 2017-04-19:
Thanks， just updated and used avr-gcc, then I have got a linker problem, any insights please?
```
client-130-203-90-97:simavr--git WenhuiZhang$ sudo make -v
GNU Make 4.0
Built for i386-apple-darwin13.0.0
Copyright (C) 1988-2013 Free Software Foundation, Inc.
License GPLv3+: GNU GPL version 3 or later <http://gnu.org/licenses/gpl.html>
This is free software: you are free to change and redistribute it.
There is NO WARRANTY, to the extent permitted by law.
client-130-203-90-97:simavr--git WenhuiZhang$ vim tests/test_atmega48_disabled_timer.c
client-130-203-90-97:simavr--git WenhuiZhang$ make
Makefile:30: warning: overriding recipe for target 'install'
Makefile:24: warning: ignoring old recipe for target 'install'
make -C simavr RELEASE=0
make[1]: Entering directory '/Users/WenhuiZhang/Library/Caches/Homebrew/simavr--git/simavr'
Makefile:30: warning: overriding recipe for target 'install'
Makefile:24: warning: ignoring old recipe for target 'install'
make -C simavr RELEASE=0
make[2]: Entering directory '/Users/WenhuiZhang/Library/Caches/Homebrew/simavr--git/simavr/simavr'
make obj config
make[3]: Entering directory '/Users/WenhuiZhang/Library/Caches/Homebrew/simavr--git/simavr/simavr'
make[3]: Nothing to be done for 'obj'.
make[3]: Nothing to be done for 'config'.
make[3]: Leaving directory '/Users/WenhuiZhang/Library/Caches/Homebrew/simavr--git/simavr/simavr'
make libsimavr run_avr
make[3]: Entering directory '/Users/WenhuiZhang/Library/Caches/Homebrew/simavr--git/simavr/simavr'
make[3]: Nothing to be done for 'libsimavr'.
make[3]: 'run_avr' is up to date.
make[3]: Leaving directory '/Users/WenhuiZhang/Library/Caches/Homebrew/simavr--git/simavr/simavr'
make[2]: Leaving directory '/Users/WenhuiZhang/Library/Caches/Homebrew/simavr--git/simavr/simavr'
make -C tests RELEASE=0
make[2]: Entering directory '/Users/WenhuiZhang/Library/Caches/Homebrew/simavr--git/simavr/tests'
make[2]: Nothing to be done for 'all'.
make[2]: Leaving directory '/Users/WenhuiZhang/Library/Caches/Homebrew/simavr--git/simavr/tests'
make -C examples RELEASE=0
make[2]: Entering directory '/Users/WenhuiZhang/Library/Caches/Homebrew/simavr--git/simavr/examples'
uname: illegal option -- o
usage: uname [-amnprsv]
for bi in board_ledramp board_i2ctest board_simduino board_ds1338 board_hd77480 board_timer_64led board_ssd1306; do make -C $bi; done
make[3]: Entering directory '/Users/WenhuiZhang/Library/Caches/Homebrew/simavr--git/simavr/examples/board_ledramp'
ledramp done
make[3]: Leaving directory '/Users/WenhuiZhang/Library/Caches/Homebrew/simavr--git/simavr/examples/board_ledramp'
make[3]: Entering directory '/Users/WenhuiZhang/Library/Caches/Homebrew/simavr--git/simavr/examples/board_i2ctest'
i2ctest done
make[3]: Leaving directory '/Users/WenhuiZhang/Library/Caches/Homebrew/simavr--git/simavr/examples/board_i2ctest'
make[3]: Entering directory '/Users/WenhuiZhang/Library/Caches/Homebrew/simavr--git/simavr/examples/board_simduino'
simduino done
make[3]: Leaving directory '/Users/WenhuiZhang/Library/Caches/Homebrew/simavr--git/simavr/examples/board_simduino'
make[3]: Entering directory '/Users/WenhuiZhang/Library/Caches/Homebrew/simavr--git/simavr/examples/board_ds1338'
ds1338demo done
make[3]: Leaving directory '/Users/WenhuiZhang/Library/Caches/Homebrew/simavr--git/simavr/examples/board_ds1338'
make[3]: Entering directory '/Users/WenhuiZhang/Library/Caches/Homebrew/simavr--git/simavr/examples/board_hd77480'
charlcd done
make[3]: Leaving directory '/Users/WenhuiZhang/Library/Caches/Homebrew/simavr--git/simavr/examples/board_hd77480'
make[3]: Entering directory '/Users/WenhuiZhang/Library/Caches/Homebrew/simavr--git/simavr/examples/board_timer_64led'
timer_64led done
make[3]: Leaving directory '/Users/WenhuiZhang/Library/Caches/Homebrew/simavr--git/simavr/examples/board_timer_64led'
make[3]: Entering directory '/Users/WenhuiZhang/Library/Caches/Homebrew/simavr--git/simavr/examples/board_ssd1306'
ssd1306demo done
make[3]: Leaving directory '/Users/WenhuiZhang/Library/Caches/Homebrew/simavr--git/simavr/examples/board_ssd1306'
make[2]: Leaving directory '/Users/WenhuiZhang/Library/Caches/Homebrew/simavr--git/simavr/examples'
make[1]: Leaving directory '/Users/WenhuiZhang/Library/Caches/Homebrew/simavr--git/simavr'
make -C tests RELEASE=0
make[1]: Entering directory '/Users/WenhuiZhang/Library/Caches/Homebrew/simavr--git/tests'
TST obj-x86_64-apple-darwin16.4.0/test_atmega48_disabled_timer.tst
In file included from tests.c:6:
/usr/local/CrossPack-AVR-20131216/avr/include/stdio.h:724:12: warning: declaration of built-in function 'fprintf' requires inclusion of the header <stdio.h>
      [-Wbuiltin-requires-header]
extern int      fprintf(FILE *__stream, const char *__fmt, ...);
                ^
/usr/local/CrossPack-AVR-20131216/avr/include/stdio.h:875:12: warning: declaration of built-in function 'vfscanf' requires inclusion of the header <stdio.h>
      [-Wbuiltin-requires-header]
extern int      vfscanf(FILE *__stream, const char *__fmt, va_list __ap);
                ^
/usr/local/CrossPack-AVR-20131216/avr/include/stdio.h:888:12: warning: declaration of built-in function 'fscanf' requires inclusion of the header <stdio.h>
      [-Wbuiltin-requires-header]
extern int      fscanf(FILE *__stream, const char *__fmt, ...);
                ^
tests.c:45:2: warning: implicit declaration of function 'atexit' is invalid in C99 [-Wimplicit-function-declaration]
        atexit(atexit_handler);
        ^
[... cut ...]
5 warnings generated.
ld: library not found for -lsimavr
clang: error: linker command failed with exit code 1 (use -v to see invocation)
Makefile:46: recipe for target 'obj-x86_64-apple-darwin16.4.0/test_atmega48_disabled_timer.tst' failed
make[1]: *** [obj-x86_64-apple-darwin16.4.0/test_atmega48_disabled_timer.tst] Error 1
make[1]: Leaving directory '/Users/WenhuiZhang/Library/Caches/Homebrew/simavr--git/tests'
Makefile:18: recipe for target 'build-tests' failed
make: *** [build-tests] Error 2

```

ladislas, 2017-04-19:
@obra any ideas on that? I don't personally use it, so I cannot help :(

obra, 2017-04-19:
I never really got simavr working and am on Linux these days :/

ladislas, 2017-08-04:
as there is no activity, I'm closing the issue. Feel free to reopen if needed

closed 2017-08-04 by ladislas