osx-cross/homebrew-avr #45: simavr

wenhuizhang, 2017-04-18:
client-130-203-90-97:simavr--git WenhuiZhang$ make 
Makefile:30: warning: overriding recipe for target 'install'
Makefile:24: warning: ignoring old recipe for target 'install'
make -C simavr RELEASE=0
make[1]: Entering directory '/Users/WenhuiZhang/Library/Caches/Homebrew/simavr--git/simavr'
Makefile:30: warning: overriding recipe for target 'install'
Makefile:24: warning: ignoring old recipe for target 'install'
make -C simavr RELEASE=0
make[2]: Entering directory '/Users/WenhuiZhang/Library/Caches/Homebrew/simavr--git/simavr/simavr'
CONF obj-x86_64-apple-darwin16.4.0/cores.deps
WARNING cores/sim_90usb162.c did not compile, check your avr-gcc toolchain
WARNING cores/sim_mega128.c did not compile, check your avr-gcc toolchain
WARNING cores/sim_mega1280.c did not compile, check your avr-gcc toolchain
WARNING cores/sim_mega1281.c did not compile, check your avr-gcc toolchain
WARNING cores/sim_mega1284.c did not compile, check your avr-gcc toolchain
WARNING cores/sim_mega128rfa1.c did not compile, check your avr-gcc toolchain
WARNING cores/sim_mega128rfr2.c did not compile, check your avr-gcc toolchain
WARNING cores/sim_mega16.c did not compile, check your avr-gcc toolchain
WARNING cores/sim_mega164.c did not compile, check your avr-gcc toolchain
WARNING cores/sim_mega168.c did not compile, check your avr-gcc toolchain
WARNING cores/sim_mega169.c did not compile, check your avr-gcc toolchain
WARNING cores/sim_mega16m1.c did not compile, check your avr-gcc toolchain
WARNING cores/sim_mega2560.c did not compile, check your avr-gcc toolchain
WARNING cores/sim_mega32.c did not compile, check your avr-gcc toolchain
WARNING cores/sim_mega324.c did not compile, check your avr-gcc toolchain
WARNING cores/sim_mega324a.c did not compile, check your avr-gcc toolchain
WARNING cores/sim_mega328.c did not compile, check your avr-gcc toolchain
WARNING cores/sim_mega48.c did not compile, check your avr-gcc toolchain
WARNING cores/sim_mega644.c did not compile, check your avr-gcc toolchain
WARNING cores/sim_mega8.c did not compile, check your avr-gcc toolchain
WARNING cores/sim_mega88.c did not compile, check your avr-gcc toolchain
WARNING cores/sim_tiny13.c did not compile, check your avr-gcc toolchain
WARNING cores/sim_tiny2313.c did not compile, check your avr-gcc toolchain
WARNING cores/sim_tiny24.c did not compile, check your avr-gcc toolchain
WARNING cores/sim_tiny25.c did not compile, check your avr-gcc toolchain
WARNING cores/sim_tiny44.c did not compile, check your avr-gcc toolchain
WARNING cores/sim_tiny45.c did not compile, check your avr-gcc toolchain
WARNING cores/sim_tiny84.c did not compile, check your avr-gcc toolchain
WARNING cores/sim_tiny85.c did not compile, check your avr-gcc toolchain
make obj config
make[3]: Entering directory '/Users/WenhuiZhang/Library/Caches/Homebrew/simavr--git/simavr/simavr'
make[3]: Nothing to be done for 'obj'.
CONF sim_core_decl.h
make[3]: Leaving directory '/Users/WenhuiZhang/Library/Caches/Homebrew/simavr--git/simavr/simavr'
make libsimavr run_avr
make[3]: Entering directory '/Users/WenhuiZhang/Library/Caches/Homebrew/simavr--git/simavr/simavr'
CC sim/sim_vcd_file.c
CC sim/sim_hex.c
CC sim/sim_avr.c
CC sim/sim_io.c
CC sim/sim_elf.c
CC sim/sim_irq.c
CC sim/sim_cmds.c
CC sim/sim_core.c
CC sim/sim_cycle_timers.c
CC sim/sim_gdb.c
CC sim/sim_utils.c
CC sim/sim_interrupts.c
CC sim/avr_watchdog.c
CC sim/avr_bitbang.c
CC sim/avr_timer.c
CC sim/avr_eeprom.c
CC sim/avr_extint.c
CC sim/avr_uart.c
CC sim/avr_ioport.c
CC sim/avr_usb.c
CC sim/avr_spi.c
CC sim/avr_adc.c
CC sim/avr_twi.c
CC sim/avr_flash.c
CC sim/avr_lin.c
In file included from cores/sim_megax.c:22:
In file included from sim/sim_avr.h:29:
sim/sim_irq.h:25:10: fatal error: 'stdint.h' file not found
#include <stdint.h>
         ^
1 error generated.
../Makefile.common:176: recipe for target 'obj-x86_64-apple-darwin16.4.0/sim_megax.o' failed
make[3]: *** [obj-x86_64-apple-darwin16.4.0/sim_megax.o] Error 1
make[3]: Leaving directory '/Users/WenhuiZhang/Library/Caches/Homebrew/simavr--git/simavr/simavr'
Makefile:34: recipe for target 'all' failed
make[2]: *** [all] Error 2
make[2]: Leaving directory '/Users/WenhuiZhang/Library/Caches/Homebrew/simavr--git/simavr/simavr'
Makefile:15: recipe for target 'build-simavr' failed
make[1]: *** [build-simavr] Error 2
make[1]: Leaving directory '/Users/WenhuiZhang/Library/Caches/Homebrew/simavr--git/simavr'
Makefile:15: recipe for target 'build-simavr' failed
make: *** [build-simavr] Error 2

ladislas, 2017-04-18:
same as #44 -- you need to install avr-gcc

wenhuizhang, 2017-04-19:
Thanks , I have installed avr-gcc, and while compiling, faced some other issues.

ladislas, 2017-08-04:
as there is no activity, I'm closing the issue. Feel free to reopen if needed

closed 2017-08-04 by ladislas