osx-cross/homebrew-avr #46: Link Time Optimisation 6.2.0

tuna-f1sh, 2017-04-25:
I use homebrew-avr with [Arduino-mk](https://github.com/sudar/Arduino-Makefile) and have been for a while. Updating to the latest formula, I suffered a segmentation fault at the linking stage.

Doing some digging, this is due to the `-flto` flag, set by default for Arduino-mk (and presumably arduino builds) but missing from the 6.2.0 homebrew build. I'm not that well versed with the gcc compiler but [Link Time Optimization](https://gcc.gnu.org/wiki/LinkTimeOptimization) (lto) seems to be merged into the latest branch and was working in the previous homebrew formulae - is there a reason it's missing from the homebrew build?

One can workaround it be [manually setting the C++ flags](https://github.com/sudar/Arduino-Makefile/issues/486) - excluding `-flto` - but it would be nice to have it working by default.

Thanks.

larsimmisch, 2017-04-25:
@tuna-f1sh: can you show a minimal example of the compile/link rules in your Makefile? In particular, do you use ld for linking or gcc?

Note to self: I'm not sure what causes this, because `-enable-lto` is the default according to https://gcc.gnu.org/install/configure.html. But there are some mentions that LTO is a linker plugin, so we may have to make sure the gcc lto plugin is picked up by binutils.

ladislas, 2017-04-25:
I have the same issue. Did not notice it before as i was overriding the flags.

larsimmisch, 2017-04-25:
@tuna-f1sh Just looked at Arduino-mk: disregard my question. `$CC` is invoked as the linker

tuna-f1sh, 2017-04-25:
Appears to be a bug in the compiler. The [Arch package had the same problem](https://bugs.archlinux.org/task/50848). There is a work around but seems to be fixed in the latest avr-libc?[An issue in the Arduino repository](https://github.com/arduino/Arduino/issues/660#issuecomment-167271054) too.

ladislas, 2017-04-25:
that's super interesting! :) 

maybe using avr-gcc 6.3.0 and Binutils 2.26.1 will fix the issue?

https://gcc.gnu.org/gcc-6/changes.html

larsimmisch, 2017-04-25:
The underlying issue seems to be this (https://github.com/arduino/Arduino/issues/660#issuecomment-216265321):

> Small update: lto build on 6.1 segfaults on libraries using virtual inheritance. Hence, adding -fno-devirtualize to c and cpp extra flags solves the segfault.

I would argue that this is a gcc bug - whatever it does, it should not segfault. Can you try compiling with `-fno-devirtualize` and `-flto`?

In any case, this should probably be reported to gcc and it looks as if the best we can do right now is to suggest workarounds - and removing `-flto` does work for you.

Have you got time to prepare a minimal reproduction case?

larsimmisch, 2017-04-25:
@ladislas yes, good idea. But we need a reproducible test case

ladislas, 2017-04-25:
Just tried and it doesn't work...

here is my test case: https://github.com/ladislas/Bare-Arduino-Project/tree/lto-bug

on branch `lto-bug` in `./src/BarProject` just run `make` on OSX.

ladislas, 2017-08-04:
@tuna-f1sh this is being discussed here: https://github.com/sudar/Arduino-Makefile/issues/486

updating arduino-mk should fix the issue.

closed 2017-08-04 by ladislas