osx-cross/homebrew-avr #48: Older versions of gcc

ladislas, 2017-04-25:
As of today, we have 3 versions of GCC:

- latest (6.3.0)
- 4.9
- 4.8

We don't have 5.x.

I'm wondering if we should have them all, or none at all. Doesn't really make sense to jump from 4 to 6.

In hombrew they have all the versions from 4.6 to 5.

My opinion would be to keep only the latest version of each big version:

- 4.x
- 5.x
- 6.x
- etc.

but don't bother with multiple versions of each.

And regarding binutils and avr-libc, only keep the latest version. If they are not compatible with older version of gcc, drop support of those versions of gcc.

what do you thing?

larsimmisch, 2017-04-25:
Yes, I think having the latest version of each major version would be great. So remove 4.8 and add 5.3.

We could also think about 4.9.3 instead of 4.9 (from https://gcc.gnu.org/releases.html)

ladislas, 2017-04-25:
agreed. I'll take care of it!

also, I think that following the new homebrew rule for naming version like `avr-gcc@5.3` would be best.

ladislas, 2017-04-25:
so i gave it more thoughts -- I will name the formula this way

ladislas, 2017-08-04:
closing this issue as I just merged #54

closed 2017-08-04 by ladislas