osx-cross/homebrew-avr #49: 6.3.0 doesn't have pgmspace.h

adzenith, 2017-04-27:
I'm trying to compile [QMK](https://github.com/qmk/qmk_firmware). It used to work fine with 6.2.0, but it looks like 6.3.0 is missing a required header.
6.2.0 had this file:
`/usr/local/Cellar/avr-gcc/6.2.0/avr/include/avr/pgmspace.h`
6.3.0 appears to have no equivalent. Am I doing something wrong? I just `brew update`d.

Thanks!

ladislas, 2017-04-27:
you have to recompile everything from scratch, specifically avr-libc

larsimmisch, 2017-04-27:
It would be nice if we could trigger this automatically, but I thought about it a while back and had no good idea how to do it without resorting to very hacky workarounds

ladislas, 2017-04-27:
this is linked to the fact that we cannot have to versions available at the same time.

someone was talking about sandboxing at some points, it might help solve the problem, but I'm not an expert with homebrew. any ideas?

ladislas, 2017-04-27:
@larsimmisch i'm gonna try that when i get time: https://github.com/cosmo0920/homebrew-avr/blob/e2d70836645309705df172ba0f2e96b5807fef4e/avr-gcc.rb with the `resource` thing

adzenith, 2017-04-27:
Just to confirm my understanding, it's as simple as:
```
brew uninstall avr-libc
brew install avr-libc
```
Is that correct?

larsimmisch, 2017-04-27:
Yes, that should do it.

adzenith, 2017-04-30:
Thanks for the help - it worked great. I'm gonna go ahead and close this.

closed 2017-04-30 by adzenith