osx-cross/homebrew-avr #51: avr-gcc-{4.9.3,6.3.0} fails to build

johu, 2017-05-08:
1. Default install 
```
~ ❯❯❯ brew install avr-libc                                                                                                                                                                                                                                                 ⏎
==> Installing avr-libc from osx-cross/avr
==> Installing dependencies for osx-cross/avr/avr-libc: avr-gcc
==> Installing osx-cross/avr/avr-libc dependency: avr-gcc
==> Downloading ftp://gcc.gnu.org/pub/gcc/releases/gcc-6.3.0/gcc-6.3.0.tar.bz2
Already downloaded: /Users/johannes.huber/Library/Caches/Homebrew/avr-gcc-6.3.0.tar.bz2
==> ../configure --target=avr --prefix=/usr/local/Cellar/avr-gcc/6.3.0 --enable-languages=c,c++ --with-gnu-as --with-gnu-ld --with-ld=/usr/local/opt/avr-binutils/bin/avr-ld --with-as=/usr/local/opt/avr-binutils/bin/avr-as --disable-nls --disable-shared --disable-threads
==> make
Last 15 lines from /Users/johannes.huber/Library/Logs/Homebrew/avr-gcc/02.make:
  "_inflateInit_", referenced from:
      lto_end_uncompression(lto_compression_stream*) in libbackend.a(lto-compress.o)
ld: symbol(s) not found for architecture x86_64
  "_zError", referenced from:
      lto_end_compression(lto_compression_stream*) in libbackend.a(lto-compress.o)
      lto_end_uncompression(lto_compression_stream*) in libbackend.a(lto-compress.o)
clang: error: linker command failed with exit code 1 (use -v to see invocation)
make[2]: *** [cc1plus] Error 1
make[2]: *** Waiting for unfinished jobs....
ld: symbol(s) not found for architecture x86_64
clang: error: linker command failed with exit code 1 (use -v to see invocation)
make[2]: *** [cc1] Error 1
rm gcc.pod
make[1]: *** [all-gcc] Error 2
make: *** [all] Error 2

If reporting this issue please do so at (not Homebrew/brew or Homebrew/core):
https://github.com/osx-cross/homebrew-avr/issues
```

2. tried with lower gcc
```
~ ❯❯❯ brew install avr-gcc49
==> Installing avr-gcc49 from osx-cross/avr
==> Downloading ftp://gcc.gnu.org/pub/gcc/releases/gcc-4.9.3/gcc-4.9.3.tar.bz2
######################################################################## 100,0%
==> ../configure --target=avr --prefix=/usr/local/Cellar/avr-gcc49/4.9.3 --enable-languages=c,c++ --with-gnu-as --with-gnu-ld --with-ld=/usr/local/opt/avr-binutils/bin/avr-ld --with-as=/usr/local/opt/avr-binutils/bin/avr-as --disable-nls --disable-shared --disable-threa
==> make
Last 15 lines from /Users/johannes.huber/Library/Logs/Homebrew/avr-gcc49/02.make:
      lto_end_uncompression(lto_compression_stream*) in libbackend.a(lto-compress.o)
  "_inflateEnd", referenced from:
      lto_end_uncompression(lto_compression_stream*) in libbackend.a(lto-compress.o)
  "_inflateInit_", referenced from:
      lto_end_uncompression(lto_compression_stream*) in libbackend.a(lto-compress.o)
  "_zError", referenced from:
      lto_end_compression(lto_compression_stream*) in libbackend.a(lto-compress.o)
      lto_end_uncompression(lto_compression_stream*) in libbackend.a(lto-compress.o)
ld: symbol(s) not found for architecture x86_64
clang: error: linker command failed with exit code 1 (use -v to see invocation)
make[2]: *** [cc1] Error 1
make[2]: *** Waiting for unfinished jobs....
rm gcc.pod
make[1]: *** [all-gcc] Error 2
make: *** [all] Error 2
```

ladislas, 2017-05-08:
are you sure you installed xcode and/or the command line tool correctly? what is the output of `xcode-select -v && xcode-select -p`

can you run the following:

```
$ brew remove avr-binutils avr-gcc avr-libc
$ brew untap osx-cross/avr
$ brew update && brew upgrade && brew cleanup -scf
$ brew tap osx-cross/avr
$ brew install avr-libc
```
let me know what happens.

also you're not supposed to install gcc directly.

johu, 2017-05-08:
```
❯❯❯ xcode-select -v && xcode-select -p
xcode-select version 2347.
/Applications/Xcode.app/Contents/Developer

❯❯❯ brew remove avr-gcc
Error: No such keg: /usr/local/Cellar/avr-gcc

❯❯❯ brew remove avr-libc
Error: No such keg: /usr/local/Cellar/avr-libc

❯❯❯ brew remove avr-binutils                                                                                                                                                                                                                                
Uninstalling /usr/local/Cellar/avr-binutils/2.28... (140 files, 12.0MB)

❯❯❯ brew untap osx-cross/avr
Untapping osx-cross/avr... (47 files, 68.5KB)
Untapped 9 formulae

❯❯❯ brew update && brew upgrade && brew cleanup -scf
Updated 2 taps (caskroom/cask, homebrew/core).
No changes to formulae.
==> Upgrading 2 outdated packages, with result:
mysql 5.7.18_1, node 7.10.0
==> Upgrading mysql
==> Downloading https://homebrew.bintray.com/bottles/mysql-5.7.18_1.sierra.bottle.tar.gz
######################################################################## 100,0%
==> Pouring mysql-5.7.18_1.sierra.bottle.tar.gz
==> Using the sandbox
==> Caveats
We've installed your MySQL database without a root password. To secure it run:
    mysql_secure_installation

MySQL is configured to only allow connections from localhost by default

To connect run:
    mysql -uroot

To have launchd start mysql now and restart at login:
  brew services start mysql
Or, if you don't want/need a background service you can just run:
  mysql.server start
==> Summary
🍺  /usr/local/Cellar/mysql/5.7.18_1: 320 files, 232.9MB
==> Upgrading node
==> Downloading https://homebrew.bintray.com/bottles/node-7.10.0.sierra.bottle.tar.gz
######################################################################## 100,0%
==> Pouring node-7.10.0.sierra.bottle.tar.gz
==> Caveats
Bash completion has been installed to:
  /usr/local/etc/bash_completion.d
==> Summary
🍺  /usr/local/Cellar/node/7.10.0: 3,078 files, 39.7MB
Removing: /usr/local/Cellar/mysql/5.7.18... (320 files, 232.9MB)
Removing: /usr/local/Cellar/node/7.9.0... (3,078 files, 39.7MB)
Removing: /Users/johannes.huber/Library/Caches/Homebrew/git-2.12.2.sierra.bottle.tar.gz... (12.2MB)
Removing: /Users/johannes.huber/Library/Caches/Homebrew/mysql-5.7.18.sierra.bottle.1.tar.gz... (71.9MB)
Removing: /Users/johannes.huber/Library/Caches/Homebrew/mysql@5.6-5.6.36.sierra.bottle.1.tar.gz... (44.8MB)
Removing: /Users/johannes.huber/Library/Caches/Homebrew/node-7.9.0.sierra.bottle.tar.gz... (10.6MB)
==> This operation has freed approximately 412MB of disk space.

❯❯❯ brew tap osx-cross/avr
==> Tapping osx-cross/avr
Cloning into '/usr/local/Homebrew/Library/Taps/osx-cross/homebrew-avr'...
remote: Counting objects: 13, done.
remote: Compressing objects: 100% (13/13), done.
remote: Total 13 (delta 3), reused 4 (delta 0), pack-reused 0
Unpacking objects: 100% (13/13), done.

❯❯❯ brew install avr-libc
==> Installing avr-libc from osx-cross/avr
==> Installing dependencies for osx-cross/avr/avr-libc: avr-binutils, avr-gcc
==> Installing osx-cross/avr/avr-libc dependency: avr-binutils
==> Downloading http://ftp.gnu.org/gnu/binutils/binutils-2.28.tar.bz2
Already downloaded: /Users/johannes.huber/Library/Caches/Homebrew/avr-binutils-2.28.tar.bz2
==> Downloading https://raw.githubusercontent.com/osx-cross/homebrew-avr/master/avr-binutils-size.patch
Already downloaded: /Users/johannes.huber/Library/Caches/Homebrew/avr-binutils--patch-86ef7717c1de00b47e92275ca7c9d64c7c2484e3646d86cd5955bded2cdc7c1f.patch
==> Patching
==> Applying avr-binutils-size.patch
patching file binutils/size.c
Hunk #8 succeeded at 399 (offset 1 line).
Hunk #9 succeeded at 408 (offset 1 line).
Hunk #10 succeeded at 424 (offset 1 line).
Hunk #11 succeeded at 456 (offset 1 line).
Hunk #12 succeeded at 509 (offset 1 line).
Hunk #13 succeeded at 875 (offset 8 lines).
==> ../configure --disable-debug --disable-dependency-tracking --target=avr --prefix=/usr/local/Cellar/avr-binutils/2.28 --infodir=/usr/local/Cellar/avr-binutils/2.28/share/info --mandir=/usr/local/Cellar/avr-binutils/2.28/share/man --disable-werror --disable-nls
==> make
==> make install
🍺  /usr/local/Cellar/avr-binutils/2.28: 140 files, 12.0MB, built in 1 minute 20 seconds
==> Installing osx-cross/avr/avr-libc dependency: avr-gcc
==> Downloading ftp://gcc.gnu.org/pub/gcc/releases/gcc-6.3.0/gcc-6.3.0.tar.bz2
Already downloaded: /Users/johannes.huber/Library/Caches/Homebrew/avr-gcc-6.3.0.tar.bz2
==> ../configure --target=avr --prefix=/usr/local/Cellar/avr-gcc/6.3.0 --enable-languages=c,c++ --with-gnu-as --with-gnu-ld --with-ld=/usr/local/opt/avr-binutils/bin/avr-ld --with-as=/usr/local/opt/avr-binutils/bin/avr-as --disable-nls --disable-shared --disable-threads
==> make
Last 15 lines from /Users/johannes.huber/Library/Logs/Homebrew/avr-gcc/02.make:
  "_inflateInit_", referenced from:
      lto_end_uncompression(lto_compression_stream*) in libbackend.a(lto-compress.o)
  "_zError", referenced from:
      lto_end_compression(lto_compression_stream*) in libbackend.a(lto-compress.o)
      lto_end_uncompression(lto_compression_stream*) in libbackend.a(lto-compress.o)
ld: symbol(s) not found for architecture x86_64
clang: error: linker command failed with exit code 1 (use -v to see invocation)
make[2]: *** [cc1] Error 1
make[2]: *** Waiting for unfinished jobs....
ld: symbol(s) not found for architecture x86_64
clang: error: linker command failed with exit code 1 (use -v to see invocation)
make[2]: *** [cc1plus] Error 1
rm gcc.pod
make[1]: *** [all-gcc] Error 2
make: *** [all] Error 2

If reporting this issue please do so at (not Homebrew/brew or Homebrew/core):
https://github.com/osx-cross/homebrew-avr/issues

These open issues may also help:
avr-gcc-{4.9.3,6.3.0} fails to build https://github.com/osx-cross/homebrew-avr/issues/51
```

Please let me know if you need more output.

larsimmisch, 2017-05-08:
lto again, from the looks of it. I

johu, 2017-05-08:
Maybe [this helps to fix the issue](http://stackoverflow.com/questions/25020140/undefined-symbols-for-architecture-x86-64-mavericks-yosemite-el-capitan).

ladislas, 2017-05-08:
@larsimmisch yes lto again, but it's strange, I can compile the thing... 

@johu can you try the following:

```
# cd to the tap
$ cd /usr/local/Homebrew/Library/Taps/osx-cross/homebrew-avr

# checkout the update-formula branch
$ git checkout update-formula

# try to install avr-gcc
$ brew install avr-gcc
```

and let me know what happens...

ladislas, 2017-05-08:
@johu can you also show us the output of `brew doctor`?

larsimmisch, 2017-05-08:
@ladislas it worked for me too, so I suspect there is some cruft left over

johu, 2017-05-08:
> @johu can you also show us the output of brew doctor?

```
❯❯❯ brew doctor
Please note that these warnings are just used to help the Homebrew maintainers
with debugging if you file an issue. If everything you use Homebrew for is
working fine: please don't worry and just ignore them. Thanks!

Warning: "config" scripts exist outside your system or Homebrew directories.
`./configure` scripts often look for *-config scripts to determine if
software packages are installed, and what additional flags to use when
compiling and linking.

Having additional scripts in your path can confuse software installed via
Homebrew if the config script overrides a system or Homebrew provided
script of the same name. We found the following "config" scripts:
  /opt/ImageMagick/bin/Magick++-config
  /opt/ImageMagick/bin/Magick-config
  /opt/ImageMagick/bin/MagickCore-config
  /opt/ImageMagick/bin/MagickWand-config
  /opt/ImageMagick/bin/Wand-config

Warning: Unbrewed dylibs were found in /usr/local/lib.
If you didn't put them there on purpose they could cause problems when
building Homebrew formulae, and may need to be deleted.

Unexpected dylibs:
  /usr/local/lib/libcrypto.0.9.8.dylib
  /usr/local/lib/libcrypto.1.0.0.dylib
  /usr/local/lib/libcrypto.dylib
  /usr/local/lib/libklcsagt.dylib
  /usr/local/lib/libklcskca.dylib
  /usr/local/lib/libklcsnagt.dylib
  /usr/local/lib/libklcsrt.dylib
  /usr/local/lib/libklcsstd.dylib
  /usr/local/lib/libklcstr.dylib
  /usr/local/lib/libklmspack.0.1.0.dylib
  /usr/local/lib/libklmspack.0.dylib
  /usr/local/lib/libklmspack.dylib
  /usr/local/lib/libssl.0.9.8.dylib
  /usr/local/lib/libssl.1.0.0.dylib
  /usr/local/lib/libssl.dylib
  /usr/local/lib/libz.1.2.5.dylib
  /usr/local/lib/libz.1.2.6.dylib
  /usr/local/lib/libz.1.dylib
  /usr/local/lib/libz.dylib
```

ladislas, 2017-05-08:
thanks @johu.

When trying to reinstall, can you add the `--verbose` argument so we can see what exactly went wrong?

johu, 2017-05-08:
@ladislas with --verbose output https://dpaste.de/1zq7/raw

ladislas, 2017-05-08:
@johu well... I don't know what might be causing that. Maybe you have some crufts left over as @larsimmisch mentioned but I'm afraid I cannot help you.

if you gave another computer, could you try again?

larsimmisch, 2017-05-08:
@johu I would try to move `/usr/local/lib/libz*` away and then retry.

johu, 2017-05-09:
Thanks a lot. Maybe this manual step to clean up old libs could be documented, if somebody encounters the same problem.  
```
==> Summary
🍺  /usr/local/Cellar/avr-gcc/6.3.0: 738 files, 618.9MB, built in 11 minutes 15 seconds
```

closed 2017-05-09 by johu

ladislas, 2017-05-09:
@johu good to know it worked! 

yeah, this sometimes happen. homebrew has internal documentation on that and running brew doctor is part of the process. The problem is that we can't just tell people to remove everything as somethings might be really useful. it's really on a case by case basis...

but in general cleaning up brew every other month is a good thing to do :)

sslupsky, 2018-02-08:
If anyone is reading further on this topic, I had a compile problem compiling avr-gcc@4 with Xcode 9. 

Last 15 lines from /Users/stevenslupsky/Library/Logs/Homebrew/avr-gcc@4/02.make:
/Applications/Xcode.app/Contents/Developer/Toolchains/XcodeDefault.xctoolchain/usr/include/c++/v1/exception:173:5: error: no member named 'fancy_abort' in namespace 'std::__1'; did you mean simply 'fancy_abort'?
    _VSTD::abort();
    ^~~~~~~
/Applications/Xcode.app/Contents/Developer/Toolchains/XcodeDefault.xctoolchain/usr/include/c++/v1/__config:392:15: note: expanded from macro '_VSTD'
#define _VSTD std::_LIBCPP_NAMESPACE
              ^
../../gcc/system.h:685:13: note: 'fancy_abort' declared here
extern void fancy_abort (const char *, int, const char *) ATTRIBUTE_NORETURN;
            ^
1 error generated.
make[2]: *** [c/c-objc-common.o] Error 1
make[2]: *** Waiting for unfinished jobs....
rm gcc.pod
make[1]: *** [all-gcc] Error 2
make: *** [all] Error 2

I downgraded to Xcode to 8.3.3 and the toolchain brewed successfully.