osx-cross/homebrew-avr #52: Formulae found in multiple taps loops

vcavallo, 2017-06-27:
Getting the following error trying to install avr-libc:

```
$ brew install avr-libc
Error: Formulae found in multiple taps:
 * larsimmisch/avr/avr-libc
 * osx-cross/avr/avr-libc

Please use the fully-qualified name e.g. larsimmisch/avr/avr-libc to refer the formula.
$ brew install osx-cross/avr/avr-libc
==> Installing avr-libc from osx-cross/avr
Error: Formulae found in multiple taps:
 * larsimmisch/avr/avr-gcc
 * osx-cross/avr/avr-gcc

Please use the fully-qualified name e.g. larsimmisch/avr/avr-gcc to refer the formula.
```

I've already `brew tap osx-cross/avr` and installed xcode command line tools. Let me know what other information you need from me.  
Thanks!

vcavallo, 2017-06-27:
Some further troubleshooting and frankly blind goofing around:

```
 $ brew untap osx-cross/avr
Untapping osx-cross/avr... (65 files, 109KB)
Untapped 9 formulae
 $ brew tap osx-cross/avr
==> Tapping osx-cross/avr
Cloning into '/usr/local/Homebrew/Library/Taps/osx-cross/homebrew-avr'...
remote: Counting objects: 13, done.
remote: Compressing objects: 100% (13/13), done.
remote: Total 13 (delta 3), reused 2 (delta 0), pack-reused 0
Unpacking objects: 100% (13/13), done.
Tapped 9 formulae (46 files, 67.5KB)
 $ brew install osx-cross/avr/avr-libc
==> Installing avr-libc from osx-cross/avr
Error: Formulae found in multiple taps:
 * larsimmisch/avr/avr-gcc
 * osx-cross/avr/avr-gcc

Please use the fully-qualified name e.g. larsimmisch/avr/avr-gcc to refer the formula.
 $ brew install osx-cross/avr/avr-gcc
==> Installing avr-gcc from osx-cross/avr
Error: Formulae found in multiple taps:
 * larsimmisch/avr/avr-binutils
 * osx-cross/avr/avr-binutils

Please use the fully-qualified name e.g. larsimmisch/avr/avr-binutils to refer the formula.
 $ brew install osx-cross/avr/avr-binutils
Warning: osx-cross/avr/avr-binutils 2.28 is already installed
```

vcavallo, 2017-06-27:
...and as is often the case when I open issues, it's fixed a minute later:

```
 $ brew untap larsimmisch/avr
Untapping larsimmisch/avr... (34 files, 25.9KB)
Untapped 3 formulae
$ brew untap osx-cross/avr
Untapping osx-cross/avr... (47 files, 68.5KB)
Untapped 9 formulae
 $ brew tap osx-cross/avr
==> Tapping osx-cross/avr
Cloning into '/usr/local/Homebrew/Library/Taps/osx-cross/homebrew-avr'...
remote: Counting objects: 13, done.
remote: Compressing objects: 100% (13/13), done.
remote: Total 13 (delta 3), reused 2 (delta 0), pack-reused 0
Unpacking objects: 100% (13/13), done.
Tapped 9 formulae (46 files, 67.5KB)
 $ brew install osx-cross/avr/avr-libc
==> Installing avr-libc from osx-cross/avr
==> Installing dependencies for osx-cross/avr/avr-libc: mpfr, libmpc, avr-gcc
[.....etc.....]
[.....etc.....]
[.....etc.....]
```

Might be worth mentioning somewhere that if someone has both tapped they should remove the `larsimmisch` one?

ladislas, 2017-06-27:
yup you came up with the solution :) 

thanks for letting us know, I'll close the issue now.

closed 2017-06-27 by ladislas