osx-cross/homebrew-avr #55: avr-gcc 7.1.0 will not find <avr/io.h>

ubidefeo, 2017-07-26:
just updated but had to `brew switch avr-gcc 6.3.0` to have my code compile again.

```
Compiling C: main.c
avr-gcc -c -mmcu=atmega8515 -I. -gdwarf-2  -Os -funsigned-char -funsigned-bitfields -fpack-struct -fshort-enums -Wall -Wstrict-prototypes -Wa,-adhlns=./main.lst  -std=gnu99 -MMD -MP -MF .dep/main.o.d main.c -o main.o
main.c:34:10: fatal error: avr/io.h: No such file or directory
 #include <avr/io.h>
          ^~~~~~~~~~
compilation terminated.
make: *** [main.o] Error 1
```

ddavidebor, 2017-08-02:
same issue here with gcc 7, on a project that compiles correctly on Arch with the same avr-gcc version

ladislas, 2017-08-02:
you need to recompile avrlibc for each avr version.

this is fixed in branch https://github.com/osx-cross/homebrew-avr/tree/update-7.1-6.4

will merge soon to master

codmpm, 2017-08-02:
Having the problem right now. How can I use the update-7.1-6.4 version?

ladislas, 2017-08-04:
@codmpm navigate to `lekaleka /usr/local/Homebrew/Library/Taps/osx-cross/homebrew-avr/` and switch branch

codmpm, 2017-08-04:
Thank you!

But sadly I only have a master branch:
[image]

ladislas, 2017-08-04:
@codmpm i just merge the changes, brew update should pull them and it should work. if not untap and tap again ;) 

and as for only seeing the master branch, I had the same issue, you need to:

`git config remote.origin.fetch "+refs/heads/*:refs/remotes/origin/*"``

fixed it for me :)

codmpm, 2017-08-04:
That did the trick - thank you!

[image]

codmpm, 2017-08-04:
Sorry to bother you again... but I think I've messed something up:

[image]

ladislas, 2017-08-04:
oh sorry! I have to update the readme. You just need to run `brew install avr-gcc` :)

ladislas, 2017-08-04:
but as said, it is now merged in master, so no need to switch branches.

codmpm, 2017-08-04:
It is working from master. Thank you so much!

closed 2017-08-04 by ladislas