osx-cross/homebrew-avr #58: Missing avr-libc breaks simavr formula

porglezomp, 2017-08-21:
```
~ $ brew install --HEAD simavr
==> Installing simavr from osx-cross/avr
Error: No available formula with the name "avr-libc" (dependency of osx-cross/avr/simavr)
==> Searching for a previously deleted formula...
Warning: homebrew/core is shallow clone. To get complete history run:
  git -C "$(brew --repo homebrew/core)" fetch --unshallow

Error: No previously deleted formula found.
==> Searching for similarly named formulae...
==> Searching local taps...
Error: No similarly named formulae found.
==> Searching taps...
==> Searching taps on GitHub...
Error: No formulae found in taps.
```
I suspect this was caused by #54 removing the `avr-libc` formula, and not changing `simavr` to use the resource, but I don't know how to test with a local formula to confirm this.

ladislas, 2017-08-21:
I just created a PR with the fix - https://github.com/osx-cross/homebrew-avr/pull/59

Can you try it? 

how to here: https://help.github.com/articles/checking-out-pull-requests-locally/

ladislas, 2017-08-21:
well... doesn't seem to work... 

@obra @jacquesg - any ideas guys on what we should do?

porglezomp, 2017-08-22:
Hm, I was messing around working with it locally, and accidentally got it to run fine with the same package after I installed `avr-gcc` first. I had a few broken steps in between, so I'm not sure if that's all that's changed. Does setting `simavr` to depend on `avr-gcc` work?

ladislas, 2017-08-22:
yep that might be it!

ladislas, 2017-09-04:
Just merged the changes. it's working now and pulling avr-gcc as a resource which contains avr-libc and avr-binutils.

ladislas, 2017-09-04:
thanks for your investigation @porglezomp 👍

commit cfd87fd8 (2017-08-21): add avrlibc as a resource to simavr - fix #58
commit 4b351246 (2017-08-21): add avrlibc as a resource to simavr - fix #58
commit 9e268578 (2017-09-04): add avr-gcc as a resource to simavr - fix #58
closed 2017-09-04 by ladislas