osx-cross/homebrew-avr #61: Fails in upgrade avr-gcc, cause of fails in fetching avr-libc 2.0 from server.

imeteora, 2017-08-25:
Runs upgrading avr-gcc to 7.2.0, always fails in downloading avr-libc 2.0.0 from TW's server. Here is console's log:

$ brew upgrade       
==> Upgrading 1 outdated package, with result:
osx-cross/avr/avr-gcc 7.2.0
==> Upgrading osx-cross/avr/avr-gcc 
==> Downloading https://gcc.gnu.org/pub/gcc/releases/gcc-7.2.0/gcc-7.2.0.tar.xz
Already downloaded: /Users/zhudelun/Library/Caches/Homebrew/avr-gcc-7.2.0.tar.xz
==> ../configure --target=avr --prefix=/usr/local/Cellar/avr-gcc/7.2.0 --enable-languages=c,c++ --with-ld=/usr/local/opt
==> make
==> make install
==> Downloading https://download.savannah.gnu.org/releases/avr-libc/avr-libc-2.0.0.tar.bz2
==> Downloading from http://ftp.yzu.edu.tw/nongnu/avr-libc/avr-libc-2.0.0.tar.bz2

curl: (7) Failed to connect to ftp.yzu.edu.tw port 80: Operation timed out
Error: Failed to download resource "avr-gcc--avr-libc"
Download failed: https://download.savannah.gnu.org/releases/avr-libc/avr-libc-2.0.0.tar.bz2

How to fix it?

ladislas, 2017-08-25:
does pasting `https://download.savannah.gnu.org/releases/avr-libc/avr-libc-2.0.0.tar.bz2` in your browser work?

it seems like the the link resolves to the mirror closest to you and then fails.
if we can find a mirror that's working all the time, I can add that to the formula.

ladislas, 2017-08-25:
update: actually this link http://ftp.yzu.edu.tw/nongnu/avr-libc/avr-libc-2.0.0.tar.bz2 works for me

imeteora, 2017-08-25:
Oh, I see. I think that my company's bad network or the Chinese GFW cause this issue happens for me.  I will try it again in my home later. Thanks.

closed 2017-08-25 by imeteora