osx-cross/homebrew-avr #62: Not able to update toolchain

MCUdude, 2017-08-26:
Hi!
About a year ago I installed the AVR toolchain on my mac using [this guide](http://maxembedded.com/2015/06/setting-up-avr-gcc-toolchain-on-linux-and-mac-os-x/). It worked like a charm, and I currently have avr-gcc 4.9.3 and avr-libc 2.0.0 installed.

I'm working [on a project](https://github.com/MCUdude/optiboot_flash) where I want to compile optiboot for the new ATmega48PB/88PB/168PB/328PB chips, but I'm getting an error saying that the device isn't supported (or in the list of supported devices). 

In order to update the toolchain I tried to run the following command; resulting in an error:
```
$ brew upgrade avr-libc
Error: No available formula with the name "avr-libc" 
```

... And trying to install avr-gcc@4 didn't work either
```
$ brew install avr-gcc@4
==> Installing avr-gcc@4 from osx-cross/avr
==> Downloading ftp://gcc.gnu.org/pub/gcc/releases/gcc-4.9.4/gcc-4.9.4.tar.bz2
Already downloaded: /Users/Hans/Library/Caches/Homebrew/avr-gcc@4-4.9.4.tar.bz2

==> ../configure --target=avr --prefix=/usr/local/Cellar/avr-gcc@4/4.9.4 --enable-languages=c,c++ --with-gnu-as --with-gnu-
==> make
Last 15 lines from /Users/Hans/Library/Logs/Homebrew/avr-gcc@4/02.make:
	       -v header_name="config.h system.h coretypes.h options.h tm.h" < optionlist > options.c
if [ xinfo = xinfo ]; then \
		makeinfo --split-size=5000000 --split-size=5000000 --no-split -I . -I ../../gcc/doc \
			-I ../../gcc/doc/include -o doc/gcc.info ../../gcc/doc/gcc.texi; \
	fi
../../gcc/doc//invoke.texi:1535: Unknown command `4-20170826-74408-aqhezd/gcc-4.9.4/build/gcc/../../gcc'.
../../gcc/doc//invoke.texi:1534: @include `/private/tmp/avr-gcc/../libiberty/at-file.texi': No such file or directory.
makeinfo: Removing output file `doc/gcc.info' due to errors; use --force to preserve.
make[2]: *** [doc/gcc.info] Error 1
make[2]: *** Waiting for unfinished jobs....
/bin/sh ../../gcc/../move-if-change tmp-mlib.h multilib.h
echo timestamp > s-mlib
rm gcc.pod
make[1]: *** [all-gcc] Error 2
make: *** [all] Error 2

If reporting this issue please do so at (not Homebrew/brew or Homebrew/core):
https://github.com/osx-cross/homebrew-avr/issues
```

How can I update the toolchain in order to compile for the new *PB series?

ladislas, 2017-08-28:
Hi @MCUdude --

We updated the way the formula work. Now you don't need to install avr-libc and avr-gcc separately because avr-libc is now a resource of avr-gcc.

You should use the latest version of avr-gcc. I'm using it all the time on old and new code and it works perfectly. A lot of bugs have been fixed since 4.9 and a lot of nice features and optimizations have been added.

It might be better to start from scratch to avoid any issues in the future. Run the following:

```
# remove old versions of avr-gcc
$ brew remove avr-gcc avr-libc avr-binutils

# untap / retap the tap
$ brew untap osx-cross/avr && brew cleanup -scf && brew tap osx-cross/avr

# install the toolchain
$ brew install avr-gcc
```

Enjoy! 👍

MCUdude, 2017-08-28:
Thanks, but I'm still getting an error.
I really want to use avr-gcc 4.9, since I'm compiling a bootloader that has to be less than 512B.
I was able to install avr-binutils, but not the rest of the packages.

```
$ brew install avr-gcc@4
==> Installing avr-gcc@4 from osx-cross/avr
==> Installing dependencies for osx-cross/avr/avr-gcc@4: avr-binutils
==> Installing osx-cross/avr/avr-gcc@4 dependency: avr-binutils
==> Downloading http://ftp.gnu.org/gnu/binutils/binutils-2.28.tar.bz2
Already downloaded: /Users/Hans/Library/Caches/Homebrew/avr-binutils-2.28.tar.bz2
==> Downloading https://raw.githubusercontent.com/osx-cross/homebrew-avr/master/avr-binutils-size.patch
Already downloaded: /Users/Hans/Library/Caches/Homebrew/avr-binutils--patch-4748f87aee912f954be9968a5a01f61e4f1897adf21e2549d0ac988b9fe8ef1d.patch
==> Patching
==> Applying avr-binutils-size.patch
patching file binutils/size.c
==> ../configure --prefix=/usr/local/Cellar/avr-binutils/2.28 --infodir=/usr/local/Cellar/avr-binutils/2.28/share/info --ma
==> make
==> make install
🍺  /usr/local/Cellar/avr-binutils/2.28: 140 files, 12.0MB, built in 2 minutes 10 seconds
==> Installing osx-cross/avr/avr-gcc@4
==> Downloading ftp://gcc.gnu.org/pub/gcc/releases/gcc-4.9.4/gcc-4.9.4.tar.bz2
Already downloaded: /Users/Hans/Library/Caches/Homebrew/avr-gcc@4-4.9.4.tar.bz2
==> ../configure --target=avr --prefix=/usr/local/Cellar/avr-gcc@4/4.9.4 --enable-languages=c,c++ --with-gnu-as --with-gnu-
==> make
Last 15 lines from /Users/Hans/Library/Logs/Homebrew/avr-gcc@4/02.make:
	       -v header_name="config.h system.h coretypes.h options.h tm.h" < optionlist > options.c
if [ xinfo = xinfo ]; then \
		makeinfo --split-size=5000000 --split-size=5000000 --no-split -I . -I ../../gcc/doc \
			-I ../../gcc/doc/include -o doc/gcc.info ../../gcc/doc/gcc.texi; \
	fi
../../gcc/doc//invoke.texi:1535: Unknown command `4-20170828-35401-girvtj/gcc-4.9.4/build/gcc/../../gcc'.
../../gcc/doc//invoke.texi:1534: @include `/private/tmp/avr-gcc/../libiberty/at-file.texi': No such file or directory.
makeinfo: Removing output file `doc/gcc.info' due to errors; use --force to preserve.
make[2]: *** [doc/gcc.info] Error 1
make[2]: *** Waiting for unfinished jobs....
/bin/sh ../../gcc/../move-if-change tmp-mlib.h multilib.h
echo timestamp > s-mlib
rm gcc.pod
make[1]: *** [all-gcc] Error 2
make: *** [all] Error 2

If reporting this issue please do so at (not Homebrew/brew or Homebrew/core):
https://github.com/osx-cross/homebrew-avr/issues
```

ladislas, 2017-08-28:
> I really want to use avr-gcc 4.9, since I'm compiling a bootloader that has to be less than 512B.

what makes you think it will not be the case with gcc 7.2?

MCUdude, 2017-08-28:
I don't. It's just that I want to stick with the compiler version that Arduino IDE + the optiboot maintainers are using.

Does this means that there's no way to use brew for installing avr-gcc 4.9? If avr-gcc 7.2 is so great, why does Atmel studio ship with avr-gcc 4.9.2? No attack, just curious 🙂

ladislas, 2017-08-28:
>  If avr-gcc 7.2 is so great, why does Atmel studio ship with avr-gcc 4.9.2?

I'd say because they are afraid of the future and don't want people to use the latest C++ features to make their code more readable and easier to write 😉 

Anyway I just pushed some changes and things should be fixed now. just `brew upgrade && brew install avr-gcc@4` and you should be good to go. Took me around 30 minutes to compile so be patient and use `-v` if you want to see the progress.

closed 2017-08-28 by ladislas

ladislas, 2017-08-28:
and also, you can have the different versions living side by side now. `avr-gcc@4` won't be linked to `/usr/local/bin` by default. You can force that with `brew link avr-gcc@4` or have your makefile point to the right directory.

I'm curious if you could also install version 7.2 and compare the .hex files generated.

MCUdude, 2017-08-28:
> I'm curious if you could also install version 7.2 and compare the .hex files generated.

I certainly will! I'm away the next week, but I'll post the hex files later.

BTW which package contains the different microcontroller options? Turns out the default avr-gcc@4 installation doesn't support _any_ of the new *PB series

ladislas, 2017-08-29:
I honestly don't know, but from the [documentation](https://gcc.gnu.org/onlinedocs/gcc/AVR-Options.html), it seems that you need to use `--mmcu=mcu` to something specific. The PB mcu's appear in the documentation in 2015. You might want to try with avr-gcc@5 or @6 if it is supported.

commit 967b33d5 (2017-08-28): Merge branch 'fix-older-versions' - fix #62