osx-cross/homebrew-avr #63: hangs on make install

miguelmota, 2017-09-23:
```
$ brew install avr-gcc
Updating Homebrew...
==> Auto-updated Homebrew!
Updated 2 taps (caskroom/cask, homebrew/core).
==> Updated Formulae
apache-geode     apibuilder-cli   brotli           docker2aci       efl              faad2            grails           grpc             lean             ott              paket            sysdig

==> Installing avr-gcc from osx-cross/avr
==> Downloading https://gcc.gnu.org/pub/gcc/releases/gcc-7.2.0/gcc-7.2.0.tar.xz
Already downloaded: /Users/mota/Library/Caches/Homebrew/avr-gcc-7.2.0.tar.xz
==> ../configure --target=avr --prefix=/usr/local/Cellar/avr-gcc/7.2.0 --enable-languages=c,c++ --with-ld=/usr/local/opt/avr-binutils/bin/avr-ld --with-as=/usr/local/opt/avr-binutils/bin/avr-as --disable-
==> make
==> make install
==> Downloading https://download.savannah.gnu.org/releases/avr-libc/avr-libc-2.0.0.tar.bz2
==> Downloading from http://mirror.csclub.uwaterloo.ca/nongnu/avr-libc/avr-libc-2.0.0.tar.bz2
######################################################################## 100.0%
==> ./configure --build=i686-apple-darwin16.7.0 --prefix=/usr/local/Cellar/avr-gcc/7.2.0 --host=avr
==> make install
```

```
$ uname -a
Darwin Chromecast-b.local 16.7.0 Darwin Kernel Version 16.7.0: Thu Jun 15 17:36:27 PDT 2017; root:xnu-3789.70.16~2/RELEASE_X86_64 x86_64
```

ladislas, 2017-09-23:
I'm not sure I understand the issue here...

it can take up to 40 minutes to compile GCC depending on your system.

If you want to make sure it still work, use `brew install avr-gcc --verbose`

closed 2017-09-24 by miguelmota

ladislas, 2017-09-24:
it was a pleasure to help!

miguelmota, 2017-09-24:
Thanks! it just took a while like you said