osx-cross/homebrew-avr #66: Cannot install

alkopop79, 2017-11-24:
It seems that qt5 depends on this this library. So far I have not managed to install it. 

```
alkopop79@MacBook-Pro ~> brew install qt5
Error: Calling Formula.sha1 is disabled!
Use Formula.sha256 instead.
/usr/local/Homebrew/Library/Taps/larsimmisch/homebrew-avr/avr-binutils.rb:7:in `<class:AvrBinutils>'
Please report this to the larsimmisch/avr tap!
If reporting this issue please do so at (not Homebrew/brew or Homebrew/core):
  https://github.com/larsimmisch/homebrew-avr/issues
alkopop79@MacBook-Pro ~> brew tap osx-cross/avr
alkopop79@MacBook-Pro ~> brew install avr-gcc
Error: Formulae found in multiple taps: 
       * osx-cross/avr/avr-gcc
       * larsimmisch/avr/avr-gcc

Please use the fully-qualified name e.g. osx-cross/avr/avr-gcc to refer the formula.
```

ladislas, 2017-11-24:
You should `brew untap larsimmisch/avr` and then `brew cleanup -scf` and uninstall all the `avr` related tools to start from scratch.

also make sure to run `brew update && brew upgrade`

But I don't see how qt5 could depend on this formula as there is no mention of us in [their formula](https://github.com/Homebrew/homebrew-core/blob/master/Formula/qt.rb).

ladislas, 2017-11-24:
I just tried installing qt5 and it works like a charm. it is even bottled.

closed 2017-11-24 by larsimmisch