osx-cross/homebrew-avr #67: Conflict of avr-gcc with gcc@7 from homebrew tap

guneemwelloeux, 2017-12-11:
Hi,

I just installed avr-gcc@7 and the install failed at the linking stage:
```
Error: The `brew link` step did not complete successfully
The formula built, but is not symlinked into /usr/local
Could not symlink lib/gcc/7/libcc1.0.so
Target /usr/local/lib/gcc/7/libcc1.0.so
```

In fact, I already had gcc@7 installed, and this is the reason for the clash.
Wouldn't it be possible to add a dependency from avr-gcc to gcc@7 instead of recompiling gcc in the avr-gcc formula?

Thanks
Will

ladislas, 2017-12-11:
Thanks for reporting that. I'll look into it.

> Wouldn't it be possible to add a dependency from avr-gcc to gcc@7 instead of recompiling gcc in the avr-gcc formula?

No because avr-gcc is compiled with a specific target (avr) which is not the case with vanilla gcc.

leojrfs, 2017-12-11:
I can confirm this issue

ladislas, 2017-12-11:
are you sure you have pulled the latest version of the formula?

a similar issue was reported here #41 and the solution was found here: https://github.com/osx-cross/homebrew-avr/commit/50b95f3884c0ce0f450d489cb1a443282f1567b5

guneemwelloeux, 2017-12-11:
brew update said everything was up to date, so I believe I am using the latest version. I don't have access to my macbook at the moment, but I'll try and give more info tonight, in particular to make sure I have installed gcc7 from the correct tap, and using the latest formula.

ladislas, 2017-12-11:
I've found the issue. The path to libdir used by avr-gcc is `../gcc/..` that's why it's conflicting.

i'll update the formula asap.

guneemwelloeux, 2017-12-11:
If I understood the formula correctly, that means that we'll be able to live with both toolchains installed and brew-linked simultaneously, which is just perfect.
Thanks for the speedy resolution!

ladislas, 2017-12-11:
just merged it

closed 2017-12-11 by ladislas

guneemwelloeux, 2017-12-11:
I ran `brew update` just now, but it is not reporting avr-gcc as being updated. Do I need to force remove and install again?

leojrfs, 2017-12-11:
brew update and then brew reinstall it

ladislas, 2017-12-13:
@guneemwelloeux you can `brew remove` before if needed.

guneemwelloeux, 2017-12-14:
Everything worked fine after the reinstall, thank you for the fix!

commit fca082b8 (2017-12-11): change libdir path - fix #67