osx-cross/homebrew-avr #68: cannot install on high sierra

KrzysztofKowalski, 2017-12-26:
command used:
brew install --HEAD osx-cross/avr/avr-gcc osx-cross/avr/avarice osx-cross/avr/simavr 

log:

==> Installing avr-gcc from osx-cross/avr
==> Installing dependencies for osx-cross/avr/avr-gcc: gmp, mpfr, libmpc, avr-binutils
==> Installing osx-cross/avr/avr-gcc dependency: gmp
==> Downloading https://homebrew.bintray.com/bottles/gmp-6.1.2_1.high_sierra.bottle.tar.gz
==> Downloading from https://akamai.bintray.com/ea/eadb377c507f5d04e8d47861fa76471be6c09dc54991540e125ee1cbc04fecd6?__gda__=exp=15
######################################################################## 100.0%
==> Pouring gmp-6.1.2_1.high_sierra.bottle.tar.gz
🍺  /usr/local/Cellar/gmp/6.1.2_1: 18 files, 3.1MB
==> Installing osx-cross/avr/avr-gcc dependency: mpfr
==> Downloading https://homebrew.bintray.com/bottles/mpfr-3.1.6.high_sierra.bottle.tar.gz
==> Downloading from https://akamai.bintray.com/ff/ff2f02099a071f15f73ac776026c30e33bb43f9b389b19b87f575cd9bd4ac0bb?__gda__=exp=15
######################################################################## 100.0%
==> Pouring mpfr-3.1.6.high_sierra.bottle.tar.gz
🍺  /usr/local/Cellar/mpfr/3.1.6: 26 files, 3.6MB
==> Installing osx-cross/avr/avr-gcc dependency: libmpc
==> Downloading https://homebrew.bintray.com/bottles/libmpc-1.0.3_1.high_sierra.bottle.tar.gz
######################################################################## 100.0%
==> Pouring libmpc-1.0.3_1.high_sierra.bottle.tar.gz
🍺  /usr/local/Cellar/libmpc/1.0.3_1: 12 files, 345.1KB
==> Installing osx-cross/avr/avr-gcc dependency: avr-binutils
==> Downloading http://ftp.gnu.org/gnu/binutils/binutils-2.29.tar.bz2
######################################################################## 100.0%
==> Downloading https://raw.githubusercontent.com/osx-cross/homebrew-avr/master/avr-binutils-size.patch
######################################################################## 100.0%
==> Patching
==> Applying avr-binutils-size.patch
patching file binutils/size.c
==> ../configure --prefix=/usr/local/Cellar/avr-binutils/2.29 --infodir=/usr/local/Cellar/avr-binutils/2.29/share/info --mandir=/u
==> make
==> make install
🍺  /usr/local/Cellar/avr-binutils/2.29: 140 files, 12.2MB, built in 2 minutes 30 seconds
==> Installing osx-cross/avr/avr-gcc --HEAD
==> Cloning svn://gcc.gnu.org/svn/gcc/trunk
==> ../configure --target=avr --prefix=/usr/local/Cellar/avr-gcc/HEAD-255997 --libdir=/usr/local/Cellar/avr-gcc/HEAD-255997/lib/av
==> make
==> make install
==> Downloading https://download.savannah.gnu.org/releases/avr-libc/avr-libc-2.0.0.tar.bz2
==> Downloading from https://mirror.netcologne.de/savannah/avr-libc/avr-libc-2.0.0.tar.bz2
######################################################################## 100.0%
==> ./configure --build=i686-apple-darwin17.3.0 --prefix=/usr/local/Cellar/avr-gcc/HEAD-255997 --host=avr
==> make install
Last 15 lines from /Users/k/Library/Logs/Homebrew/avr-gcc/05.make:
avr-gcc -DHAVE_CONFIG_H -I. -I../../..  -I../../../common -I../../../include -I../../../include  -Wall -W -Wstrict-prototypes -mmcu=avr2 -D__COMPILING_AVR_LIBC__ -mcall-prologues -Os  -MT asctime_r.o -MD -MP -MF .deps/asctime_r.Tpo -c -o asctime_r.o ../../../libc/time/asctime_r.c
during RTL pass: expand
../../../libc/time/asctime_r.c: In function 'asctime_r':
../../../libc/time/asctime_r.c:57:16: internal compiler error: in convert_memory_address_addr_space_1, at explow.c:300
      buffer[i] = ascdays[d++];
      ~~~~~~~~~~^~~~~~~~~~~~~~
libbacktrace could not find executable to open
Please submit a full bug report,
with preprocessed source if appropriate.
See <https://gcc.gnu.org/bugs/> for instructions.
make[4]: *** [asctime_r.o] Error 1
make[3]: *** [install-recursive] Error 1
make[2]: *** [install-recursive] Error 1
make[1]: *** [install-recursive] Error 1
make: *** [install-recursive] Error 1

If reporting this issue please do so at (not Homebrew/brew or Homebrew/core):
https://github.com/osx-cross/homebrew-avr/issues

ladislas, 2018-01-03:
have you tried removing the `--HEAD`?

werty37, 2018-01-16:
Works for me on macOS High Sierra 10.13.2 (17C88)
```
brew tap osx-cross/avr
brew install avr-gcc
```

```
Updating Homebrew...
==> Auto-updated Homebrew!
Updated 2 taps (homebrew/core, caskroom/cask).
==> Updated Formulae
app-engine-go-64    gobuster            unixodbc            wwwoffle
astyle              irssi               verilator           xvid
autogen             kubeless            whohas              yaze-ag
backupninja         mikutter            wiggle              zabbix

==> Tapping osx-cross/avr
Cloning into '/usr/local/Homebrew/Library/Taps/osx-cross/homebrew-avr'...
remote: Counting objects: 14, done.
remote: Compressing objects: 100% (12/12), done.
remote: Total 14 (delta 4), reused 4 (delta 0), pack-reused 0
Unpacking objects: 100% (14/14), done.
Tapped 8 formulae (47 files, 66KB)
Sujiths-MacBook-Pro:~ sujith$ brew install avr-gcc
==> Installing avr-gcc from osx-cross/avr
==> Installing dependencies for osx-cross/avr/avr-gcc: avr-binutils
==> Installing osx-cross/avr/avr-gcc dependency: avr-binutils
==> Downloading http://ftp.gnu.org/gnu/binutils/binutils-2.29.tar.bz2
######################################################################## 100.0%
==> Downloading https://raw.githubusercontent.com/osx-cross/homebrew-avr/master/
######################################################################## 100.0%
==> Patching
==> Applying avr-binutils-size.patch
patching file binutils/size.c
==> ../configure --prefix=/usr/local/Cellar/avr-binutils/2.29 --infodir=/usr/loc
==> make
==> make install
🍺  /usr/local/Cellar/avr-binutils/2.29: 140 files, 12.2MB, built in 2 minutes 36 seconds
==> Installing osx-cross/avr/avr-gcc
==> Downloading https://gcc.gnu.org/pub/gcc/releases/gcc-7.2.0/gcc-7.2.0.tar.xz
######################################################################## 100.0%
==> ../configure --target=avr --prefix=/usr/local/Cellar/avr-gcc/7.2.0 --libdir=
==> make
==> make install
==> Downloading https://download.savannah.gnu.org/releases/avr-libc/avr-libc-2.0
==> Downloading from https://mirrors.up.pt/pub/nongnu/avr-libc/avr-libc-2.0.0.ta
######################################################################## 100.0%
==> ./configure --build=i686-apple-darwin17.3.0 --prefix=/usr/local/Cellar/avr-g
==> make install
🍺  /usr/local/Cellar/avr-gcc/7.2.0: 1,682 files, 175MB, built in 19 minutes 42 seconds
```

ladislas, 2018-01-19:
I've tried reinstalling, it works as well on my computer.

@KrzysztofKowalski without more information, it's hard to know what's going wrong. I'm closing the issue but feel free to reopen if the problem persists.

closed 2018-01-19 by ladislas