osx-cross/homebrew-avr #69: Xcode and homebrew-avr

werty37, 2018-01-16:
I would like to know how to use homebrew-avr with Xcode. Can you please list the steps please, and i can help in creating a wiki with those details.

ladislas, 2018-01-16:
what do you mean use it with Xcode?

werty37, 2018-01-16:
How do i use Xcode as the IDE with homebrew-avr so that i can code, compile, flash for avr mcus? I am sorry if this is not relevant; I thought it was possible with homebrew-avr.

ladislas, 2018-01-16:
they seem to already describe all the steps to make it work. 

https://github.com/jawher/xavr/#usage

you just need to install avr-gcc prior to running their python script.

I'm not an expert in Xcode, I use neovim to program in C/C++ and I have all the features of a good IDE, code completion and error included.

if you're familiar with vim/neovim, you can look here: https://github.com/ladislas/Bare-Arduino-Project

werty37, 2018-01-16:
@ladislas The x-avr project seems dead. But anyway i ll look into it. Thanks.

closed 2018-01-16 by werty37