osx-cross/homebrew-avr #70: Installation is very CPU intensive

evanlooij, 2018-01-23:
Today I installed homebrew-avr and was surprised when the fan on my MacBook started revving up. I checked Terminal and saw the title bar jumping between gcc and ruby with the fan whirring and the laptop practically shuddering -- very steampunk but I almost killed the installation convinced that it was stuck in an endless loop. 

In the end, the installation did succeed, taking almost half an hour. I've installed Crosspack-AVR with last week (and uninstalled it today) and neither process took more than ten minutes with minimal fuss. Is this an issue?

-------------Terminal------------------------------
> ==> Tapping osx-cross/avr
Cloning into '/usr/local/Homebrew/Library/Taps/osx-cross/homebrew-avr'...
remote: Counting objects: 14, done.
remote: Compressing objects: 100% (12/12), done.
remote: Total 14 (delta 4), reused 4 (delta 0), pack-reused 0
Unpacking objects: 100% (14/14), done.
Tapped 8 formulae (47 files, 66KB)
tirion:local elisevanlooij$ brew install avr-gcc
==> Installing avr-gcc from osx-cross/avr
==> Installing dependencies for osx-cross/avr/avr-gcc: gmp, mpfr, libmpc, avr-binutils
==> Installing osx-cross/avr/avr-gcc dependency: gmp
==> Downloading https://homebrew.bintray.com/bottles/gmp-6.1.2_1.high_sierra.bot
==> Downloading from https://akamai.bintray.com/ea/eadb377c507f5d04e8d47861fa764
######################################################################## 100.0%
==> Pouring gmp-6.1.2_1.high_sierra.bottle.tar.gz
🍺  /usr/local/Cellar/gmp/6.1.2_1: 18 files, 3.1MB
==> Installing osx-cross/avr/avr-gcc dependency: mpfr
==> Downloading https://homebrew.bintray.com/bottles/mpfr-4.0.0.high_sierra.bott
==> Downloading from https://akamai.bintray.com/e0/e09cdf149a648968e92fb36f88e00
######################################################################## 100.0%
==> Pouring mpfr-4.0.0.high_sierra.bottle.tar.gz
🍺  /usr/local/Cellar/mpfr/4.0.0: 28 files, 4.6MB
==> Installing osx-cross/avr/avr-gcc dependency: libmpc
==> Downloading https://homebrew.bintray.com/bottles/libmpc-1.1.0.high_sierra.bo
######################################################################## 100.0%
==> Pouring libmpc-1.1.0.high_sierra.bottle.tar.gz
🍺  /usr/local/Cellar/libmpc/1.1.0: 12 files, 353.8KB
==> Installing osx-cross/avr/avr-gcc dependency: avr-binutils
==> Downloading http://ftp.gnu.org/gnu/binutils/binutils-2.29.tar.bz2
######################################################################## 100.0%
==> Downloading https://raw.githubusercontent.com/osx-cross/homebrew-avr/master/
######################################################################## 100.0%
==> Patching
==> Applying avr-binutils-size.patch
patching file binutils/size.c
==> ../configure --prefix=/usr/local/Cellar/avr-binutils/2.29 --infodir=/usr/loc
==> make
==> make install
🍺  /usr/local/Cellar/avr-binutils/2.29: 140 files, 12.2MB, built in 2 minutes 29 seconds
==> Installing osx-cross/avr/avr-gcc
==> Downloading https://gcc.gnu.org/pub/gcc/releases/gcc-7.2.0/gcc-7.2.0.tar.xz
######################################################################## 100.0%
==> ../configure --target=avr --prefix=/usr/local/Cellar/avr-gcc/7.2.0 --libdir=
==> make
==> make install
==> Downloading https://download.savannah.gnu.org/releases/avr-libc/avr-libc-2.0
==> Downloading from https://mirrors.up.pt/pub/nongnu/avr-libc/avr-libc-2.0.0.ta
######################################################################## 100.0%
==> ./configure --build=i686-apple-darwin17.3.0 --prefix=/usr/local/Cellar/avr-g
==> make install
🍺  /usr/local/Cellar/avr-gcc/7.2.0: 1,682 files, 174.6MB, built in 24 minutes 57 seconds

leojrfs, 2018-01-23:
You should know that it is compiling a compiler. So it is CPU intensive 😀

leojrfs, 2018-01-23:
You can open a new issue if you think it should also be provided as binary (bottle or so).

closed 2018-01-23 by leojrfs