osx-cross/homebrew-avr #74: Fails to build on High Sierra

DagAgren, 2018-02-06:
Getting this when trying to install.

    $ brew install avr-gcc
    ==> Installing avr-gcc from osx-cross/avr
    ==> Installing dependencies for osx-cross/avr/avr-gcc: avr-binutils
    ==> Installing osx-cross/avr/avr-gcc dependency: avr-binutils
    ==> Downloading http://ftp.gnu.org/gnu/binutils/binutils-2.30.tar.bz2
    Already downloaded: /Users/dag/Library/Caches/Homebrew/avr-binutils-2.30.tar.bz2
    ==> Downloading https://raw.githubusercontent.com/osx-cross/homebrew-avr/master/
    Already downloaded: /Users/dag/Library/Caches/Homebrew/avr-binutils--patch-4748f87aee912f954be9968a5a01f61e4f1897adf21e2549d0ac988b9fe8ef1d.patch
    ==> Patching
    ==> Applying avr-binutils-size.patch
    patching file binutils/size.c
    ==> ../configure --prefix=/usr/local/Cellar/avr-binutils/2.30 --infodir=/usr/loc
    ==> make
    Last 15 lines from /Users/dag/Library/Logs/Homebrew/avr-binutils/02.make:
    clang -DHAVE_CONFIG_H -I. -I../../binutils  -I. -I../../binutils -I../bfd -I../../binutils/../bfd -I../../binutils/../include -DLOCALEDIR="\"/usr/local/Cellar/avr-binutils/2.30/share/locale\"" -Dbin_dummy_emulation=bin_vanilla_emulation  -W -Wall -Wstrict-prototypes -Wmissing-prototypes -Wshadow -I../../binutils/../zlib -g -O2 -MT is-strip.o -MD -MP -MF .deps/is-strip.Tpo -c -o is-strip.o ../../binutils/is-strip.c
    20 errors generated.
    make[4]: *** [readelf.o] Error 1
    make[4]: *** Waiting for unfinished jobs....
    mv -f .deps/wrstabs.Tpo .deps/wrstabs.Po
    mv -f .deps/elfedit.Tpo .deps/elfedit.Po
    mv -f .deps/is-strip.Tpo .deps/is-strip.Po
    mv -f .deps/nm.Tpo .deps/nm.Po
    mv -f .deps/objcopy.Tpo .deps/objcopy.Po
    mv -f .deps/dwarf.Tpo .deps/dwarf.Po
    mv -f .deps/ieee.Tpo .deps/ieee.Po
    make[3]: *** [all-recursive] Error 1
    make[2]: *** [all] Error 2
    make[1]: *** [all-binutils] Error 2
    make: *** [all] Error 2
    
    If reporting this issue please do so at (not Homebrew/brew or Homebrew/core):
    https://github.com/osx-cross/homebrew-avr/issues

(Why does this not include binaries? Building this beast can easily take an hour. This is a huge waste of resources, and providing binaries would also avoid mystery errors like this.)

ladislas, 2018-02-06:
Seems related to #68 -- but I don't have the issue on High Sierra myself so it's a bit hard to help.

try running the following:

```sh
$ brew remove avr-gcc avr-binutils
$ brew untap osx-cross/avr
$ brew cleanup -scf
$ brew tap osx-cross/avr
$ brew install avr-gcc
```

ladislas, 2018-02-06:
and regarding including binaries, I'm not aware of a simple way to do so for taps.

feel free to open a pull request to add that feature :)

DagAgren, 2018-02-06:
No change. I don't think it is related to that issue, the error seems significantly different, but homebrew unhelpfully cuts off all the useful parts of the error.

ladislas, 2018-02-06:
try installing with `--verbose`

DagAgren, 2018-02-07:
This seems to be the error, repeated quite a few times:
    ../../binutils/readelf.c:9250:10: error: incomplete definition of type 'struct tm'
                        tmp->tm_year + 1900, tmp->tm_mon + 1, tmp->tm_mday,
                        ~~~^
    /usr/include/secure/_stdio.h:57:62: note: expanded from macro 'snprintf'
      __builtin___snprintf_chk (str, len, 0, __darwin_obsz(str), __VA_ARGS__)
                                                                 ^~~~~~~~~~~
    ../../binutils/readelf.c:9241:9: note: forward declaration of 'struct tm'
            struct tm * tmp;

ladislas, 2018-02-07:
can you provide us with the full output in a https://pastebin.com/

ladislas, 2018-02-07:
I was able to compile it myself
[image]

larsimmisch, 2018-02-07:
I'm on High Sierra, and had no problems.

@DagAgren Can you run `brew doctor` and check if there is something in `/usr/local` that maybe shouldn't be there? `struct tm` is normally defined in `time.h`.

maartenpaauw, 2018-02-07:
Just installed it (on High Sierra) without any problems.

DagAgren, 2018-02-07:
Thanks, `brew doctor` was exactly what was needed. ffmpeg has somehow decided to just dump all its headers in /usr/local/include, messing everything up.

closed 2018-02-07 by DagAgren

ladislas, 2019-06-05:
@DagAgren Good news, bottles (binaries) are now available :)