osx-cross/homebrew-avr #75: Not able to install avr-gcc@4 or avr-gcc@5

MCUdude, 2018-02-10:
Hi!
I'm not able to install avr-gcc@4 or avr-gcc@5 when using brew. Here's the error when i tried avr-gcc@5:

```
brew install osx-cross/avr/avr-gcc@5
==> Installing avr-gcc@5 from osx-cross/avr
==> Downloading ftp://gcc.gnu.org/pub/gcc/releases/gcc-5.4.0/gcc-5.4.0.tar.bz2
######################################################################## 100.0%
==> ../configure --target=avr --prefix=/usr/local/Cellar/avr-gcc@5/5.4.0 --libdi
==> make
Last 15 lines from /Users/Hans/Library/Logs/Homebrew/avr-gcc@5/02.make:
/Library/Developer/CommandLineTools/usr/include/c++/v1/functional:1398:2: error: no member named 'fancy_abort' in namespace 'std::__1'; did you mean simply 'fancy_abort'?
        _VSTD::abort();
        ^~~~~~~
/Library/Developer/CommandLineTools/usr/include/c++/v1/__config:392:15: note: expanded from macro '_VSTD'
#define _VSTD std::_LIBCPP_NAMESPACE
              ^
../../gcc/system.h:700:13: note: 'fancy_abort' declared here
extern void fancy_abort (const char *, int, const char *) ATTRIBUTE_NORETURN;
            ^
1 error generated.
make[2]: *** [auto-profile.o] Error 1
make[2]: *** Waiting for unfinished jobs....
rm gcc.pod
make[1]: *** [all-gcc] Error 2
make: *** [all] Error 2

If reporting this issue please do so at (not Homebrew/brew or Homebrew/core):
https://github.com/osx-cross/homebrew-avr/issues

```

The error looks to me identical to the one I get when trying to install avr-gcc@4. What's wrong?

ladislas, 2018-02-11:
@sslupsky had the same issue:

https://github.com/osx-cross/homebrew-avr/issues/51#issuecomment-364172149

downgrading to previous xcode versions seems to work.

ladislas, 2018-02-11:
@MCUdude try running `brew doctor`

Dsignmind, 2018-02-16:
@MCUdude @ladislas I am running into a similar problem. I don't have the ability to downgrade XCode as I also develop mobile apps that require iOS11. 
Getting the exact same error when I try to install avr-gcc@5 after following steps from [#62 ](https://github.com/osx-cross/homebrew-avr/issues/62)
Any help appreciated.

ladislas, 2018-02-16:
just fixed it.

quick question though: why not use avr-gcc 7.3?

closed 2018-02-16 by ladislas

Dsignmind, 2018-02-16:
@ladislas Thank you, trying now! 
To be honest I would, but I'm collaborating on a project that is using an older AVR++ library (pre gcc 6.0.0) and to do so would mean fixing/changing a lot of reinterpret_cast errors I am receiving from the existing code.

sslupsky, 2018-02-16:
@Dsignmind You can have multiple versions of Xcode installed.  You can download Xcode 8, rename it some something like Xcode8.3.3 (do not have any spaces in the name), move it to your Applications folder before you run it then run it.  Then open preferences and you can select which toolchain to use (Xcode 9 or Xcode 8).

I had a problem compiling avr-gcc 4.5.3 with avr-libc1.7.1.  Source code required a hack to make it compile.  I added the following to the brew formula:

inreplace "gcc/config/avr/driver-avr.c", "if (0 == argc)\n    return;", "if (0 == argc)\n    return NULL;"

commit 05ca290c (2018-02-16): fix build with Xcode9 - fix #75