osx-cross/homebrew-avr #76: Fails to Build on High Sierra

JacobJerrell, 2018-02-24:
Getting this when trying to install:

```
Last 15 lines from /Users/j792280/Library/Logs/Homebrew/avr-gcc/02.make:
checking for gethostbyname... yes

checking for exported symbols... yes
checking for library containing dlopen... none required
checking for -fPIC -shared... yes
checking for socketpair... yes
checking for select... yes
checking for fork... yes
configure: updating cache ./config.cache
configure: creating ./config.status
config.status: creating Makefile
config.status: creating cc1plugin-config.h
config.status: executing depfiles commands
config.status: executing libtool commands
make: *** [all] Error 2
```

I've run through the following troubleshooting as suggested in other similar issues:

Using ``brew doctor``, a lot of stuff was in the node folder regarding openssl, and v8. I moved the entire /usr/local/include/node folder to the desktop and ran:

```
$ brew remove avr-gcc avr-binutils
$ brew untap osx-cross/avr
$ brew cleanup -scf
$ brew upgrade
$ brew tap osx-cross/avr
$ brew install avr-gcc
```

Sorry for the Tarantino script, but the above error is what happened after the above troubleshooting. Any attempts to install prior to that resulted in the brew install hanging at ``make`` for a very very long time (I was warned that it would be at least 20 minutes. I'm on a 2017 MacBook Pro, and waited greater than an hour)

ladislas, 2018-02-25:
could you share the result of `brew doctor`?

If you don't have too many packages installed, you could try removing brew completely and starting from scratch. it happens to me sometimes and really help.

JacobJerrell, 2018-02-25:
I have a number of packages installed because I work for a software company... currently is not the time to uproot my system, but if all else fails I can wait until things settled down a bit and then go down that road... After putting the node folder back into the /usr/local/include directory, this is the result of ``brew doctor``

```
Warning: "config" scripts exist outside your system or Homebrew directories.
`./configure` scripts often look for *-config scripts to determine if
software packages are installed, and what additional flags to use when
compiling and linking.

Having additional scripts in your path can confuse software installed via
Homebrew if the config script overrides a system or Homebrew provided
script of the same name. We found the following "config" scripts:
  /anaconda3/bin/icu-config
  /anaconda3/bin/freetype-config
  /anaconda3/bin/xslt-config
  /anaconda3/bin/libpng16-config
  /anaconda3/bin/python3.6m-config
  /anaconda3/bin/libpng-config
  /anaconda3/bin/xml2-config
  /anaconda3/bin/python3-config
  /anaconda3/bin/curl-config
  /anaconda3/bin/ncursesw6-config
  /anaconda3/bin/pcre-config
  /anaconda3/bin/python3.6-config

Warning: Unbrewed header files were found in /usr/local/include.
If you didn't put them there on purpose they could cause problems when
building Homebrew formulae, and may need to be deleted.

Unexpected header files:
  /usr/local/include/node/android-ifaddrs.h
  /usr/local/include/node/libplatform/libplatform-export.h
  /usr/local/include/node/libplatform/libplatform.h
  /usr/local/include/node/libplatform/v8-tracing.h
  /usr/local/include/node/node.h
  /usr/local/include/node/node_api.h
  /usr/local/include/node/node_api_types.h
  /usr/local/include/node/node_buffer.h
  /usr/local/include/node/node_object_wrap.h
  /usr/local/include/node/node_version.h
  /usr/local/include/node/openssl/aes.h
  /usr/local/include/node/openssl/archs/BSD-x86/opensslconf.h
  /usr/local/include/node/openssl/archs/BSD-x86_64/opensslconf.h
  /usr/local/include/node/openssl/archs/VC-WIN32/opensslconf.h
  /usr/local/include/node/openssl/archs/VC-WIN64A/opensslconf.h
  /usr/local/include/node/openssl/archs/aix-gcc/opensslconf.h
  /usr/local/include/node/openssl/archs/aix64-gcc/opensslconf.h
  /usr/local/include/node/openssl/archs/darwin-i386-cc/opensslconf.h
  /usr/local/include/node/openssl/archs/darwin64-x86_64-cc/opensslconf.h
  /usr/local/include/node/openssl/archs/linux-aarch64/opensslconf.h
  /usr/local/include/node/openssl/archs/linux-armv4/opensslconf.h
  /usr/local/include/node/openssl/archs/linux-elf/opensslconf.h
  /usr/local/include/node/openssl/archs/linux-ppc/opensslconf.h
  /usr/local/include/node/openssl/archs/linux-ppc64/opensslconf.h
  /usr/local/include/node/openssl/archs/linux-x32/opensslconf.h
  /usr/local/include/node/openssl/archs/linux-x86_64/opensslconf.h
  /usr/local/include/node/openssl/archs/linux32-s390x/opensslconf.h
  /usr/local/include/node/openssl/archs/linux64-s390x/opensslconf.h
  /usr/local/include/node/openssl/archs/solaris-x86-gcc/opensslconf.h
  /usr/local/include/node/openssl/archs/solaris64-x86_64-gcc/opensslconf.h
  /usr/local/include/node/openssl/asn1.h
  /usr/local/include/node/openssl/asn1_mac.h
  /usr/local/include/node/openssl/asn1t.h
  /usr/local/include/node/openssl/bio.h
  /usr/local/include/node/openssl/blowfish.h
  /usr/local/include/node/openssl/bn.h
  /usr/local/include/node/openssl/buffer.h
  /usr/local/include/node/openssl/camellia.h
  /usr/local/include/node/openssl/cast.h
  /usr/local/include/node/openssl/cmac.h
  /usr/local/include/node/openssl/cms.h
[... 69 lines cut ...]
  /usr/local/include/node/uv-errno.h
  /usr/local/include/node/uv-linux.h
  /usr/local/include/node/uv-os390.h
  /usr/local/include/node/uv-posix.h
  /usr/local/include/node/uv-sunos.h
  /usr/local/include/node/uv-threadpool.h
  /usr/local/include/node/uv-unix.h
  /usr/local/include/node/uv-version.h
  /usr/local/include/node/uv-win.h
  /usr/local/include/node/uv.h
  /usr/local/include/node/v8-debug.h
  /usr/local/include/node/v8-inspector-protocol.h
  /usr/local/include/node/v8-inspector.h
  /usr/local/include/node/v8-platform.h
  /usr/local/include/node/v8-profiler.h
  /usr/local/include/node/v8-testing.h
  /usr/local/include/node/v8-util.h
  /usr/local/include/node/v8-value-serializer-version.h
  /usr/local/include/node/v8-version-string.h
  /usr/local/include/node/v8-version.h
  /usr/local/include/node/v8.h
  /usr/local/include/node/v8config.h
  /usr/local/include/node/zconf.h
  /usr/local/include/node/zlib.h
```

ladislas, 2018-02-26:
I've tried re-installing it myself on High Sierra and it works perfectly.

I'm not sure what's causing the issue as the log is not very useful...

trying `brew remove anaconda node` might help, but i'm not sure.

as said earlier, removing hombrew and resinstalling it might fix it.
(quick tip: take a screenshot of `brew list` so you know which package you need to reinstall)

JacobJerrell, 2018-03-08:
I blew out homebrew and reinstalled. The first thing I tried to install was avr-gcc, and this is the result:

```==> ../configure --target=avr --prefix=/usr/local/Cellar/avr-gcc/7.3.0 --libdir=/usr/local/Cellar/avr-gcc/7.3.0/lib/avr-gcc/7
==> make
Last 15 lines from /Users/myUsername/Library/Logs/Homebrew/avr-gcc/02.make:
/private/tmp/avr-gcc-20180307-69759-6u86j8/gcc-7.3.0/build/./gcc/xgcc -B/private/tmp/avr-gcc-20180307-69759-6u86j8/gcc-7.3.0/build/./gcc/ -B/usr/local/Cellar/avr-gcc/7.3.0/avr/bin/ -B/usr/local/Cellar/avr-gcc/7.3.0/avr/lib/ -isystem /usr/local/Cellar/avr-gcc/7.3.0/avr/include -isystem /usr/local/Cellar/avr-gcc/7.3.0/avr/sys-include    -g -O2 -mmcu=avr31 -O2  -g -O2 -DIN_GCC  -DCROSS_DIRECTORY_STRUCTURE  -W -Wall -Wno-narrowing -Wwrite-strings -Wcast-qual -Wstrict-prototypes -Wmissing-prototypes -Wold-style-definition  -isystem ./include   -DDF=SF -Dinhibit_libc -mcall-prologues -Os -g -DIN_LIBGCC2 -fbuilding-libgcc -fno-stack-protector -Dinhibit_libc  -DDF=SF -Dinhibit_libc -mcall-prologues -Os -I. -I. -I../../.././gcc -I../../../../libgcc -I../../../../libgcc/. -I../../../../libgcc/../gcc -I../../../../libgcc/../include  -DHAVE_CC_TLS -DUSE_EMUTLS -o _satfractSQUSQ.o -MT _satfractSQUSQ.o -MD -MP -MF _satfractSQUSQ.dep -DL_satfract -DFROM_SQ -DTO_USQ -c ../../../../libgcc/fixed-bit.c -fvisibility=hidden -DHIDE_EXPORTS
/private/tmp/avr-gcc-20180307-69759-6u86j8/gcc-7.3.0/build/./gcc/xgcc -B/private/tmp/avr-gcc-20180307-69759-6u86j8/gcc-7.3.0/build/./gcc/ -B/usr/local/Cellar/avr-gcc/7.3.0/avr/bin/ -B/usr/local/Cellar/avr-gcc/7.3.0/avr/lib/ -isystem /usr/local/Cellar/avr-gcc/7.3.0/avr/include -isystem /usr/local/Cellar/avr-gcc/7.3.0/avr/sys-include    -g -O2 -mmcu=avr31 -O2  -g -O2 -DIN_GCC  -DCROSS_DIRECTORY_STRUCTURE  -W -Wall -Wno-narrowing -Wwrite-strings -Wcast-qual -Wstrict-prototypes -Wmissing-prototypes -Wold-style-definition  -isystem ./include   -DDF=SF -Dinhibit_libc -mcall-prologues -Os -g -DIN_LIBGCC2 -fbuilding-libgcc -fno-stack-protector -Dinhibit_libc  -DDF=SF -Dinhibit_libc -mcall-prologues -Os -I. -I. -I../../.././gcc -I../../../../libgcc -I../../../../libgcc/. -I../../../../libgcc/../gcc -I../../../../libgcc/../include  -DHAVE_CC_TLS -DUSE_EMUTLS -o _satfractSQUDQ.o -MT _satfractSQUDQ.o -MD -MP -MF _satfractSQUDQ.dep -DL_satfract -DFROM_SQ -DTO_UDQ -c ../../../../libgcc/fixed-bit.c -fvisibility=hidden -DHIDE_EXPORTS
/private/tmp/avr-gcc-20180307-69759-6u86j8/gcc-7.3.0/build/./gcc/xgcc -B/private/tmp/avr-gcc-20180307-69759-6u86j8/gcc-7.3.0/build/./gcc/ -B/usr/local/Cellar/avr-gcc/7.3.0/avr/bin/ -B/usr/local/Cellar/avr-gcc/7.3.0/avr/lib/ -isystem /usr/local/Cellar/avr-gcc/7.3.0/avr/include -isystem /usr/local/Cellar/avr-gcc/7.3.0/avr/sys-include    -g -O2 -mmcu=avr31 -O2  -g -O2 -DIN_GCC  -DCROSS_DIRECTORY_STRUCTURE  -W -Wall -Wno-narrowing -Wwrite-strings -Wcast-qual -Wstrict-prototypes -Wmissing-prototypes -Wold-style-definition  -isystem ./include   -DDF=SF -Dinhibit_libc -mcall-prologues -Os -g -DIN_LIBGCC2 -fbuilding-libgcc -fno-stack-protector -Dinhibit_libc  -DDF=SF -Dinhibit_libc -mcall-prologues -Os -I. -I. -I../../.././gcc -I../../../../libgcc -I../../../../libgcc/. -I../../../../libgcc/../gcc -I../../../../libgcc/../include  -DHAVE_CC_TLS -DUSE_EMUTLS -o _satfractSQUTQ.o -MT _satfractSQUTQ.o -MD -MP -MF _satfractSQUTQ.dep -DL_satfract -DFROM_SQ -DTO_UTQ -c ../../../../libgcc/fixed-bit.c -fvisibility=hidden -DHIDE_EXPORTS
/private/tmp/avr-gcc-20180307-69759-6u86j8/gcc-7.3.0/build/./gcc/xgcc -B/private/tmp/avr-gcc-20180307-69759-6u86j8/gcc-7.3.0/build/./gcc/ -B/usr/local/Cellar/avr-gcc/7.3.0/avr/bin/ -B/usr/local/Cellar/avr-gcc/7.3.0/avr/lib/ -isystem /usr/local/Cellar/avr-gcc/7.3.0/avr/include -isystem /usr/local/Cellar/avr-gcc/7.3.0/avr/sys-include    -g -O2 -mmcu=avr31 -O2  -g -O2 -DIN_GCC  -DCROSS_DIRECTORY_STRUCTURE  -W -Wall -Wno-narrowing -Wwrite-strings -Wcast-qual -Wstrict-prototypes -Wmissing-prototypes -Wold-style-definition  -isystem ./include   -DDF=SF -Dinhibit_libc -mcall-prologues -Os -g -DIN_LIBGCC2 -fbuilding-libgcc -fno-stack-protector -Dinhibit_libc  -DDF=SF -Dinhibit_libc -mcall-prologues -Os -I. -I. -I../../.././gcc -I../../../../libgcc -I../../../../libgcc/. -I../../../../libgcc/../gcc -I../../../../libgcc/../include  -DHAVE_CC_TLS -DUSE_EMUTLS -o _satfractSQUHA.o -MT _satfractSQUHA.o -MD -MP -MF _satfractSQUHA.dep -DL_satfract -DFROM_SQ -DTO_UHA -c ../../../../libgcc/fixed-bit.c -fvisibility=hidden -DHIDE_EXPORTS
/private/tmp/avr-gcc-20180307-69759-6u86j8/gcc-7.3.0/build/./gcc/xgcc -B/private/tmp/avr-gcc-20180307-69759-6u86j8/gcc-7.3.0/build/./gcc/ -B/usr/local/Cellar/avr-gcc/7.3.0/avr/bin/ -B/usr/local/Cellar/avr-gcc/7.3.0/avr/lib/ -isystem /usr/local/Cellar/avr-gcc/7.3.0/avr/include -isystem /usr/local/Cellar/avr-gcc/7.3.0/avr/sys-include    -g -O2 -mmcu=avr31 -O2  -g -O2 -DIN_GCC  -DCROSS_DIRECTORY_STRUCTURE  -W -Wall -Wno-narrowing -Wwrite-strings -Wcast-qual -Wstrict-prototypes -Wmissing-prototypes -Wold-style-definition  -isystem ./include   -DDF=SF -Dinhibit_libc -mcall-prologues -Os -g -DIN_LIBGCC2 -fbuilding-libgcc -fno-stack-protector -Dinhibit_libc  -DDF=SF -Dinhibit_libc -mcall-prologues -Os -I. -I. -I../../.././gcc -I../../../../libgcc -I../../../../libgcc/. -I../../../../libgcc/../gcc -I../../../../libgcc/../include  -DHAVE_CC_TLS -DUSE_EMUTLS -o _satfractSQUSA.o -MT _satfractSQUSA.o -MD -MP -MF _satfractSQUSA.dep -DL_satfract -DFROM_SQ -DTO_USA -c ../../../../libgcc/fixed-bit.c -fvisibility=hidden -DHIDE_EXPORTS
[... cut ...] QQ.o -MT _satfractDQQQ.o -MD -MP -MF _satfractDQQQ.dep -DL_satfract -DFROM_DQ -DTO_QQ -c ../../../../libgcc/fixed-bit.c -fvisibility=hidden -DHIDE_EXPORTS
make[4]: *** [_satfractSQUTA.o] Error 1
make[4]: *** Waiting for unfinished jobs....
make[3]: *** [multi-do] Error 1
make[2]: *** [all-multi] Error 2
make[1]: *** [all-target-libgcc] Error 2
make: *** [all] Error 2

If reporting this issue please do so at (not Homebrew/brew or Homebrew/core):
https://github.com/osx-cross/homebrew-avr/issues```

ladislas, 2018-03-08:
can you run `brew install avr-gcc --verbose` and provide me with the full output?

are you running the latest version of XCode and/or command line tools?

JacobJerrell, 2018-03-12:
Sorry it's taking so long to get back...

[--verbose output (last 129 lines because of terminal memory limit)](https://pastebin.com/tdmcsmTD)

00.options.out:
`osx-cross/avr/avr-gcc`

[01.configure](https://pastebin.com/pBAJq9QQ)
[01.configure.cc](https://pastebin.com/aht1j0kM)

02.make and 02.make.cc are too large for pastebin... email?

ladislas, 2018-03-12:
Honestly I don't know what could go wrong... 

@leojrfs @larsimmisch any ideas?

JacobJerrell, 2018-03-20:
I'm on a managed device -- company provided. I have admin access and aside from some nuances that seem to affect other High Sierra users, I haven't had any issues. Thought I might put that out there since it seems to be throwing everyone for a loop.

JacobJerrell, 2018-03-26:
I resolved the issue by skipping homebrew and compiling the everything locally.

1. Installed binutils 
1.a. ./configure --target=avr
2. Installed gcc
2.a. $PWD/../gcc-7.3.0/configure --prefix=/opt/gcc-7.3.0 --without-headers --with-newlib --target=avr --enable-languages=c --disable-libssp
2.b. The configure command that worked for me is a combination of a couple fixes I found on other discussion boards.
2.c. make (wait.... for a very long time)
2.d. make install

Seems like the big fixes were `--without-headers` and `--disable-libssp`. This solution is a combination of various install walkthrough's and issue resolutions found on the web. Maybe it will help someone else

closed 2018-03-26 by JacobJerrell