osx-cross/homebrew-avr #77: avr-gcc@7 will not build on High Sierra (again!)

fauxpark, 2018-06-30:
I see there are a few previous issues similar to mine, but none of the solutions in them (if any) have helped me.

I need to stay on avr-gcc version 7 for now as the QMK keyboard firmware and/or LUFA apparently have some issues with the new version at the moment. Now avr-gcc itself won't build when I attempt to install it.

[Here is the output of 02.make](https://pastebin.com/rw7WteZg). Lines 3252-3254 seem to be where the action is.

`brew doctor` didn't find anything. I tried untap/tap, cleanup, deleting cached downloads etc, no dice. Installing Homebrew onto an older Macbook Pro that hadn't had its Xcode command line tools updated in a while gave the same result before and after updating (so it doesn't appear to be the culprit).

ladislas, 2018-06-30:
you should be good now.

closed 2018-06-30 by ladislas

fauxpark, 2018-06-30:
Thanks, it is indeed building now.

commit 3f4e52ca (2018-06-30): don't build documentation - fix #77