osx-cross/homebrew-avr #78: brew install avr-gcc fails

maxshaffer, 2018-07-24:
$ brew install avr-gcc
Updating Homebrew…
==> Auto-updated Homebrew!
Updated 1 tap (homebrew/cask).
No changes to formulae.

==> Installing avr-gcc from osx-cross/avr
==> Installing dependencies for osx-cross/avr/avr-gcc: avr-binutils
==> Installing osx-cross/avr/avr-gcc dependency: avr-binutils
==> Downloading http://ftp.gnu.org/gnu/binutils/binutils-2.30.tar.bz2
Already downloaded: /Users/max/Library/Caches/Homebrew/avr-binutils-2.30.tar.bz2
==> Downloading https://raw.githubusercontent.com/osx-cross/homebrew-avr/master/avr-binutils-size.patch
Already downloaded: /Users/max/Library/Caches/Homebrew/avr-binutils--patch-4748f87aee912f954be9968a5a01f61e4f1897adf21e2549d0ac988b9fe8ef1d.patch
==> Patching
==> Applying avr-binutils-size.patch
patching file binutils/size.c
==> ../configure --prefix=/usr/local/Cellar/avr-binutils/2.30 --infodir=/usr/local/Cellar/avr-binutils/2.30/share/info --mandir=/usr/local/Cellar/avr-binutils/2.30/share/man --target=avr --disable-nls --disable-werror
==> make
==> make install
🍺  /usr/local/Cellar/avr-binutils/2.30: 158 files, 12.5MB, built in 5 minutes 16 seconds
==> Installing osx-cross/avr/avr-gcc
==> Downloading https://gcc.gnu.org/pub/gcc/releases/gcc-8.1.0/gcc-8.1.0.tar.xz
Already downloaded: /Users/max/Library/Caches/Homebrew/avr-gcc-8.1.0.tar.xz
==> ../configure --target=avr --prefix=/usr/local/Cellar/avr-gcc/8.1.0 --libdir=/usr/local/Cellar/avr-gcc/8.1.0/lib/avr-gcc/8 --enable-languages=c,c++ --with-ld=/usr/local/opt/avr-binutils/bin/avr-ld --with-as=/usr/local/opt/avr-binutils/bin/avr-as --disable-nls --disable
==> make
Last 15 lines from /Users/max/Library/Logs/Homebrew/avr-gcc/02.make:
/private/tmp/avr-gcc-20180723-19951-ub9qsk/gcc-8.1.0/build/./gcc/nm -pg _udivmodhi4_s.o | awk 'NF == 3 && $2 !~ /^[UN]$/ && $3 !~ /.*_compat/ && $3 !~ /.*@.*/ { print "\t.hidden", $3 }' > _udivmodhi4.visT
/private/tmp/avr-gcc-20180723-19951-ub9qsk/gcc-8.1.0/build/./gcc/nm -pg _divmodhi4_s.o | awk 'NF == 3 && $2 !~ /^[UN]$/ && $3 !~ /.*_compat/ && $3 !~ /.*@.*/ { print "\t.hidden", $3 }' > _divmodhi4.visT
/private/tmp/avr-gcc-20180723-19951-ub9qsk/gcc-8.1.0/build/./gcc/nm -pg _divmodpsi4_s.o | awk 'NF == 3 && $2 !~ /^[UN]$/ && $3 !~ /.*_compat/ && $3 !~ /.*@.*/ { print "\t.hidden", $3 }' > _divmodpsi4.visT
/private/tmp/avr-gcc-20180723-19951-ub9qsk/gcc-8.1.0/build/./gcc/nm -pg _udivmodpsi4_s.o | awk 'NF == 3 && $2 !~ /^[UN]$/ && $3 !~ /.*_compat/ && $3 !~ /.*@.*/ { print "\t.hidden", $3 }' > _udivmodpsi4.visT
mv -f _divmodhi4.visT _divmodhi4.vis
make[4]: mv: Bad executable (or shared library)
mv -f _udivmodpsi4.visT _udivmodpsi4.vis
make[4]: *** [_divmodhi4.vis] Error 1
make[4]: *** Waiting for unfinished jobs....
mv -f _divmodpsi4.visT _divmodpsi4.vis
mv -f _udivmodhi4.visT _udivmodhi4.vis
make[3]: *** [multi-do] Error 1
make[2]: *** [all-multi] Error 2
make[1]: *** [all-target-libgcc] Error 2
make: *** [all] Error 2

If reporting this issue please do so at (not Homebrew/brew or Homebrew/core):
https://github.com/osx-cross/homebrew-avr/issues

ladislas, 2018-07-25:
Thanks @maxshaffer for filling an issue.

But without more information, I can't help.

Please use `--verbose` when installing and share the complete output through [pastbin](https://pastebin.com/).

Please also share your system information and the list of formulae already installed.

Have you tried running `brew update && brew upgrade` first? What is the output of `brew doctor`?

try uninstalling everything and reinstalling as explained here: https://github.com/osx-cross/homebrew-avr/issues/51#issuecomment-299814905

maxshaffer, 2018-07-30:
Yes, I ran brew update && brew upgrade. I am up to date. brew doctor returns "Your system is ready to brew". I ran brew install avr-gcc --verbose, but the installation took over 12 hours and I finally had to cancel it prematurely.

ladislas, 2018-08-04:
I still need info about your system. I can help you with what you shared. I tried uninstalling and recompiling and it works well on my machine.

The 12 hours installation is clearly not a good sign. It usually takes between 30min to 1 hour to compile GCC.

Closing the issue now, feel free to reopen if you have new updates.

closed 2018-08-04 by ladislas