osx-cross/homebrew-avr #79: brew install avr-gcc fails on High Sierra

daikiojm, 2018-08-05:
I attempted to install avr-gcc with brew, but the following error occurs and it fails.
looked at past Issue and tried brew's update and clean, but the situation did not change.

## Install result

Getting this when trying to install:

```
$ brew install osx-cross/avr/avr-gcc
==> Installing avr-gcc from osx-cross/avr
==> Downloading https://gcc.gnu.org/pub/gcc/releases/gcc-8.1.0/gcc-8.1.0.tar.xz
Already downloaded: /Users/daikiojm/Library/Caches/Homebrew/avr-gcc-8.1.0.tar.xz
==> ../configure --target=avr --prefix=/usr/local/Cellar/avr-gcc/8.1.0 --libdir=/usr/local/Cellar/avr-gcc/8.1.0/lib/avr-gcc/8 --enable-languages=c,c++ --with-ld=/usr/local/opt/avr-binuti
==> make
Last 15 lines from /Users/daikiojm/Library/Logs/Homebrew/avr-gcc/02.make:
/usr/local/include/isl/map.h:496:21: note: 'isl_map_domain' declared here
__isl_give isl_set *isl_map_domain(__isl_take isl_map *bmap);
                    ^
../../gcc/graphite-sese-to-poly.c:524:26: error: cannot initialize a parameter of type 'isl_map *' with an rvalue of type 'isl_space *'
                            isl_space_domain (isl_map_get_space (acc)));
                                              ^~~~~~~~~~~~~~~~~~~~~~~
/usr/local/include/isl/map.h:496:56: note: passing argument to parameter 'bmap' here
__isl_give isl_set *isl_map_domain(__isl_take isl_map *bmap);
                                                       ^
fatal error: too many errors emitted, stopping now [-ferror-limit=]
20 errors generated.
make[2]: *** [graphite-sese-to-poly.o] Error 1
rm gcc.pod
make[1]: *** [all-gcc] Error 2
make: *** [all] Error 2

If reporting this issue please do so at (not Homebrew/brew or Homebrew/core):
https://github.com/osx-cross/homebrew-avr/issues
```

There is output with `--verbose` 

https://pastebin.com/91JGqU0k

## Environment

My environment confirmed with the `brew config`

```
$ brew config
HOMEBREW_VERSION: 1.7.1
ORIGIN: https://github.com/Homebrew/brew
HEAD: d6a245c05110c240e680da594c2cc6038998b82e
Last commit: 13 days ago
Core tap ORIGIN: https://github.com/Homebrew/homebrew-core
Core tap HEAD: 2792778ca306e223ca1b3cf32e6891190ab1370f
Core tap last commit: 18 hours ago
HOMEBREW_PREFIX: /usr/local
CPU: octa-core 64-bit kabylake
Homebrew Ruby: 2.3.7 => /System/Library/Frameworks/Ruby.framework/Versions/2.3/usr/bin/ruby
Clang: 9.1 build 902
Git: 2.18.0 => /usr/local/bin/git
Curl: 7.54.0 => /usr/bin/curl
Java: N/A
macOS: 10.13.6-x86_64
CLT: 9.4.1.0.1.1528165917
Xcode: 9.4.1
XQuartz: N/A
```

ladislas, 2018-08-10:
thanks @joshuata for pointing that out! I'm going to look into it.

The only thing that surprises me is that we are still on GCC 8.1.0 and we don't depend on ISL...

ladislas, 2018-08-14:
@joshuata let me know if it works now. it does on my machine.

closed 2018-08-14 by ladislas

joshuata, 2018-08-14:
@ladislas It's working! Thanks!

commit a2360ef7 (2018-08-14): update to gcc 8.2.0 - fix #79