osx-cross/homebrew-avr #80: Can't build avr-gcc@7 on High Sierra

henrikjonhed, 2018-08-15:
Hi, I cant build avr-gcc@7 om high sierra, 02.make looks like this in the end:

`
...

clang++ -fno-PIE -c   -g -O2 -DIN_GCC  -DCROSS_DIRECTORY_STRUCTURE  -fno-strict-aliasing -fno-exceptions -fno-rtti -fasynchronous-unwind-tables -W -Wall -Wno-narrowing -Wwrite-strings -Wcast-qual -Wmissing-format-attribute -Woverloaded-virtual -pedantic -Wno-long-long -Wno-variadic-macros -Wno-overlength-strings   -DHAVE_CONFIG_H -I. -I. -I../../gcc -I../../gcc/. -I../../gcc/../include -I../../gcc/../libcpp/include  -I../../gcc/../libdecnumber -I../../gcc/../libdecnumber/dpd -I../libdecnumber -I../../gcc/../libbacktrace   -o graphite-dependences.o -MT graphite-dependences.o -MMD -MP -MF ./.deps/graphite-dependences.TPo ../../gcc/graphite-dependences.c
clang++ -fno-PIE -c   -g -O2 -DIN_GCC  -DCROSS_DIRECTORY_STRUCTURE  -fno-strict-aliasing -fno-exceptions -fno-rtti -fasynchronous-unwind-tables -W -Wall -Wno-narrowing -Wwrite-strings -Wcast-qual -Wmissing-format-attribute -Woverloaded-virtual -pedantic -Wno-long-long -Wno-variadic-macros -Wno-overlength-strings   -DHAVE_CONFIG_H -I. -I. -I../../gcc -I../../gcc/. -I../../gcc/../include -I../../gcc/../libcpp/include  -I../../gcc/../libdecnumber -I../../gcc/../libdecnumber/dpd -I../libdecnumber -I../../gcc/../libbacktrace   -o graphite-optimize-isl.o -MT graphite-optimize-isl.o -MMD -MP -MF ./.deps/graphite-optimize-isl.TPo ../../gcc/graphite-optimize-isl.c
clang++ -fno-PIE -c   -g -O2 -DIN_GCC  -DCROSS_DIRECTORY_STRUCTURE  -fno-strict-aliasing -fno-exceptions -fno-rtti -fasynchronous-unwind-tables -W -Wall -Wno-narrowing -Wwrite-strings -Wcast-qual -Wmissing-format-attribute -Woverloaded-virtual -pedantic -Wno-long-long -Wno-variadic-macros -Wno-overlength-strings   -DHAVE_CONFIG_H -I. -I. -I../../gcc -I../../gcc/. -I../../gcc/../include -I../../gcc/../libcpp/include  -I../../gcc/../libdecnumber -I../../gcc/../libdecnumber/dpd -I../libdecnumber -I../../gcc/../libbacktrace   -o graphite-poly.o -MT graphite-poly.o -MMD -MP -MF ./.deps/graphite-poly.TPo ../../gcc/graphite-poly.c
clang++ -fno-PIE -c   -g -O2 -DIN_GCC  -DCROSS_DIRECTORY_STRUCTURE  -fno-strict-aliasing -fno-exceptions -fno-rtti -fasynchronous-unwind-tables -W -Wall -Wno-narrowing -Wwrite-strings -Wcast-qual -Wmissing-format-attribute -Woverloaded-virtual -pedantic -Wno-long-long -Wno-variadic-macros -Wno-overlength-strings   -DHAVE_CONFIG_H -I. -I. -I../../gcc -I../../gcc/. -I../../gcc/../include -I../../gcc/../libcpp/include  -I../../gcc/../libdecnumber -I../../gcc/../libdecnumber/dpd -I../libdecnumber -I../../gcc/../libbacktrace   -o graphite-scop-detection.o -MT graphite-scop-detection.o -MMD -MP -MF ./.deps/graphite-scop-detection.TPo ../../gcc/graphite-scop-detection.c
clang++ -fno-PIE -c   -g -O2 -DIN_GCC  -DCROSS_DIRECTORY_STRUCTURE  -fno-strict-aliasing -fno-exceptions -fno-rtti -fasynchronous-unwind-tables -W -Wall -Wno-narrowing -Wwrite-strings -Wcast-qual -Wmissing-format-attribute -Woverloaded-virtual -pedantic -Wno-long-long -Wno-variadic-macros -Wno-overlength-strings   -DHAVE_CONFIG_H -I. -I. -I../../gcc -I../../gcc/. -I../../gcc/../include -I../../gcc/../libcpp/include  -I../../gcc/../libdecnumber -I../../gcc/../libdecnumber/dpd -I../libdecnumber -I../../gcc/../libbacktrace   -o graphite-sese-to-poly.o -MT graphite-sese-to-poly.o -MMD -MP -MF ./.deps/graphite-sese-to-poly.TPo ../../gcc/graphite-sese-to-poly.c
../../gcc/graphite-optimize-isl.c:57:19: error: use of undeclared identifier 'isl_space_dim'
  unsigned dims = isl_space_dim (space, isl_dim_set);
                  ^
../../gcc/graphite-dependences.c:48:16: error: use of undeclared identifier 'isl_space_get_tuple_id'; did you mean 'isl_map_get_tuple_id'?
  isl_id *id = isl_space_get_tuple_id (d, isl_dim_in);
               ^~~~~~~~~~~~~~~~~~~~~~
               isl_map_get_tuple_id
/usr/local/include/isl/map.h:102:20: note: 'isl_map_get_tuple_id' declared here
__isl_give isl_id *isl_map_get_tuple_id(__isl_keep isl_map *map,
                   ^
../../gcc/graphite-dependences.c:48:40: error: cannot initialize a parameter of type 'isl_map *' with an lvalue of type 'isl_space *'
  isl_id *id = isl_space_get_tuple_id (d, isl_dim_in);
                                       ^
/usr/local/include/isl/map.h:102:61: note: passing argument to parameter 'map' here
__isl_give isl_id *isl_map_get_tuple_id(__isl_keep isl_map *map,
                                                            ^
../../gcc/graphite-optimize-isl.c:60:3: error: use of undeclared identifier 'isl_space_free'
  isl_space_free (space);
  ^
../../gcc/graphite-dependences.c:51:3: error: use of undeclared identifier 'isl_space_free'
  isl_space_free (d);
  ^
../../gcc/graphite-dependences.c:179:16: error: use of undeclared identifier 'isl_space_dim'
  int nb_out = isl_space_dim (space, isl_dim_out);
               ^
2 errors generated.
make[2]: *** [graphite-optimize-isl.o] Error 1
make[2]: *** Waiting for unfinished jobs....
../../gcc/graphite-dependences.c:185:3: error: use of undeclared identifier 'isl_space_free'
  isl_space_free (space);
  ^
../../gcc/graphite-dependences.c:195:11: error: use of undeclared identifier 'isl_space_dim'
  int n = isl_space_dim (space, isl_dim_out);
          ^
../../gcc/graphite-dependences.c:197:3: error: use of undeclared identifier 'isl_space_free'
  isl_space_free (space);
  ^
../../gcc/graphite-dependences.c:272:34: error: use of undeclared identifier 'isl_space_range'; did you mean 'isl_map_range'?
  isl_map *lex = isl_map_lex_le (isl_space_range (space));
                                 ^~~~~~~~~~~~~~~
                                 isl_map_range
/usr/local/include/isl/map.h:497:21: note: 'isl_map_range' declared here
__isl_give isl_set *isl_map_range(__isl_take isl_map *map);
                    ^
../../gcc/graphite-dependences.c:272:51: error: cannot initialize a parameter of type 'isl_map *' with an lvalue of type 'isl_space *'
  isl_map *lex = isl_map_lex_le (isl_space_range (space));
                                                  ^~~~~
/usr/local/include/isl/map.h:497:55: note: passing argument to parameter 'map' here
__isl_give isl_set *isl_map_range(__isl_take isl_map *map);
                                                      ^
../../gcc/graphite-isl-ast-to-gimple.c:98:7: error: use of undeclared identifier 'isl_id_free'
      isl_id_free (it->first);
      ^
../../gcc/graphite-isl-ast-to-gimple.c:293:3: error: use of undeclared identifier 'isl_id_free'
  isl_id_free (tmp_isl_id);
  ^
9 errors generated.
make[2]: *** [graphite-dependences.o] Error 1
../../gcc/graphite-isl-ast-to-gimple.c:629:5: error: use of undeclared identifier 'isl_id_free'
    isl_id_free (res->first);
[... 132 lines cut ...]
Core tap last commit: 3 hours ago
HOMEBREW_PREFIX: /usr/local
HOMEBREW_CACHE: /Users/henrikjonhed/Library/Caches/Homebrew
CPU: octa-core 64-bit haswell
Homebrew Ruby: 2.3.7 => /System/Library/Frameworks/Ruby.framework/Versions/2.3/usr/bin/ruby
Clang: 9.1 build 902
Git: 2.17.0 => /usr/local/bin/git
Curl: 7.54.0 => /usr/bin/curl
Java: 1.8.0_102
macOS: 10.13.6-x86_64
CLT: 9.4.1.0.1.1528165917
Xcode: 9.4.1
XQuartz: 2.7.9 => /opt/X11

HOMEBREW_CC: clang
HOMEBREW_CXX: clang++
MAKEFLAGS: -j8
CMAKE_PREFIX_PATH: /usr/local
CMAKE_INCLUDE_PATH: /usr/include/libxml2:/System/Library/Frameworks/OpenGL.framework/Versions/Current/Headers
CMAKE_LIBRARY_PATH: /System/Library/Frameworks/OpenGL.framework/Versions/Current/Libraries
PKG_CONFIG_PATH: /usr/local/opt/mpfr/lib/pkgconfig
PKG_CONFIG_LIBDIR: /usr/lib/pkgconfig:/usr/local/Homebrew/Library/Homebrew/os/mac/pkgconfig/10.13
ACLOCAL_PATH: /usr/local/share/aclocal
PATH: /usr/local/Homebrew/Library/Homebrew/shims/mac/super:/usr/local/opt/avr-binutils/bin:/usr/bin:/bin:/usr/sbin:/sbin
`

ladislas, 2018-08-15:
@henrikjonhed it should be working now, let me know.

closed 2018-08-15 by ladislas

henrikjonhed, 2018-08-15:
Works now! Thanks for the fast fix 🙌

commit ef705e99 (2018-08-15): add isl patch to fix build - fix #80