osx-cross/homebrew-avr #82: simavr fail to install with last version of macos sierra

patrickelectric, 2018-08-24:
```
brew tap osx-cross/avr
brew install --HEAD simavr
==> Installing simavr from osx-cross/avr
==> Cloning https://github.com/buserror/simavr.git
Updating /Users/patrick/Library/Caches/Homebrew/simavr--git
==> Checking out branch master
Already on 'master'
Your branch is up to date with 'origin/master'.
HEAD is now at b3a8e8f Merge pull request #301 from aznoohwee/bugfix/polling-flag-within-isr
==> make all HOMEBREW_PREFIX=/usr/local RELEASE=1
==> make install DESTDIR=/usr/local/Cellar/simavr/HEAD-b3a8e8f HOMEBREW_PREFIX=/usr/local RELEASE=1
Last 15 lines from /Users/patrick/Library/Logs/Homebrew/simavr/02.make:
LD obj-x86_64-apple-darwin17.7.0/run_avr.elf
ln -sf obj-x86_64-apple-darwin17.7.0/run_avr.elf run_avr
mkdir -p /usr/local/Cellar/simavr/HEAD-b3a8e8f/include/simavr/avr
install -m644 sim/*.h /usr/local/Cellar/simavr/HEAD-b3a8e8f/include/simavr/
install -m644 sim_core_*.h /usr/local/Cellar/simavr/HEAD-b3a8e8f/include/simavr/
install -m644 sim/avr/*.h /usr/local/Cellar/simavr/HEAD-b3a8e8f/include/simavr/avr/
mkdir -p /usr/local/Cellar/simavr/HEAD-b3a8e8f/lib
install obj-x86_64-apple-darwin17.7.0/libsimavr.a /usr/local/Cellar/simavr/HEAD-b3a8e8f/lib/
mkdir -p /usr/local/Cellar/simavr/HEAD-b3a8e8f/lib/pkgconfig/
sed -e "s|PREFIX|/usr/local/Cellar/simavr/HEAD-b3a8e8f|g" -e "s|VERSION|unknown|g" \
		simavr-avr.pc >/usr/local/Cellar/simavr/HEAD-b3a8e8f/lib/pkgconfig/simavr-avr.pc
sed -e "s|PREFIX|/usr/local/Cellar/simavr/HEAD-b3a8e8f|g" -e "s|VERSION|unknown|g" \
		simavr.pc >/usr/local/Cellar/simavr/HEAD-b3a8e8f/lib/pkgconfig/simavr.pc
mkdir -p /usr/local/Cellar/simavr/HEAD-b3a8e8f/bin
install obj-x86_64-apple-darwin17.7.0/run_avr.elf /usr/local/Cellar/simavr/HEAD-b3a8e8f/bin/simavr

If reporting this issue please do so at (not Homebrew/brew or Homebrew/core):
https://github.com/osx-cross/homebrew-avr/issues
```

jacquesg, 2018-08-25:
This is in the error log:

```
2018-08-25 09:34:51 +0200

make
install
DESTDIR=/usr/local/Cellar/simavr/HEAD-b3a8e8f
HOMEBREW_PREFIX=/usr/local
RELEASE=1

/Applications/Xcode.app/Contents/Developer/usr/bin/make -C simavr install RELEASE=1 DESTDIR=/usr/local/Cellar/simavr/HEAD-b3a8e8f PREFIX=/usr/local/Cellar/simavr/HEAD-b3a8e8f
/Applications/Xcode.app/Contents/Developer/usr/bin/make -C examples/parts install RELEASE=1 DESTDIR=/usr/local/Cellar/simavr/HEAD-b3a8e8f PREFIX=/usr/local/Cellar/simavr/HEAD-b3a8e8f
fatal: No names found, cannot describe anything.
mkdir -p /usr/local/Cellar/simavr/HEAD-b3a8e8f/include/simavr/parts
install -m644 *.h /usr/local/Cellar/simavr/HEAD-b3a8e8f/include/simavr/parts/
install obj-x86_64-apple-darwin17.7.0/libsimavrparts.a /usr/local/Cellar/simavr/HEAD-b3a8e8f/lib/
install: /usr/local/Cellar/simavr/HEAD-b3a8e8f/lib/: No such file or directory
make[1]: *** [install] Error 71
make: *** [install-parts] Error 2
make: *** Waiting for unfinished jobs....
/Applications/Xcode.app/Contents/Developer/usr/bin/make obj config
fatal: No names found, cannot describe anything.
make[2]: Nothing to be done for `obj'.
make[2]: Nothing to be done for `config'.
/Applications/Xcode.app/Contents/Developer/usr/bin/make libsimavr run_avr
fatal: No names found, cannot describe anything.
make[2]: Nothing to be done for `libsimavr'.
LD obj-x86_64-apple-darwin17.7.0/run_avr.elf
ln -sf obj-x86_64-apple-darwin17.7.0/run_avr.elf run_avr
mkdir -p /usr/local/Cellar/simavr/HEAD-b3a8e8f/include/simavr/avr
install -m644 sim/*.h /usr/local/Cellar/simavr/HEAD-b3a8e8f/include/simavr/
install -m644 sim_core_*.h /usr/local/Cellar/simavr/HEAD-b3a8e8f/include/simavr/
install -m644 sim/avr/*.h /usr/local/Cellar/simavr/HEAD-b3a8e8f/include/simavr/avr/
mkdir -p /usr/local/Cellar/simavr/HEAD-b3a8e8f/lib
install obj-x86_64-apple-darwin17.7.0/libsimavr.a /usr/local/Cellar/simavr/HEAD-b3a8e8f/lib/
mkdir -p /usr/local/Cellar/simavr/HEAD-b3a8e8f/lib/pkgconfig/
sed -e "s|PREFIX|/usr/local/Cellar/simavr/HEAD-b3a8e8f|g" -e "s|VERSION|unknown|g" \
                simavr-avr.pc >/usr/local/Cellar/simavr/HEAD-b3a8e8f/lib/pkgconfig/simavr-avr.pc
sed -e "s|PREFIX|/usr/local/Cellar/simavr/HEAD-b3a8e8f|g" -e "s|VERSION|unknown|g" \
                simavr.pc >/usr/local/Cellar/simavr/HEAD-b3a8e8f/lib/pkgconfig/simavr.pc
mkdir -p /usr/local/Cellar/simavr/HEAD-b3a8e8f/bin
install obj-x86_64-apple-darwin17.7.0/run_avr.elf /usr/local/Cellar/simavr/HEAD-b3a8e8f/bin/simavr
```

ladislas, 2018-09-29:
I'm not using simarv myself and don't have much time to investigate.

@patrickelectric feel free to propose a PR with a fix.

Closing the issue for now.

patrickelectric, 2018-09-29:
the problem still exist and in my mind the issue should be open until a solution appears, I don't see why closing, this will only make people to create new issues based on this one.

Besides that, I can take a look if someone points the way to go.

ladislas, 2018-09-29:
I reopened it but as I said I don’t know how to fix this and I don’t have time.

ladislas, 2018-09-30:
I can confirm that your fix fixed it 👍

ladislas, 2018-09-30:
I've changed the formulae to point to your fix while we wait for the PR to be accepted here: https://github.com/buserror/simavr/pull/306

patrickelectric, 2018-11-06:
Is someone maintaining https://github.com/buserror/simavr ?

ladislas, 2018-11-10:
@patrickelectric doesn't seem so...

If you want, we could fork it under osx-cross and apply your changes here.

ladislas, 2019-02-26:
@patrickelectric as your PR has been merged, the issue is now resolved.

Thanks for your help! :)

patrickelectric, 2019-02-26:
You can check the status here:
https://github.com/buserror/simavr/pull/306

ladislas, 2019-02-27:
Oh yeah my bad!

ladislas, 2019-11-27:
@patrickelectric buserror/simavr#306 has been closed...

any chance we could fix that in the formula? and a line to do this `$(MKDIR) $(DESTDIR)/lib/pkgconfig`?

ladislas, 2020-01-08:
@patrickelectric closing the issue now. Feel free to open a PR with a fix.

closed 2020-01-08 by ladislas

patrickelectric, 2020-01-08:
Well, the simavr dev does not want to fix it and also does not want to merge my PR with the fix. No idea about the actual status and I'm not using it anymore.

ladislas, 2020-01-09:
@patrickelectric as I don't like having a broken formula, I decided to tackle the bug. The solution was actually super simple: set `ENV.deparallelize` before calling `make`.

It now works like a charm.

commit 3da8324b (2018-09-30): WIP - fix mac installation (waiting for PR to be merged https://github.com/buserror/simavr/pull/306 - fix #82
commit 04a8abc4 (2020-01-09): fix simavr build issue by setting ENV.deparallelize - fix #82