osx-cross/homebrew-avr #85: Download link of avr-libc is down; cannot install avr-gcc

rraihansaputra, 2018-09-29:
```
⚡ brew install avr-gcc@7
Updating Homebrew...
==> Installing avr-gcc@7 from osx-cross/avr
==> Downloading https://gcc.gnu.org/pub/gcc/releases/gcc-7.3.0/gcc-7.3.0.tar.xz
######################################################################## 100.0%
==> Patching
patching file gcc/graphite.h
==> ../configure --target=avr --prefix=/usr/local/Cellar/avr-gcc@7/7.3.0 --libdir=/usr/local/Cellar/a
==> make
==> make install
==> Downloading https://download.savannah.gnu.org/releases/avr-libc/avr-libc-2.0.0.tar.bz2

curl: (7) Failed to connect to mirror.freedif.org port 443: Operation timed out
Error: An exception occured within a child process:
  DownloadError: Failed to download resource "avr-gcc@7--avr-libc"
Download failed: https://download.savannah.gnu.org/releases/avr-libc/avr-libc-2.0.0.tar.bz2
```

The mirror linked from homebrew is down for me (https://mirror.freedif.org/GNU-Sa/avr-libc/avr-libc-2.0.0.tar.bz2). I have obtained the correct file from http://download-mirror.savannah.gnu.org/releases/avr-libc/ but don't know what to do with it.

ladislas, 2018-09-29:
should work now, let me know.

closed 2018-09-29 by ladislas

rraihansaputra, 2018-09-29:
I did fork it and changed the link as you did on your fix https://github.com/osx-cross/homebrew-avr/commit/75e059ef17072d2f7d75e228126b070bd1d43c0c so it should work. Thanks!

commit 75e059ef (2018-09-29): add mirror for avr-libc - fix #85