osx-cross/homebrew-avr #87: brew install avr-gcc@6 fails on mojave

kaijajan, 2018-11-23:
I attemped to install avr-gcc 6.4 on mojave with brew, but the following error occurs while doing make
anyone use avr-gcc 6.4 on mojave and successfully installed?

Build error:

```
Last 15 lines from /Users/archer2532/Library/Logs/Homebrew/avr-gcc@6/02.make:
/usr/local/include/isl/map.h:496:21: note: 'isl_map_domain' declared here
__isl_give isl_set *isl_map_domain(__isl_take isl_map *bmap);
                    ^
../../gcc/graphite-sese-to-poly.c:675:26: error: cannot initialize a parameter of type 'isl_map *' with an rvalue of type 'isl_space *'
                            isl_space_domain (isl_map_get_space (acc)));
                                              ^~~~~~~~~~~~~~~~~~~~~~~
/usr/local/include/isl/map.h:496:56: note: passing argument to parameter 'bmap' here
__isl_give isl_set *isl_map_domain(__isl_take isl_map *bmap);
                                                       ^
fatal error: too many errors emitted, stopping now [-ferror-limit=]
20 errors generated.
make[2]: *** [graphite-sese-to-poly.o] Error 1
rm gcc.pod
make[1]: *** [all-gcc] Error 2
make: *** [all] Error 2
```

Brew Config:

```
HOMEBREW_VERSION: 1.8.3
ORIGIN: https://github.com/Homebrew/brew
HEAD: 4021aa80dfbaa369ee357a955c0216f9566133ac
Last commit: 6 days ago
Core tap ORIGIN: https://github.com/Homebrew/homebrew-core
Core tap HEAD: a266871741361ef22b342106d13b85c669ede727
Core tap last commit: 5 hours ago
HOMEBREW_PREFIX: /usr/local
CPU: octa-core 64-bit ivybridge
Homebrew Ruby: 2.3.7 => /usr/local/Homebrew/Library/Homebrew/vendor/portable-ruby/2.3.7/bin/ruby
Clang: 10.0 build 1000
Git: 2.17.2 => /Applications/Xcode.app/Contents/Developer/usr/bin/git
Curl: 7.54.0 => /usr/bin/curl
macOS: 10.14-x86_64
CLT: 10.1.0.0.1.1539992718
Xcode: 10.1
```

ladislas, 2018-11-23:
@kaijajan it should be good know, can you test it?

closed 2018-11-23 by ladislas

kaijajan, 2018-11-23:
it works now, thank you so much =D

commit aaabd6d8 (2018-11-23): activate patch for isl - fix #87