osx-cross/homebrew-avr #88: Fails to build on High Sierra (node)

pinglivin, 2018-12-03:
This is task depends of #76.
When I installed node using Homebrew, I get same error as was wroten in task #76.
I used command  
`brew link --overwrite --dry-run node`
that resolved this problem. After when I apply this command and repeat `brew doctor` I get message 

> MacBook-Pro-8:~ xxx$ brew doctor
Please note that these warnings are just used to help the Homebrew maintainers
with debugging if you file an issue. If everything you use Homebrew for is
working fine: please don't worry or file an issue; just ignore this. Thanks!
Warning: Unbrewed header files were found in /usr/local/include.
If you didn't put them there on purpose they could cause problems when
building Homebrew formulae, and may need to be deleted.
Unexpected header files:
  /usr/local/include/node/v8-inspector-protocol.h
  /usr/local/include/node/v8-inspector.h

Please help me resolved this problem. Thanks.

ladislas, 2018-12-03:
As you mentioned, this issue was already investigated but no solution was found.

The OP suggested building from source: https://github.com/osx-cross/homebrew-avr/issues/76#issuecomment-376019810

I have node and avr-gcc installed with absolutely no issues.

Try uninstalling node and untapping homebrew-avr, then tap again and try installing avr-gcc before node.

`brew upgrade` might also help.

ladislas, 2019-01-30:
No updates here, closing the issue for now. Feel free to reopen if the problem persists.

closed 2019-01-30 by ladislas