osx-cross/homebrew-avr #90: avr-gcc 8.2.0 has a known bug

cgkades, 2019-02-03:
Hey guys, looks like 8.2.0 has a bug that causes issues when trying to flash a keyboard with qmk and avrdude. I tried to downgrade in brew but either I'm doing something wrong, or older versions aren't available. 

Here's a link to a discussion about it https://github.com/qmk/qmk_firmware/issues/3657

ladislas, 2019-02-26:
Hi @cgkades,
Are you trying to downgrade to 8.1.0 or to 7.x.x?

For the first, you can edit the forumula, change the link and the sha256.

For the second, you can install it using `brew install avr-gcc@7`

GCC 8.3.0 is out, do you know if the bug is fixed?

cgkades, 2019-02-26:
I was just trying to downgrade to any other version, and for some reason I just wasn't finding that avr-gcc@7 syntax. I do not know for sure if the bug was fixed in 8.3, sorry

ladislas, 2019-02-27:
Alright! I'll close the issue and update to 8.3.0. if the issue persist let us know!

closed 2019-02-27 by ladislas