osx-cross/homebrew-avr #91: brew install avr-gcc fails on Mojave

frankimhof, 2019-02-18:
### Error when installing

`Installing osx-cross/avr/avr-gcc
==> Downloading https://ftp.gnu.org/gnu/gcc/gcc-8.2.0/gcc-8.2.0.tar.xz
######################################################################## 100.0%
==> Patching
patching file gcc/graphite.h
==> ../configure --target=avr --prefix=/usr/local/Cellar/avr-gcc/8.2.0 --libdir=
==> make
Last 15 lines from /Users/frankimhof/Library/Logs/Homebrew/avr-gcc/02.make:
libtool: link: sed -e 's,^,_,' < ../../libcc1/libcc1plugin.sym > .libs/libcc1plugin-symbols.expsym`

`libtool: link: clang++ -Wl,-undefined -Wl,dynamic_lookup -o .libs/libcc1plugin.0.so -bundle  
.libs/libcc1plugin.o .libs/callbacks.o .libs/connection.o .libs/marshall.o    -Wl,-no_pie ../libiberty/pic/libiberty.a   -Wl,-exported_symbols_list,.libs/libcc1plugin-symbols.expsym
libtool: link: dsymutil .libs/libcc1plugin.0.so || :
warning: no debug symbols in executable (-arch x86_64)`

`libtool: link: (cd ".libs" && rm -f "libcc1plugin.so" && ln -s "libcc1plugin.0.so" "libcc1plugin.so")
mv -f .deps/libcp1plugin.Tpo .deps/libcp1plugin.Plo
/bin/sh ./libtool --tag=CXX   --mode=link clang++ -W -Wall  -fvisibility=hidden -g -O2 -module -export-symbols ../../libcc1/libcp1plugin.sym  '-Wl,-no_pie' -o libcp1plugin.la -rpath /usr/local/Cellar/avr-gcc/8.2.0/lib/avr-gcc/8/gcc/avr/8.2.0/plugin libcp1plugin.lo callbacks.lo connection.lo marshall.lo   -Wc,../libiberty/pic/libiberty.a 
libtool: link: ( cd ".libs" && rm -f "libcc1plugin.la" && ln -s "../libcc1plugin.la" "libcc1plugin.la" )
libtool: link: sed -e 's,^,_,' < ../../libcc1/libcp1plugin.sym > .libs/libcp1plugin-symbols.expsym
libtool: link: clang++ -Wl,-undefined -Wl,dynamic_lookup -o .libs/libcp1plugin.0.so -bundle  .libs/libcp1plugin.o .libs/callbacks.o .libs/connection.o .libs/marshall.o    -Wl,-no_pie ../libiberty/pic/libiberty.a   -Wl,-exported_symbols_list,.libs/libcp1plugin-symbols.expsym
libtool: link: dsymutil .libs/libcp1plugin.0.so || :
warning: no debug symbols in executable (-arch x86_64)`

`libtool: link: (cd ".libs" && rm -f "libcp1plugin.so" && ln -s "libcp1plugin.0.so" "libcp1plugin.so")
libtool: link: ( cd ".libs" && rm -f "libcp1plugin.la" && ln -s "../libcp1plugin.la" "libcp1plugin.la" )
make: *** [all] Error 2`

### Environment
`HOMEBREW_VERSION: 2.0.2
ORIGIN: https://github.com/Homebrew/brew
HEAD: 117c24f4b6294e037431d3a850ced6955d53e26f
Last commit: 30 hours ago
Core tap ORIGIN: https://github.com/Homebrew/homebrew-core
Core tap HEAD: 782131f79f3eb840ca6d162d2ffe09bbb56c48ea
Core tap last commit: 89 minutes ago
HOMEBREW_PREFIX: /usr/local
HOMEBREW_LOGS: /Users/frankimhof/Library/Logs/Homebrew
CPU: dual-core 64-bit penryn
Homebrew Ruby: 2.3.7 => /System/Library/Frameworks/Ruby.framework/Versions/2.3/usr/bin/ruby
Clang: 10.0 build 1000
Git: 2.17.2 => /Applications/Xcode.app/Contents/Developer/usr/bin/git
Curl: 7.54.0 => /usr/bin/curl
Java: 11.0.1
macOS: 10.14.2-x86_64
CLT: 10.1.0.0.1.1539992718
Xcode: 10.1`

ladislas, 2019-02-19:
Well it does build perfectly well on my machine using macOS Mojave as well.

Have you tried `brew remove`-ing `avr-gcc`, `brew cleanup`-ing cache and starting again?

Was a previous version installed? Can you list all your installed formulae?

frankimhof, 2019-02-19:
Hey
Thanks for your fast answer.
### Cleanup and Remove
> $ brew remove avr-gcc
Error: No such keg: /usr/local/Cellar/avr-gcc
$ brew cleanup
Removing: /Users/frankimhof/Library/Caches/Homebrew/node--11.6.0.mojave.bottle.tar.gz... (12.7MB)
Removing: /Users/frankimhof/Library/Logs/Homebrew/icu4c... (68B)
Removing: /Users/frankimhof/Library/Logs/Homebrew/node... (68B)

### Install

> $ brew install avr-gcc
Updating Homebrew...
==> Auto-updated Homebrew!
Updated 1 tap (homebrew/core).
==> New Formulae
mage
==> Updated Formulae
caddy               flake8              jfrog-cli-go        ship
cassandra           google-java-format  links               topgrade
checkbashisms       imagemagick         mariadb@10.1
embulk              ipython             mysql@5.7
eslint              jenkins             pixman
==> Installing avr-gcc from osx-cross/avr
==> Downloading https://ftp.gnu.org/gnu/gcc/gcc-8.2.0/gcc-8.2.0.tar.xz
Already downloaded: /Users/frankimhof/Library/Caches/Homebrew/downloads/241c8efec83436d0c5a6fe75389062072fef694334fbbb253d612b0ef7e82cf4--gcc-8.2.0.tar.xz
==> Patching
patching file gcc/graphite.h
==> ../configure --target=avr --prefix=/usr/local/Cellar/avr-gcc/8.2.0 --libdir=
==> make
Last 15 lines from /Users/frankimhof/Library/Logs/Homebrew/avr-gcc/02.make:
mv -f .deps/libcp1plugin.Tpo .deps/libcp1plugin.Plo
/bin/sh ./libtool --tag=CXX   --mode=link clang++ -W -Wall  -fvisibility=hidden -g -O2 -module -export-symbols ../../libcc1/libcp1plugin.sym  '-Wl,-no_pie' -o libcp1plugin.la -rpath /usr/local/Cellar/avr-gcc/8.2.0/lib/avr-gcc/8/gcc/avr/8.2.0/plugin libcp1plugin.lo callbacks.lo connection.lo marshall.lo   -Wc,../libiberty/pic/libiberty.a 
libtool: link: sed -e 's,^,_,' < ../../libcc1/libcc1plugin.sym > .libs/libcc1plugin-symbols.expsym
libtool: link: clang++ -Wl,-undefined -Wl,dynamic_lookup -o .libs/libcc1plugin.0.so -bundle  .libs/libcc1plugin.o .libs/callbacks.o .libs/connection.o .libs/marshall.o    -Wl,-no_pie ../libiberty/pic/libiberty.a   -Wl,-exported_symbols_list,.libs/libcc1plugin-symbols.expsym
libtool: link: sed -e 's,^,_,' < ../../libcc1/libcp1plugin.sym > .libs/libcp1plugin-symbols.expsym
libtool: link: clang++ -Wl,-undefined -Wl,dynamic_lookup -o .libs/libcp1plugin.0.so -bundle  .libs/libcp1plugin.o .libs/callbacks.o .libs/connection.o .libs/marshall.o    -Wl,-no_pie ../libiberty/pic/libiberty.a   -Wl,-exported_symbols_list,.libs/libcp1plugin-symbols.expsym
libtool: link: dsymutil .libs/libcc1plugin.0.so || :
warning: no debug symbols in executable (-arch x86_64)
libtool: link: (cd ".libs" && rm -f "libcc1plugin.so" && ln -s "libcc1plugin.0.so" "libcc1plugin.so")
libtool: link: dsymutil .libs/libcp1plugin.0.so || :
warning: no debug symbols in executable (-arch x86_64)
libtool: link: ( cd ".libs" && rm -f "libcc1plugin.la" && ln -s "../libcc1plugin.la" "libcc1plugin.la" )
libtool: link: (cd ".libs" && rm -f "libcp1plugin.so" && ln -s "libcp1plugin.0.so" "libcp1plugin.so")
libtool: link: ( cd ".libs" && rm -f "libcp1plugin.la" && ln -s "../libcp1plugin.la" "libcp1plugin.la" )
make: *** [all] Error 2

### Brew List

> $ brew list
avr-binutils	gmp		icu4c		isl		libmpc		mpfr

### Other
I had CrossPack installed before, but I removed it. Don't know if this could be the problem.

ladislas, 2019-02-19:
I've never used crosspack, but that might be.

as you don't have a lot of formulae, can you remove them all and try again?

frankimhof, 2019-02-20:
### Uninstall
> $ brew remove --force $(brew list) --ignore-dependencies
Uninstalling avr-binutils... (157 files, 12.4MB)
Uninstalling gmp... (18 files, 3.1MB)
Uninstalling icu4c... (254 files, 68.4MB)
Uninstalling isl... (71 files, 3.9MB)
Uninstalling libmpc... (12 files, 354.2KB)
Uninstalling mpfr... (28 files, 4.6MB)
$ brew cleanup
$ 

But I still have the same errors. When installing.

ladislas, 2019-02-20:
could you run `brew install avr-gcc --verbose` and share the whole output through [pastbin](https://pastebin.com/) or equivalent?

frankimhof, 2019-02-20:
yes, here is the link
https://justpaste.it/5eprr

ladislas, 2019-02-22:
I can see the error but I don't understand where it comes from and I can't reproduce on my system...

Was it working before? Do you have the latest Xcode and Command line Tools installed? What computer are you using? What is the ouput of `brew doctor`?

frankimhof, 2019-02-26:
I never tried it before. Here is my information:

### System
MacBook Pro 5.3
Intel Core 2 Duo
2.8 GHz
### MacOS and XCode
MacOS 10.14.2 (Dosdude1-Version, because I have an old Mac)
XCode 10.1 with developper command line tools installed.

### Brew Doctor

> 

> $ brew doctor
Please note that these warnings are just used to help the Homebrew maintainers
with debugging if you file an issue. If everything you use Homebrew for is
working fine: please don't worry or file an issue; just ignore this. Thanks!

Warning: "config" scripts exist outside your system or Homebrew directories.
`./configure` scripts often look for *-config scripts to determine if
software packages are installed, and what additional flags to use when
compiling and linking.

Having additional scripts in your path can confuse software installed via
Homebrew if the config script overrides a system or Homebrew provided
script of the same name. We found the following "config" scripts:
  /usr/local/CrossPack-AVR/bin/libusb-config

Warning: Unbrewed dylibs were found in /usr/local/lib.
If you didn't put them there on purpose they could cause problems when
building Homebrew formulae, and may need to be deleted.

Unexpected dylibs:
  /usr/local/lib/SUVMMFaker.dylib
  /usr/local/lib/libjlinkpic32.6.34.6.dylib

Warning: Unbrewed header files were found in /usr/local/include.
If you didn't put them there on purpose they could cause problems when
building Homebrew formulae, and may need to be deleted.

Unexpected header files:
  /usr/local/include/node/libplatform/libplatform-export.h
  /usr/local/include/node/libplatform/libplatform.h
  /usr/local/include/node/libplatform/v8-tracing.h
  /usr/local/include/node/node.h
  /usr/local/include/node/node_api.h
  /usr/local/include/node/node_api_types.h
  /usr/local/include/node/node_buffer.h
  /usr/local/include/node/node_object_wrap.h
  /usr/local/include/node/node_version.h
  /usr/local/include/node/openssl/aes.h
  /usr/local/include/node/openssl/archs/BSD-x86_64/asm/crypto/buildinf.h
  /usr/local/include/node/openssl/archs/BSD-x86_64/asm/crypto/include/internal/bn_conf.h
  /usr/local/include/node/openssl/archs/BSD-x86_64/asm/crypto/include/internal/dso_conf.h
  /usr/local/include/node/openssl/archs/BSD-x86_64/asm/include/openssl/opensslconf.h
  /usr/local/include/node/openssl/archs/BSD-x86_64/asm/include/progs.h
  /usr/local/include/node/openssl/archs/BSD-x86_64/no-asm/crypto/buildinf.h
  /usr/local/include/node/openssl/archs/BSD-x86_64/no-asm/crypto/include/internal/bn_conf.h
  /usr/local/include/node/openssl/archs/BSD-x86_64/no-asm/crypto/include/internal/dso_conf.h
  /usr/local/include/node/openssl/archs/BSD-x86_64/no-asm/include/openssl/opensslconf.h
  /usr/local/include/node/openssl/archs/BSD-x86_64/no-asm/include/progs.h
  /usr/local/include/node/openssl/archs/VC-WIN32/asm/crypto/buildinf.h
  /usr/local/include/node/openssl/archs/VC-WIN32/asm/crypto/include/internal/bn_conf.h
  /usr/local/include/node/openssl/archs/VC-WIN32/asm/crypto/include/internal/dso_conf.h
  /usr/local/include/node/openssl/archs/VC-WIN32/asm/include/openssl/opensslconf.h
  /usr/local/include/node/openssl/archs/VC-WIN32/asm/include/progs.h
  /usr/local/include/node/openssl/archs/VC-WIN32/no-asm/crypto/buildinf.h
  /usr/local/include/node/openssl/archs/VC-WIN32/no-asm/crypto/include/internal/bn_conf.h
  /usr/local/include/node/openssl/archs/VC-WIN32/no-asm/crypto/include/internal/dso_conf.h
[... 267 lines cut ...]
  /usr/local/include/node/uv/win.h
  /usr/local/include/node/v8-inspector-protocol.h
  /usr/local/include/node/v8-inspector.h
  /usr/local/include/node/v8-platform.h
  /usr/local/include/node/v8-profiler.h
  /usr/local/include/node/v8-testing.h
  /usr/local/include/node/v8-util.h
  /usr/local/include/node/v8-value-serializer-version.h
  /usr/local/include/node/v8-version-string.h
  /usr/local/include/node/v8-version.h
  /usr/local/include/node/v8.h
  /usr/local/include/node/v8config.h
  /usr/local/include/node/zconf.h
  /usr/local/include/node/zlib.h

Warning: Homebrew's sbin was not found in your PATH but you have installed
formulae that put executables in /usr/local/sbin.
Consider setting the PATH for example like so
  echo 'export PATH="/usr/local/sbin:$PATH"' >> ~/.bash_profile

Warning: Broken symlinks were found. Remove them with `brew cleanup`:
  /usr/local/lib/libmchpusb-1.0.dylib

### Conclusion
I will be using avr-gcc. I don't want to overstay my welcome with my strange system. But thanks for your investigation efforts!

ladislas, 2019-02-26:
Thanks! 

As stated by brew doctor, you have a lot of things in your path that might confuse the installation of avr-gcc.

i would remove crosspack and all the node's files and even uninstall brew completely and start from scratch.

https://docs.brew.sh/FAQ#how-do-i-uninstall-homebrew

frankimhof, 2019-03-02:
I uninstalled brew, node and crosspatch.
I even deleted all files in /usr/local that had to do with node.
Then i reinstalled brew and node (with brew) which was successful.
> $ brew doctor
Your system is ready to brew.
So i tried reinstalling avr-gcc but it still gave me the same errors.

If you have anything else I could try, let me know.

ladislas, 2019-03-03:
This issue seems to be this one: 

```
configure: error: in `/private/tmp/avr-gcc-20190302-23684-1jbum7h/gcc-8.3.0/build/avr/libgcc':

configure: error: cannot compute suffix of object files: cannot compile

See `config.log' for more details.

make[1]: *** [configure-target-libgcc] Error 1

make[1]: *** Waiting for unfinished jobs....
```

You can try the following solution: https://stackoverflow.com/a/13025807/2205264
But I'm not sure it will help.

CircleOnCircles, 2019-03-14:
> This issue seems to be this one:
> 
> ```
> configure: error: in `/private/tmp/avr-gcc-20190302-23684-1jbum7h/gcc-8.3.0/build/avr/libgcc':
> 
> configure: error: cannot compute suffix of object files: cannot compile
> 
> See `config.log' for more details.
> 
> make[1]: *** [configure-target-libgcc] Error 1
> 
> make[1]: *** Waiting for unfinished jobs....
> ```
> You can try the following solution: [stackoverflow.com/a/13025807/2205264](https://stackoverflow.com/a/13025807/2205264)
> But I'm not sure it will help.

This works!

ladislas, 2019-03-15:
@CircleOnCircles thanks for the feedback!

were you having the same issue as @frankimhof?

I don't understand why this doesn't work for you whereas it does work for me.

ladislas, 2019-03-20:
I've tried reinstalling on a brand new computer with no issues.
As I can't reproduce, I'll close the issue for now.

Feel free to reopen if you have more information.

closed 2019-03-20 by ladislas