osx-cross/homebrew-avr #93: 8.3.0 keeps re-installing

scottsuch, 2019-03-04:
If I run `brew update; brew cask upgrade` this package will install over and over again.

```==> Upgrading osx-cross/avr/avr-gcc
==> Downloading https://ftp.gnu.org/gnu/gcc/gcc-8.3.0/gcc-8.3.0.tar.xz
Already downloaded: /Users/sgorsuch/Library/Caches/Homebrew/downloads/790ecda70af93e36763edcb183a8d4e91e433bee9bad9a60360addae9d8e82be--gcc-8.3.0.tar.xz
==> Patching
==> ../configure --target=avr --prefix=/usr/local/Cellar/avr-gcc/8.3.0 --libdir=/usr/local/Cellar/avr-gcc/8.3.0/lib/avr-gcc/8 --enable-languages=c,c++ --with-ld=/usr/local/opt/avr-binutils/bin/avr-ld --with-as=/
Last 15 lines from /Users/sgorsuch/Library/Logs/Homebrew/avr-gcc/01.configure:
checking for gnatmake... no
checking whether compiler driver understands Ada... no
checking how to compare bootstrapped objects... cmp --ignore-initial=16 $$f1 $$f2
checking for objdir... .libs
checking for the correct version of gmp.h... yes
checking for the correct version of mpfr.h... no
configure: error: Building GCC requires GMP 4.2+, MPFR 2.4.0+ and MPC 0.8.0+.
Try the --with-gmp, --with-mpfr and/or --with-mpc options to specify
their locations.  Source code for these libraries can be found at
their respective hosting sites as well as at
ftp://gcc.gnu.org/pub/gcc/infrastructure/.  See also
http://gcc.gnu.org/install/prerequisites.html for additional info.  If
you obtained GMP, MPFR and/or MPC from a vendor distribution package,
make sure that you have installed both the libraries and the header
files.  They may be located in separate packages.

If reporting this issue please do so at (not Homebrew/brew or Homebrew/core):
https://github.com/osx-cross/homebrew-avr/issues

These open issues may also help:
brew install avr-gcc fails on Mojave https://github.com/osx-cross/homebrew-avr/issues/91```

ladislas, 2019-03-05:
Can you share the output of:

- [ ] `brew config`
- [ ] `brew doctor`
- [ ] `brew list`
- [ ] `brew cask list`

scottsuch, 2019-03-05:
I just performed an uninstall and reinstall yesterday and now am unable to reproduce this behavior. Sorry bothering you with this.

closed 2019-03-05 by scottsuch

ladislas, 2019-03-05:
no worries! glad it's fixed :)