osx-cross/homebrew-avr #96: building avarice fails

ladislas, 2019-03-21:
building avarice fails with the following error:

```
==> Cloning https://svn.code.sf.net/p/avarice/code/trunk
Updating /Users/ladislas/Library/Caches/Homebrew/avarice--svn-HEAD
svn update
Updating '.':
At revision 375.
svn export --force . /private/tmp/d20190321-33587-67zq49
cp -pR /private/tmp/d20190321-33587-67zq49/avarice/. /private/tmp/avarice-20190321-33587-ns668z/avarice
chmod -Rf +w /private/tmp/d20190321-33587-67zq49
Error: An exception occurred within a child process:
  Errno::ENOENT: No such file or directory @ dir_chdir - avarice
```

@carlossless - you were the one providing us with the formula in the first place, any idea what we should change to fix this?

ladislas, 2019-03-21:
@carlossless no need, I've found the issue :)

closed 2019-03-21 by ladislas

carlossless, 2019-03-27:
@ladislas cool, glad you got it working!

commit e7cb8310 (2019-03-21): fix avarice building issue - fix #86 & fix #96