osx-cross/homebrew-avr #98: avr-gcc@7

exoizzy, 2019-03-22:
im new to homebrew and terminal so bear with me. im trying to start working on a custom mechanical keyboard and was following these instructions [https://docs.qmk.fm/#/getting_started_build_tools](https://docs.qmk.fm/#/getting_started_build_tools) and got to the `brew install avr-gcc@7` step and it gave me this: 

> Installing avr-gcc@7 from osx-cross/avr
> ==> Downloading https://gcc.gnu.org/pub/gcc/releases/gcc-7.3.0/gcc-7.3.0.tar.xz
> ######################################################################## 100.0%
> ==> ../configure --target=avr --prefix=/usr/local/Cellar/avr-gcc@7/7.3.0 --libdir=/usr/local/Cellar/av
> ==> make
> Last 15 lines from /Users/me/Library/Logs/Homebrew/avr-gcc@7/02.make:
>        unsigned n_row, unsigned n_col);
>                  ^
> 11 errors generated.
> make[2]: *** [graphite-isl-ast-to-gimple.o] Error 1
> ../../gcc/graphite-optimize-isl.c:57:19: error: use of undeclared identifier 'isl_space_dim'
>   unsigned dims = isl_space_dim (space, isl_dim_set);
>                   ^
> ../../gcc/graphite-optimize-isl.c:60:3: error: use of undeclared identifier 'isl_space_free'
>   isl_space_free (space);
>   ^
> 2 errors generated.
> make[2]: *** [graphite-optimize-isl.o] Error 1
> rm gcc.pod
> make[1]: *** [all-gcc] Error 2
> make: *** [all] Error 2

no clue what any of this means but I dont know why there would be errors like this here. thanks in advance for any help

fauxpark, 2019-03-22:
This is probably due to https://github.com/osx-cross/homebrew-avr/commit/2b127ae20f66ae7ec0defdafa9b2cbffc97c144a, in particular the removal of the patch that adds some ISL `#include`s.

ladislas, 2019-03-22:
@exoizzy 4560f36 should fix the issue. Can you try it with:

```shell
$ cd /usr/local/Homebrew/Library/Taps/osx-cross/homebrew-avr
$ git fetch origin pull/100/head:fix-isl-gcc@7-#98
$ git checkout fix-isl-gcc@7-#98
$ brew install ./avr-gcc@7.rb
```

rumbogs, 2019-03-23:
@ladislas: this isn't working, the local repo can't pull the branch for some reason 🤔

matarrubia, 2019-03-23:
The above did not work for me. I had to temporarily install `isl` `0.19` to work around this.

ladislas, 2019-03-25:
it should be working with 4560f36a976c4acbc1a472ebe5c06d43ccac7bc3 that I just merged into master.

I had removed the patch thinking it was fixed but turns out it is not yet.

I'll let you try and let me know if is works now.

ladislas, 2019-03-25:
It does compile on my machine:

```
→ brew config
HOMEBREW_VERSION: 2.0.5-40-geece1f0
ORIGIN: https://github.com/Homebrew/brew
HEAD: eece1f00f029e9ffcf07976dd0e001e6343ce4e1
Last commit: 16 hours ago
Core tap ORIGIN: https://github.com/Homebrew/homebrew-core
Core tap HEAD: 0bd7ca0bae2c5ce8ce10898d00362cc9114ed4b8
Core tap last commit: 3 hours ago
HOMEBREW_PREFIX: /usr/local
HOMEBREW_DEV_CMD_RUN: 1
HOMEBREW_DISPLAY: /private/tmp/com.apple.launchd.mFmGSKCOEt/org.macosforge.xquartz:0
HOMEBREW_LOGS: /Users/ladislas/Library/Logs/Homebrew
HOMEBREW_VISUAL: nvim
CPU: quad-core 64-bit broadwell
Homebrew Ruby: 2.3.7 => /System/Library/Frameworks/Ruby.framework/Versions/2.3/usr/bin/ruby
Clang: 10.0 build 1000
Git: 2.21.0 => /usr/local/bin/git
Curl: 7.54.0 => /usr/bin/curl
Java: 11
macOS: 10.14.3-x86_64
CLT: 10.1.0.0.1.1539992718
Xcode: 10.1
XQuartz: 2.7.11 => /opt/X11
```

[image]

ladislas, 2019-03-25:
I've updated gcc to 7.4.0 and the problem has been fixed, the patch is not needed anymore.

commit 7291524f (2019-03-25): update to gcc 7.4.0 - fix #98 + fix #101
closed 2019-03-25 by ladislas