osx-cross/homebrew-avr #101: avr-gcc@7 can't build on Mojave

henrikjonhed, 2019-03-24:
Hi!

I get the following error when I try to install avr-gcc@7:

```
clang++ -fno-PIE -c   -g -O2 -DIN_GCC  -DCROSS_DIRECTORY_STRUCTURE  -fno-strict-aliasing -fno-exceptions -fno-rtti -fasynchronous-unwind-tables -W -Wall -Wno-narrowing -Wwrite-strings -Wcast-qual -Wmissing-format-attribute -Woverloaded-virtual -pedantic -Wno-long-long -Wno-variadic-macros -Wno-overlength-strings   -DHAVE_CONFIG_H -I. -I. -I../../gcc -I../../gcc/. -I../../gcc/../include -I../../gcc/../libcpp/include  -I../../gcc/../libdecnumber -I../../gcc/../libdecnumber/dpd -I../libdecnumber -I../../gcc/../libbacktrace   -o hash-map-tests.o -MT hash-map-tests.o -MMD -MP -MF ./.deps/hash-map-tests.TPo ../../gcc/hash-map-tests.c
../../gcc/graphite-isl-ast-to-gimple.c:98:7: error: use of undeclared identifier 'isl_id_free'
      isl_id_free (it->first);
      ^
../../gcc/graphite-isl-ast-to-gimple.c:293:3: error: use of undeclared identifier 'isl_id_free'
  isl_id_free (tmp_isl_id);
  ^
../../gcc/graphite-isl-ast-to-gimple.c:629:5: error: use of undeclared identifier 'isl_id_free'
    isl_id_free (res->first);
    ^
../../gcc/graphite-isl-ast-to-gimple.c:669:53: error: use of undeclared identifier 'isl_id_get_user'
      ast_build_info *for_info = (ast_build_info *) isl_id_get_user (id);
                                                    ^
../../gcc/graphite-isl-ast-to-gimple.c:672:7: error: use of undeclared identifier 'isl_id_free'
      isl_id_free (id);
      ^
../../gcc/graphite-optimize-isl.c:57:19: error: use of undeclared identifier 'isl_space_dim'
  unsigned dims = isl_space_dim (space, isl_dim_set);
                  ^
../../gcc/graphite-isl-ast-to-gimple.c:876:31: error: use of undeclared identifier 'isl_id_get_user'
  poly_bb_p pbb = (poly_bb_p) isl_id_get_user (name_id);
                              ^
../../gcc/graphite-optimize-isl.c:60:3: error: use of undeclared identifier 'isl_space_free'
  isl_space_free (space);
  ^
../../gcc/graphite-dependences.c:48:16: error: use of undeclared identifier 'isl_space_get_tuple_id'; did you mean 'isl_map_get_tuple_id'?
  isl_id *id = isl_space_get_tuple_id (d, isl_dim_in);
               ^~~~~~~~~~~~~~~~~~~~~~
               isl_map_get_tuple_id
/usr/local/include/isl/map.h:102:20: note: 'isl_map_get_tuple_id' declared here
__isl_give isl_id *isl_map_get_tuple_id(__isl_keep isl_map *map,
                   ^
../../gcc/graphite-dependences.c:48:40: error: cannot initialize a parameter of type 'isl_map *' with an lvalue of type 'isl_space *'
  isl_id *id = isl_space_get_tuple_id (d, isl_dim_in);
                                       ^
/usr/local/include/isl/map.h:102:61: note: passing argument to parameter 'map' here
__isl_give isl_id *isl_map_get_tuple_id(__isl_keep isl_map *map,
                                                            ^
../../gcc/graphite-isl-ast-to-gimple.c:882:3: error: use of undeclared identifier 'isl_id_free'
  isl_id_free (name_id);
  ^
../../gcc/graphite-dependences.c:51:3: error: use of undeclared identifier 'isl_space_free'
  isl_space_free (d);
  ^
2 errors generated.
../../gcc/graphite-dependences.c:179:16: error: use of undeclared identifier 'isl_space_dim'
  int nb_out = isl_space_dim (space, isl_dim_out);
               ^
make[2]: *** [graphite-optimize-isl.o] Error 1
make[2]: *** Waiting for unfinished jobs....
../../gcc/graphite-dependences.c:185:3: error: use of undeclared identifier 'isl_space_free'
  isl_space_free (space);
  ^
../../gcc/graphite-isl-ast-to-gimple.c:2886:19: error: use of undeclared identifier 'isl_space_dim'
  int dimension = isl_space_dim (schedule_space, isl_dim_out);
                  ^
../../gcc/graphite-dependences.c:195:11: error: use of undeclared identifier 'isl_space_dim'
  int n = isl_space_dim (space, isl_dim_out);
          ^
../../gcc/graphite-isl-ast-to-gimple.c:2890:3: error: use of undeclared identifier 'isl_space_free'
  isl_space_free (schedule_space);
  ^
../../gcc/graphite-dependences.c:197:3: error: use of undeclared identifier 'isl_space_free'
  isl_space_free (space);
  ^
[... 99 lines cut ...]
../../gcc/graphite-sese-to-poly.c:265:53: error: use of undeclared identifier 'isl_space_copy'
      lhs = extract_affine (s, TREE_OPERAND (e, 0), isl_space_copy (space));
                                                    ^
../../gcc/graphite-sese-to-poly.c:529:8: error: use of undeclared identifier 'isl_space_domain'; did you mean 'isl_map_domain'?
                            isl_space_domain (isl_map_get_space (acc)));
                            ^~~~~~~~~~~~~~~~
                            isl_map_domain
/usr/local/include/isl/map.h:496:21: note: 'isl_map_domain' declared here
__isl_give isl_set *isl_map_domain(__isl_take isl_map *bmap);
                    ^
../../gcc/graphite-sese-to-poly.c:529:26: error: cannot initialize a parameter of type 'isl_map *' with an rvalue of type 'isl_space *'
                            isl_space_domain (isl_map_get_space (acc)));
                                              ^~~~~~~~~~~~~~~~~~~~~~~
/usr/local/include/isl/map.h:496:56: note: passing argument to parameter 'bmap' here
__isl_give isl_set *isl_map_domain(__isl_take isl_map *bmap);
                                                       ^
fatal error: too many errors emitted, stopping now [-ferror-limit=]
20 errors generated.
make[2]: *** [graphite-sese-to-poly.o] Error 1
rm gcc.pod
make[1]: *** [all-gcc] Error 2
make: *** [all] Error 2
```

I guess this might be a dup issue of #98. I tried to apply the patch in the instructions of that issue, but it didnt help

semonje, 2019-03-24:
I'm also having issues with this – and the instructions in the other issue(s) didn't work for me either. Mojave 10.14.3.

bonty, 2019-03-25:
Hello, I'm getting the same error as @semonje 
In my environment(Mojave 10.14.3), I've fixed this error applying the patch below

https://github.com/osx-cross/homebrew-avr/commit/ef705e991484b05742037a7ec2137fd346fb2fd4#diff-530f092d20e9f2db266aa3cc4858ae7cb

ladislas, 2019-03-25:
it should be working with 4560f36a976c4acbc1a472ebe5c06d43ccac7bc3 that I just merged into master.

I had removed the patch thinking it was fixed but turns out it is not yet.

I'll let you try and let me know if is works now.

ladislas, 2019-03-25:
It does compile on my machine:

```
→ brew config
HOMEBREW_VERSION: 2.0.5-40-geece1f0
ORIGIN: https://github.com/Homebrew/brew
HEAD: eece1f00f029e9ffcf07976dd0e001e6343ce4e1
Last commit: 16 hours ago
Core tap ORIGIN: https://github.com/Homebrew/homebrew-core
Core tap HEAD: 0bd7ca0bae2c5ce8ce10898d00362cc9114ed4b8
Core tap last commit: 3 hours ago
HOMEBREW_PREFIX: /usr/local
HOMEBREW_DEV_CMD_RUN: 1
HOMEBREW_DISPLAY: /private/tmp/com.apple.launchd.mFmGSKCOEt/org.macosforge.xquartz:0
HOMEBREW_LOGS: /Users/ladislas/Library/Logs/Homebrew
HOMEBREW_VISUAL: nvim
CPU: quad-core 64-bit broadwell
Homebrew Ruby: 2.3.7 => /System/Library/Frameworks/Ruby.framework/Versions/2.3/usr/bin/ruby
Clang: 10.0 build 1000
Git: 2.21.0 => /usr/local/bin/git
Curl: 7.54.0 => /usr/bin/curl
Java: 11
macOS: 10.14.3-x86_64
CLT: 10.1.0.0.1.1539992718
Xcode: 10.1
XQuartz: 2.7.11 => /opt/X11
```

[image]

ladislas, 2019-03-25:
I've updated gcc to 7.4.0 and the problem has been fixed, the patch is not needed anymore.

closed 2019-03-25 by ladislas

semonje, 2019-03-25:
Awesome, thanks!

henrikjonhed, 2019-03-25:
Works great now, thanks!

commit 7291524f (2019-03-25): update to gcc 7.4.0 - fix #98 + fix #101