osx-cross/homebrew-avr #103: Adding CI & Bottles to osx-cross/avr with Azure Pipelines

ladislas, 2019-03-27:
Hi @osx-cross/team --

This past few days, I've been updating some of the formulae and I've been trying to implement CI and Bottles generation for our tap to reduce the time it takes to install avr-gcc.

I was happily surprised to see that a lof of people are relying on our tap and that when something breaks, they let us know, as #98 and #101 have shown.

I first tried using Travis-CI but quickly ran into ~~stupid~~ limitations when building gcc: slow process, timeouts, log size just to name a few.

Other CI providers providing macOS images were usually expansive and did not feel like good/easy to setup solutions.

I kept googling and stumbled upon Azure Pipelines, which provide macOS images. It turns out that brew itself has moved to Azure Pipelines because it is easier to use, faster, cheaper and more reliable than Travis-CI.

It took me half a day to get it to work for avr-gcc and it compiles all versions of avr-gcc in about 30 minutes.

The PR is #102 

It will allow us to check that gcc updates and PR compile fine without having to do it on our own machines.

The next step will be to provide bottles for the different formulae. I'm still working on it.

larsimmisch, 2019-03-27:
Very cool! Thanks!

leojrfs, 2019-03-27:
Looks great! Thank you!

ladislas, 2019-04-18:
Quick update 🍺 

I've set up the Azure Pipeline and Bintray organizations to manage the process and bottles + patches (instead of using gist).

I've managed to build and publish a bottle for avr-binutils locally but I'm still figuring things out, the documentation on this subject is not that great and there seems to be a lot of concepts that are clear for homebrew developers but a bit cryptic for people coming from outside of the echo system.

I'm continuing development on my own fork to save you a lot of back and forth with testing the scripts.

If you want to test avr-binutils, you can 

```
brew install ladislas/avr/avr-binutils
```

ladislas, 2019-06-05:
@osx-cross/team Good news! 🎉

Azure Pipelines is up and running a bottles are already available for avarice, binutils and avr-gcc 9.1.0.

I'll keep adding the rest of the formulae.

The process is fully automated so any PR with a version bump will have its bottles update. A maintainer will then need to run `brew pull --bottle`

Everything is explained here: https://github.com/ladislas/homebrew-greetings

I'll close the issue now :)

closed 2019-06-05 by ladislas