osx-cross/homebrew-avr #124: avr-gcc includes outdated avr/io128a1u.h

mokus0, 2019-09-26:
When attempting to "upgrade" from Microchip's ancient avr-gcc (version 5.4.0), I find that I can no longer compile code for ATXMEGA128A1U because the avr-gcc package here includes a part-support header (avr/io128a1u.h) from 2010 - 5 years older than the one Microchip is shipping.  This older version has several serious issues such as missing and misspelled constants.

I can work around it by including a newer copy in my project, but I figured it was worth reporting anyway.

larsimmisch, 2019-09-26:
Yes, thank you! But we are the wrong project to report that to. I think this bug should be filed for avr-libc: http://savannah.nongnu.org/projects/avr-libc/ 

Can you file it there?

closed 2019-09-26 by larsimmisch

mokus0, 2019-09-26:
Thanks, I'll look into that.

larsimmisch, 2019-09-26:
Please do. Jörg will probably ask you for a patch, though.