osx-cross/homebrew-avr #129: macOS 10.15 Catalina Support

borninla, 2019-10-08:
Not sure if this is where I should ask, but is there a timeline for Catalina support?

ladislas, 2019-10-09:
as soon as it's available in Azure Pipelines.

for the moment you can compile from source. it should take around 40 minutes on a MacBook Pro Retina, 13-inch, Early 2015.

ladislas, 2019-10-09:
FYI -- https://github.com/microsoft/azure-pipelines-image-generation/issues/1277#issuecomment-539598428

closed 2019-10-09 by ladislas

ladislas, 2019-11-30:
@borninla I just upgraded to Catalina with a clean install and when running `brew install avr-gcc` it installed the Mojave bottle without any issues.

Seems to be working fine for the moment.

as soon as Azure Pipelines adds Catalina support, I'll update the build process.