osx-cross/homebrew-avr #130: Port avr-size patch to 2.33.1

anatol, 2019-10-14:
Binutils modified size.c a lot and now it conflicts with the patch referenced here

https://github.com/osx-cross/homebrew-avr/blob/e1df340224bf42372fb922e408e516c484056dce/Formula/avr-binutils.rb#L20

Arch Linux made an attempt to port it over to 2.33.1 and the current result is here
https://git.archlinux.org/svntogit/community.git/tree/trunk/avr-size.patch?h=packages/avr-binutils

It would be great to have a unified patch across all distros to have a consistent and well-tested functionality.

ladislas, 2019-10-14:
I'm okay with doing that. Can you locally test that the patch runs without any errors on 2.33.1?

anatol, 2019-10-14:
The patch compiles fine with binutils 2.33.1. I tested `avr-size` locally and it looks great so far. Arch Linux also uses this patch with the test version of `avr-binutils`.

ladislas, 2019-10-24:
Thanks @anatol! I'll change that asap unless you want to make a PR.

ladislas, 2019-11-29:
@anatol I tried updating our patch to use the arch linux version but it failed because of different headers in the diff.

the error is here: https://dev.azure.com/osx-cross/homebrew-avr/_build/results?buildId=570&view=logs&j=f48822f6-9f12-56e2-121b-f56c97331bbb&t=d24f701f-ca91-5a58-1360-f8e5d6da5696&l=139

I fixed it by changing the header and publishing the patch on bintray.

If you'd like to work on a PR to fix that, you're welcome to do so.

anatol, 2019-11-29:
ArchLinux uses `patch -p1 < avr-size.patch` to patch in the file. You could either:
* use `-p1` instead of `-p0` command line option
* modify lines 3/4 in my patch and remove filename prefixes `a/` and `b/`

ladislas, 2019-11-29:
> use -p1 instead of -p0 command line option
I'll try that.

ladislas, 2019-11-30:
So actually is was a wrong config on our side, the formula was passing `:p0`. Removing that defaulted to `:p1` and fixed the issue.

Thanks for the help @anatol!

commit d5a3cf02 (2019-11-30): Change avr binutils patch path (#135)

- use archlinux patch
- closes #130
closed 2019-11-30 by ladislas